openbmc/openbmc #3858: [Add Sensor] phosphor-hwmon is running，but web /hardware-status/sensors is null.

qkpidan, 2022-05-19:
**Note on Bugs**
I cant see anything in  #/hardware-status/sensors

**Unexpected behavior you saw**
phosphor-hwmon is running，but web /hardware-status/sensors is null.

**Expected behavior**
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Modify  opbmc - qemu/hw/arm/aspeed.c  line 540.
[image]

2. Add phosphor-hwmon_%.bbappend
```bash
FILESEXTRAPATHS:prepend:hkzy-ast2500 := "${THISDIR}/${PN}:"
EXTRA_OECONF:append:hkzy-ast2500 = " --enable-negative-errno-on-fail"

CHIPS = " \
        bus@1e78a000/i2c-bus@80/tmp105@48 \
        bus@1e78a000/i2c-bus@80/tmp105@49 \
        bus@1e78a000/i2c-bus@80/tmp105@4a \
        bus@1e78a000/i2c-bus@80/tmp105@4b \
        "
ITEMSFMT = "ahb/apb/{0}.conf"

ITEMS = "${@compose_list(d, 'ITEMSFMT', 'CHIPS')}"
ENVS = "obmc/hwmon/{0}"

SYSTEMD_ENVIRONMENT_FILE:${PN}:append:hkzy-ast2500 = " ${@compose_list(d, 'ENVS', 'ITEMS')}"
```

3. Modify  dtsi
[image]

5. See web-ui

**Screenshots**
[image]

[image]

[image]

[image]

[image]

**OpenBMC Information:**
 - /etc/os-release:
[image]

 - uname -a:
[image]

williamspatrick, 2022-05-19:
Please don't send screenshots of text and logs.  The synopsis of this issue is "I made some changes and things don't work as I expected".  I'm not sure how you expect others to debug this for you.

Are there sensor objects being created by the phosphor-hwmon processes on dbus?  Do they contain the values you expect?  If so, do the Redfish APIs give the sensor values you expect?  If so, does the webui console have any errors?

closed 2022-05-19 by williamspatrick

qkpidan, 2022-05-19:
> 

Got it. 
I am sorry about that. It wont happen again.

xyz.openbmc_project.Sensor is not processes on dbus, how can i fix it?

Thanks!

williamspatrick, 2022-05-19:
> xyz.openbmc_project.Sensor is not processes on dbus, how can i fix it?

I'm not sure what you mean by this or what you're expecting to see or what you're running to determine this.

`phosphor-hwmon` has a number of processes and you showed them running in your screenshot.  Each one of those will have its own dbus-service.  Running `busctl` by itself will show you all the current connections.  Pick the ones that are related to the `phosphor-hwmon` process (hopefully it is obvious) and run `busctl tree <service>`.  That will hopefully show you the path to the sensor object(s).  You can then run `busctl introspect <service> <path>` to get the interfaces and values at those object paths.

spinler, 2022-05-19:
Are you using phosphor-inventory-manager or entity-manager to manage your inventory?  I would suggest entity-manager, and then you can follow https://github.com/openbmc/entity-manager/blob/master/docs/my_first_sensors.md.  If you want to use phosphor-hwmon with phosphor-inventory-manager, then you need to manually add associations to the chassis so that redfish can find them.  https://github.com/openbmc/phosphor-inventory-manager#creating-associations describes that (though as mentioned above you'd need to check that phosphor-hwmon is actually hosting those sensors on D-Bus first)

AloneWarrior11011, 2025-04-09:
> **Note on Bugs** I cant see anything in #/hardware-status/sensors
> 
> **Unexpected behavior you saw** phosphor-hwmon is running，but web /hardware-status/sensors is null. [image]
> 
> **Expected behavior** [image]
> 
> **To Reproduce** Steps to reproduce the behavior:
> 
>     1. Modify  opbmc - qemu/hw/arm/aspeed.c  line 540.
>        [image]
> 
>     2. Add phosphor-hwmon_%.bbappend
> 
> 
> FILESEXTRAPATHS:prepend:hkzy-ast2500 := "${THISDIR}/${PN}:"
> EXTRA_OECONF:append:hkzy-ast2500 = " --enable-negative-errno-on-fail"
> 
> CHIPS = " \
>         bus@1e78a000/i2c-bus@80/tmp105@48 \
>         bus@1e78a000/i2c-bus@80/tmp105@49 \
>         bus@1e78a000/i2c-bus@80/tmp105@4a \
>         bus@1e78a000/i2c-bus@80/tmp105@4b \
>         "
> ITEMSFMT = "ahb/apb/{0}.conf"
> 
> ITEMS = "${@compose_list(d, 'ITEMSFMT', 'CHIPS')}"
> ENVS = "obmc/hwmon/{0}"
> 
> SYSTEMD_ENVIRONMENT_FILE:${PN}:append:hkzy-ast2500 = " ${@compose_list(d, 'ENVS', 'ITEMS')}"
> 
>     3. Modify  dtsi
>        [image]
> 
>     4. See web-ui
>        [image]
> 
> 
> **Screenshots** [image]
> 
> [image]
> 
> [image]
> 
> [image]
> 
> [image]
> 
> **OpenBMC Information:**
> 
>     * /etc/os-release:
>       [image]
> 
>     * uname -a:
>       [image]

Hey where do you have modify dtsi file to specify your sensor, could you provide the exact path on where I can add my tmp75 sensor data like you have add like : 
&i2c1{
   ....
   ....
};

in my case I have used tmp75 sensor which are found at bus number 1 on i2c address 0x4d.  on Raspberrypi 3