openbmc/openbmc #3861: How to access  ESPICFG register on ast2500 ?

junhengdi, 2022-05-28:
I want to access ESPICFG register  such as ESPICFG004 ,ESPICFG008  on ast2500 . But I do not know the base address of ESPICFG register.In ast2500 datasheet , only ESPI controller base address 0x1e6ee000 ,but it only can access ESPI register such as ESPI004, ESPI008.  I want to known how to access the ESPICFG register. thank you.

williamspatrick, 2022-05-29:
Please contact Aspeed for this information.

closed 2022-05-29 by williamspatrick

jk-ozlabs, 2022-05-29:
The `ESPICFG` registers aren't part of the MMIO register set mapped at 0x1e6ee000. Instead, these are the addresses of various configuration registers on the eSPI bus, accessed via `GET_CONFIGURATION` and `SET_CONFIGURATION` commands. So, they're available to the ESPI peer device (typically your host machine) rather than the BMC firmware.

The register set matches the set defined in section 7.2 of the eSPI base specification.