openbmc/openbmc #3862: [aspeed] uboot can't boot into the kernel

cmao0914, 2022-05-30:
**Description**
try to build an image with the existing ast2600 setting in a new meta-layer, and test if it can boot into kernel but failed.

**Unexpected behavior**
uboot printed 
`Wrong Image Format for bootm command`
`ERROR: can't get kernel image!`
`ast#` 

**Expected behavior**
uboot should jump to the kernel successfully

**Steps To Reproduce**
1. initialize a meta-layer and append the conf below
[image]

2. apply the existing kernel&uboot device tree or append the files with bbappend
3. build image **Cmd: . setup $your_machine**
**Screenshots**
[image]

**OpenBMC Information:**
uboot: v2019.04+gitAUTOINC+709868b19a
kernel: 5.15.41+gitAUTOINC+f7c5c0b497
openbmc branch commit id: ed14f8a39

**Additional context**
i checked out the code and created a new meta-layer to put my settings in the conf file, no other files were changed except for the DTS files.

cmao0914, 2022-05-31:
Hi, 
i updated my conf according to evb-ast2600.conf as follows,
but still got the same error, can anyone help me?

**my conf:**
KERNEL_DEVICETREE = "aspeed-ast2600-evb.dtb"
require conf/machine/include/ast2600.inc
require conf/machine/include/obmc-bsp-common.inc
UBOOT_MACHINE = "ast2600_openbmc_spl_defconfig"
SPL_BINARY = "spl/u-boot-spl.bin"
UBOOT_DEVICETREE = "ast2600-evb"
SOCSEC_SIGN_ENABLE = "0"
FLASH_SIZE = "65536"

**console output:**
U-Boot SPL 2019.04 (May 31 2022 - 09:05:08 +0000)
Trying to boot from RAM

U-Boot 2019.04 (May 31 2022 - 09:05:08 +0000)

SOC: AST2600-A3
RST: Power On
FMC 2nd Boot (ABR): Enable, Dual flashes, Source: Primary
eSPI Mode: SIO:Enable : SuperIO-2e
Eth: MAC0: RMII/NCSI, MAC1: RMII/NCSI, MAC2: RMII/NCSI, MAC3: RMII/NCSI
Model: AST2600 EVB
DRAM:  already initialized, 2 GiB (capacity:2048 MiB, VGA:16 MiB, ECC:off)
MMC:

sdhci_slot0@100: 1, sdhci_slot1@200: 2, emmc_slot0@100: 0
Loading Environment from SPI Flash... SF: Detected w25q512jv with page size 256 Bytes, erase size 4 KiB, total 64 MiB
*** Warning - bad CRC, using default environment

In:    serial@1e784000
Out:   serial@1e784000
Err:   serial@1e784000
Model: AST2600 EVB
Net:   PHY reset timed out

Warning: ftgmac@1e660000 (eth0) using random MAC address - 6a:1d:8f:7b:c6:26
eth0: ftgmac@1e660000
Warning: ftgmac@1e680000 (eth1) using random MAC address - 12:ee:10:1a:5c:1d
, eth1: ftgmac@1e680000PHY reset timed out

Warning: ftgmac@1e670000 (eth2) using random MAC address - da:69:ee:0e:56:59
, eth2: ftgmac@1e670000PHY reset timed out

Warning: ftgmac@1e690000 (eth3) using random MAC address - 86:0a:15:e1:47:bb
, eth3: ftgmac@1e690000
Hit any key to stop autoboot:  0
Wrong Image Format for bootm command
ERROR: can't get kernel image!
ast# printenv
baudrate=115200
bootargs=console=ttyS4,115200n8 root=/dev/ram rw
bootcmd=bootm 20100000
bootdelay=2
bootfile=all.bin
bootspi=fdt addr 20100000 && fdt header get fitsize totalsize && cp.b 20100000 ${loadaddr} ${fitsize} && bootm; echo Error loading kernel FIT image
fdtcontroladdr=fcf8d700
gatewayip=192.168.0.1
ipaddr=192.168.0.45
loadaddr=0x83000000
netmask=255.255.255.0
serverip=192.168.0.81
stderr=serial@1e784000
stdin=serial@1e784000
stdout=serial@1e784000
verify=yes

Environment size: 475/131068 bytes
ast#

williamspatrick, 2022-05-31:
`Wrong Image Format for bootm command
ERROR: can't get kernel image!`

You likely have a mismatch between the flash address given to u-boot and how you’ve assembled the image. This doesn’t look like a bug.

closed 2022-05-31 by williamspatrick

williamspatrick, 2022-05-31:
The `evb-ast2600` builds and boots in QEMU fine (I just checked).  By layout, I'm referring not just to the order but the exact addresses.  I suspect something along the lines of the FLASH_SIZE variable isn't taking effect correctly.

You'll want to look at what these addresses (ex. `FLASH_KERNEL_OFFSET`) end up being for both your system config and the `evb-ast2600`.

```
meta-phosphor/classes/image_types_phosphor.bbclass:FLASH_KERNEL_OFFSET:flash-65536 ?= "1024"
meta-phosphor/classes/image_types_phosphor.bbclass:FLASH_ROFS_OFFSET:flash-65536 ?= "10240"
meta-phosphor/classes/image_types_phosphor.bbclass:FLASH_RWFS_OFFSET:flash-65536 ?= "43008"
```

cmao0914, 2022-06-01:
Hi @williamspatrick ,
I built evb-ast2600 and it booted in QEMU fine, too. Then, I programmed the evb-ast2600 image to my SPI rom(w25q512jv) and then ran it to fail. I had also built the other platform images such as transformers and lannister besides evb-ast2600, they all got the same problem, and I have no clue about this situation. Would you mind if sending your evt-ast2600 image to me for testing? I will run it on the actual machine. Although, my machine is not ast2600 evb, but I think it should boot into kernel .

About your suggestions, I had checked it and no errors were found.

## QEMU output ##
./qemu-system-arm -m 1024 -M ast2600-evb -nographic     -drive file=./flash-evb-ast2600,format=raw,if=mtd     -net nic     -net user,hostfwd=:127.0.0.1:2222-:22,hostfwd=:127.0.0.1:2443-:443,hostfwd=udp:127.0.0.1:2623-:623,hostname=qemu
qemu-system-arm: warning: Aspeed iBT has no chardev backend
qemu-system-arm: warning: nic ftgmac100.1 has no peer
qemu-system-arm: warning: nic ftgmac100.2 has no peer
qemu-system-arm: warning: nic ftgmac100.3 has no peer

U-Boot SPL 2019.04 (May 17 2022 - 09:32:06 +0000)
Trying to boot from RAM

U-Boot 2019.04 (May 17 2022 - 09:32:06 +0000)

SOC: AST2600-A3
Secure Boot: Mode_2, RSA4096_SHA512
eSPI Mode: SIO:Enable : SuperIO-2e
Eth: MAC0: RGMII, MAC1: RGMII, MAC2: RGMII, MAC3: RGMII
Model: AST2600 EVB
DRAM:  already initialized, 1008 MiB (capacity:1024 MiB, VGA:64 MiB, ECC:off)
MMC:

sdhci_slot0@100: 1, sdhci_slot1@200: 2, emmc_slot0@100: 0
Loading Environment from SPI Flash... SF: Detected mx66u51235f with page size 256 Bytes, erase size 4 KiB, total 64 MiB
*** Warning - bad CRC, using default environment

In:    serial@1e784000
Out:   serial@1e784000
Err:   serial@1e784000
Model: AST2600 EVB
Net:
Warning: ftgmac@1e660000 (eth0) using random MAC address - 0e:b6:db:0a:f3:7a
eth0: ftgmac@1e660000
Warning: ftgmac@1e680000 (eth1) using random MAC address - ba:44:e4:29:3b:73
, eth1: ftgmac@1e680000
Warning: ftgmac@1e670000 (eth2) using random MAC address - 26:1a:d8:ca:37:cc
, eth2: ftgmac@1e670000
Warning: ftgmac@1e690000 (eth3) using random MAC address - 0a:4e:27:49:e3:a2
, eth3: ftgmac@1e690000
Hit any key to stop autoboot:  0
\## Loading kernel from FIT Image at 20100000 ...
   Using 'conf-aspeed-ast2600-evb.dtb' configuration
   Trying 'kernel-1' kernel subimage
     Description:  Linux kernel
     Type:         Kernel Image
     Compression:  uncompressed
     Data Start:   0x2010012c
     Data Size:    3831288 Bytes = 3.7 MiB
     Architecture: ARM
     OS:           Linux
     Load Address: 0x80001000
     Entry Point:  0x80001000
     Hash algo:    sha256
     Hash value:   7f54f99a478d8c76576d459d7ad27d04abc7fe9de70ad26d4316d39c6a5c57e0
   Verifying Hash Integrity ... sha256+ OK
\``````````

## console output ##
U-Boot SPL 2019.04 (May 17 2022 - 09:32:06 +0000)
Trying to boot from RAM

U-Boot 2019.04 (May 17 2022 - 09:32:06 +0000)

SOC: AST2600-A3
RST: Power On
FMC 2nd Boot (ABR): Enable, Dual flashes, Source: Primary
eSPI Mode: SIO:Enable : SuperIO-2e
Eth: MAC0: RMII/NCSI, MAC1: RMII/NCSI, MAC2: RMII/NCSI, MAC3: RMII/NCSI
Model: AST2600 EVB
DRAM:  already initialized, 2 GiB (capacity:2048 MiB, VGA:16 MiB, ECC:off)
MMC:

sdhci_slot0@100: 1, sdhci_slot1@200: 2, emmc_slot0@100: 0
Loading Environment from SPI Flash... SF: Detected w25q512jv with page size 256 Bytes, erase size 4 KiB, total 64 MiB
*** Warning - bad CRC, using default environment

In:    serial@1e784000
Out:   serial@1e784000
Err:   serial@1e784000
Model: AST2600 EVB
Net:   PHY reset timed out

Warning: ftgmac@1e660000 (eth0) using random MAC address - 72:47:b7:94:fe:fe
eth0: ftgmac@1e660000
Warning: ftgmac@1e680000 (eth1) using random MAC address - ee:b3:6b:b0:9b:51
, eth1: ftgmac@1e680000PHY reset timed out

Warning: ftgmac@1e670000 (eth2) using random MAC address - 2e:3c:bc:de:1d:19
, eth2: ftgmac@1e670000PHY reset timed out

Warning: ftgmac@1e690000 (eth3) using random MAC address - 4a:04:ba:91:e5:f3
, eth3: ftgmac@1e690000
Hit any key to stop autoboot:  0
Wrong Image Format for bootm command
ERROR: can't get kernel image!
ast#

LeeTroy, 2022-06-01:
Do you want to try prebuilt aspeed sdk image?
https://github.com/AspeedTech-BMC/openbmc/releases/download/v08.01/ast2600-default-obmc.tar.gz

cmao0914, 2022-06-01:
Hi @LeeTroy ,
no problem, i've downloaded it and will try `obmc-phosphor-image-ast2600-default.static.mtd` tomorrow, thank you!

cmao0914, 2022-06-02:
Hi @LeeTroy ,
I've run your image on my platform and got the same problem. Do you have any idea? In addition, I've downloaded meta openbmc from their git, and built the platform image called meta-grandcanyon(ast2600). It can boot into kernel successfully, please refer to the log below.

**console output from ast2600-defeault image:**
U-Boot 2019.04-v00.04.10 (Apr 11 2022 - 03:22:53 +0000)

SOC: AST2600-A3
RST: Power On
FMC 2nd Boot (ABR): Enable, Dual flashes, Source: Primary
eSPI Mode: SIO:Enable : SuperIO-2e
Eth: MAC0: RMII/NCSI, MAC1: RMII/NCSI, MAC2: RMII/NCSI, MAC3: RMII/NCSI
Model: AST2600 EVB
DRAM:  already initialized, 2 GiB (capacity:2048 MiB, VGA:16 MiB, ECC:off)
RC Bridge phy@1e6ed200 : Link down
MMC:

sdhci_slot0@100: 1, sdhci_slot1@200: 2, emmc_slot0@100: 0
Loading Environment from SPI Flash... SF: Detected w25q512jv with page size 256 Bytes, erase size 64 KiB, total 64 MiB
*** Warning - bad CRC, using default environment

In:    serial@1e784000
Out:   serial@1e784000
Err:   serial@1e784000
Model: AST2600 EVB
Net:
Warning: ftgmac@1e660000 (eth0) using random MAC address - 12:6a:ed:d7:4d:69
eth0: ftgmac@1e660000
Warning: ftgmac@1e680000 (eth1) using random MAC address - 72:e6:2a:ff:14:4c
, eth1: ftgmac@1e680000Could not get PHY for eth2: addr 0
Failed to initialize PHY: -19
, eth-1: ftgmac@1e670000Could not get PHY for eth3: addr 0
Failed to initialize PHY: -19
, eth-1: ftgmac@1e690000
Hit any key to stop autoboot:  0
libfdt fdt_check_header(): FDT_ERR_BADMAGIC
Error loading kernel FIT image
ast#

**console output from meta flash-grandcanyon image:**
U-Boot SPL 2019.04 grandcanyon-2479bee3e2 (May 31 2022 - 14:07:53 +0000)
SYS_INIT_RAM_END=10016000
CONFIG_SYS_INIT_SP_ADDR=10015000
CONFIG_MALLOC_F_ADDR=10012000
gd = sp = 10011f10
fdt=000182b8
Setup flash: write enable, addr4B, CE1 AHB 64MB window
Setup FMC_CE_CTRL = 0x00002A33
Watchdog: 300s
hwstrap write protect SCU508=0x40080000, SCU518=0x00000010
Before: CE0_CTRL=0x00000600, CE1_CTRL=0x00000400
cs0_status = 2, cs1_status = 2
After: CE0_CTRL=0x00000400, CE1_CTRL=0x00000400
SPL Could not find TPM (ret=-5)
vboot_reset 553
vboot_check_fit 245
SPL Could not find TPM (ret=-5)
Booting recovery U-Boot.
Blindly jumping to 0x20040000

U-Boot 2019.04 grandcanyon-2479bee3e2 (May 31 2022 - 14:07:53 +0000)

SOC: AST2600-A3
RST: Power On
FMC 2nd Boot (ABR): Enable, Dual flashes, Source: Primary
eSPI Mode: SIO:Enable : SuperIO-2e
Eth: MAC0: RMII/NCSI, MAC1: RMII/NCSI, MAC2: RMII/NCSI, MAC3: RMII/NCSI
Model: Aspeed BMC
DRAM:  1.8 GiB (capacity:2048 MiB, VGA:16 MiB, ECC:on, ECC size:1806 MiB)
Watchdog: 300s
MMC:
In:    serial
Out:   serial
Err:   serial
Model: Aspeed BMC
Net:   No MDIO found.
No ethernet found.
autoboot in 3 seconds (stop with 'Delete' key)...
\## Loading kernel from FIT Image at 201a0000 ...
   Using 'conf@1' configuration
   Trying 'kernel@1' kernel subimage
     Description:  Linux kernel
     Type:         Kernel Image
     Compression:  uncompressed
     Data Start:   0x201a00e8
     Data Size:    3414864 Bytes = 3.3 MiB
     Architecture: ARM
     OS:           Linux
     Load Address: 0x80008000
     Entry Point:  0x80008000
     Hash algo:    sha256
     Hash value:   7ccee3949d69d8ab7e304c4d22b2dc46f0c7fd9c7f3e8f9efa1ca935ae18f88a
\## Loading ramdisk from FIT Image at 201a0000 ...
   Using 'conf@1' configuration
   Trying 'ramdisk@1' ramdisk subimage
     Description:  RAMDISK
     Type:         RAMDisk Image
     Compression:  lzma compressed
     Data Start:   0x204e1d28
     Data Size:    25661019 Bytes = 24.5 MiB
     Architecture: ARM
     OS:           Linux
     Load Address: unavailable
     Entry Point:  unavailable
     Hash algo:    sha256
     Hash value:   32e131007fb2eb591501efbd8f657908be5837a174bb242685fe68f9263994c0
\## Loading fdt from FIT Image at 201a0000 ...
   Using 'conf@1' configuration
   Trying 'fdt@1' fdt subimage
     Description:  Flattened Device Tree blob
     Type:         Flat Device Tree
     Compression:  uncompressed
     Data Start:   0x21d5ac64
     Data Size:    47709 Bytes = 46.6 KiB
     Architecture: ARM
     Hash algo:    sha256
     Hash value:   891beef925b373dd8455dfdbfdb443a8afd33a76d89434d5da4989cb7e2d54e1
   Booting using the fdt blob at 0x21d5ac64
   Loading Kernel Image ... OK
   Loading Ramdisk to 9e787000, end 9ffffe5b ... OK
   Loading Device Tree to 8fff1000, end 8ffffa5c ... OK

measure vbs...tpm_tis_spi_probe() tpm_tis_spi_probe: missing reset GPIO
tpm_tis_spi_xfer() tpm_tis_spi_xfer: timeout on wait state
tpm_tis_spi_probe() tpm_tis_spi_probe: no device found
SPL Could not find TPM (ret=-5)
done
measure OS-Kernel...tpm_tis_spi_probe() tpm_tis_spi_probe: missing reset GPIO
tpm_tis_spi_xfer() tpm_tis_spi_xfer: timeout on wait state
tpm_tis_spi_probe() tpm_tis_spi_probe: no device found
SPL Could not find TPM (ret=-5)
done
measure Ramdisk...tpm_tis_spi_probe() tpm_tis_spi_probe: missing reset GPIO
tpm_tis_spi_xfer() tpm_tis_spi_xfer: timeout on wait state
tpm_tis_spi_probe() tpm_tis_spi_probe: no device found
SPL Could not find TPM (ret=-5)
done
measure Fdt...tpm_tis_spi_probe() tpm_tis_spi_probe: missing reset GPIO
tpm_tis_spi_xfer() tpm_tis_spi_xfer: timeout on wait state
tpm_tis_spi_probe() tpm_tis_spi_probe: no device found
SPL Could not find TPM (ret=-5)
done
tpm_tis_spi_probe() tpm_tis_spi_probe: missing reset GPIO
tpm_tis_spi_xfer() tpm_tis_spi_xfer: timeout on wait state
tpm_tis_spi_probe() tpm_tis_spi_probe: no device found
SPL Could not find TPM (ret=-5)

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0xf00

zpdsg123, 2022-09-17:
> Hi @LeeTroy , I've run your image on my platform and got the same problem. Do you have any idea? In addition, I've downloaded meta openbmc from their git, and built the platform image called [...]
Have you solved this issue? I met this problem recently,too.

wangzcheng, 2023-05-24:
Hi, @cmao0914 @zip123zip. Have you solved this problem? Now I have encountered similar problems, but my machine is an ast2500 and uses a 64M gd25b512m. However, when I use another model of 64M flash, it can function normally.