openbmc/openbmc #3863: Unable to flash the firmware to E3C246D4I

hamidrm, 2022-06-06:
I wrote an OpenBMC image to my E3C246D4I SPI flash by "socflash" which had built before and it works well curentlly.
But Now, I'm trying to write the original BMC firmware (provided by ASRock). But socflash shows below error:
"the BMC is protected, skip to update the firmware image!!!"
Also, this error still shows when I'm trying to write previous OpenBMC image.
How can I write to the SPI flash?

zevweiss, 2022-06-06:
`socflash` operates via a mechanism that most in the OpenBMC community consider a security vulnerability in the BMC (the host being able to overwrite the BMC's firmware without any sort of authentication is something I think most of us working on OpenBMC feel should not be allowed).  For details see [CVE-2019-6260, a.k.a. "pantsdown"](https://www.flamingspork.com/blog/2019/01/23/cve-2019-6260-gaining-control-of-bmc-from-the-host-processor/).

For updating to a different version of OpenBMC (e.g. a newer build), the `Operations -> Firmware -> Update firmware` feature in the web UI should work, or you can use a REST-based method as described [here](https://github.com/openbmc/docs/blob/master/architecture/code-update/code-update.md) and/or [here](https://github.com/openbmc/docs/blob/master/architecture/code-update/code-update-deprecated.md) (the docs aren't currently in a great state unfortunately).  If the native firmware-update mechanism doesn't work, please provide details and we can hopefully debug it.

For reverting to the factory firmware I think in _principle_ it should work to upload the firmware image to `/run/initramfs/image-bmc` and reboot, but I seem to recall trying that at some point and encountering problems, so to be on the safe side it might be best to not just dive right in and try that, since there's a chance it may leave your BMC somewhat bricked (though my guess is that if it did would probably be bricked with the "pantsdown" vulnerability described above still open, so you could _probably_ still recover it from the host via `socflash`, or @amboar's open source alternative [`culvert`](https://github.com/amboar/culvert)).  If you do want to pursue reverting to the factory firmware, do you have a BMC serial console available?  (That would make it easier to do so in a slightly more fool-proof way.)

hamidrm, 2022-06-07:
> `socflash` operates via a mechanism that most in the OpenBMC community consider a security vulnerability in the BMC (the host being able to overwrite the BMC's firmware without any sort of [...]

Thanks for your complete description 🙏

closed 2022-06-07 by hamidrm