openbmc/openbmc #3864: issue with hello world sdk

sdivcom-dotcom, 2022-06-07:
Good afternoon all. I have the following error when I run the command from the documentation https://github.com/openbmc/docs/blob/master/development/sdk-hello-world.md, namely ninja -C build:
/code/phosphor-state-manager$ ninja -C build
ninja: Entering directory `build'
[5/693] Compiling C++ object subprojects/stdplus/src/libstdplus.so.0.1.p/stdplus_fd_create.cpp.o
FAILED: subprojects/stdplus/src/libstdplus.so.0.1.p/stdplus_fd_create.cpp.o 
c++ -Isubprojects/stdplus/src/libstdplus.so.0.1.p -Isubprojects/stdplus/src -I../subprojects/stdplus/src -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wnon-virtual-dtor -Wextra -Wpedantic -Werror -std=c++2a -g -fPIC -MD -MQ subprojects/stdplus/src/libstdplus.so.0.1.p/stdplus_fd_create.cpp.o -MF subprojects/stdplus/src/libstdplus.so.0.1.p/stdplus_fd_create.cpp.o.d -o subprojects/stdplus/src/libstdplus.so.0.1.p/stdplus_fd_create.cpp.o -c ../subprojects/stdplus/src/stdplus/fd/create.cpp
In file included from ../subprojects/stdplus/src/stdplus/fd/impl.hpp:2,
                 from ../subprojects/stdplus/src/stdplus/fd/dupable.hpp:2,
                 from ../subprojects/stdplus/src/stdplus/fd/create.hpp:4,
                 from ../subprojects/stdplus/src/stdplus/fd/create.cpp:3:
../subprojects/stdplus/src/stdplus/fd/intf.hpp:5:10: fatal error: span: No such file or directory
    5 | #include <span>
      |          ^~~~~~
compilation terminated.
[6/693] Compiling C++ object subprojects/stdplus/src/libstdplus.so.0.1.p/stdplus_exception.cpp.o
FAILED: subprojects/stdplus/src/libstdplus.so.0.1.p/stdplus_exception.cpp.o 
c++ -Isubprojects/stdplus/src/libstdplus.so.0.1.p -Isubprojects/stdplus/src -I../subprojects/stdplus/src -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wnon-virtual-dtor -Wextra -Wpedantic -Werror -std=c++2a -g -fPIC -MD -MQ subprojects/stdplus/src/libstdplus.so.0.1.p/stdplus_exception.cpp.o -MF subprojects/stdplus/src/libstdplus.so.0.1.p/stdplus_exception.cpp.o.d -o subprojects/stdplus/src/libstdplus.so.0.1.p/stdplus_exception.cpp.o -c ../subprojects/stdplus/src/stdplus/exception.cpp
In file included from ../subprojects/stdplus/src/stdplus/exception.cpp:1:
../subprojects/stdplus/src/stdplus/exception.hpp:7:19: error: missing binary operator before token "("
    7 | #if !__has_builtin(__builtin_LINE)
      |                   ^
../subprojects/stdplus/src/stdplus/exception.cpp:4:19: error: missing binary operator before token "("
    4 | #if !__has_builtin(__builtin_LINE)
      |                   ^
[10/693] Compiling C++ object subprojects/stdplus/src/libstdplus.so.0.1.p/stdplus_dl.cpp.o
ninja: build stopped: subcommand failed.
Please help me with this problem!

sdivcom-dotcom, 2022-06-08:
> It doesn't seem like you have entered the SDK, have you? I see that 'ninja' is calling `c++` instead of some `arm32....-c++` executable.
> 
> It appears that it is attempting to use your build system's `c++` instead of the one from the SDK and whatever version that is doesn't fully support C++20 (based on the `span: No such file or directory`).

Did I understand correctly that you need to run in sdk? How do I do that?

williamspatrick, 2022-06-08:
> Did I understand correctly that you need to run in sdk? How do I do that?

The document you linked to is titled "in SDK" and the second line of it says:

```
Prerequisites: Completed Development Environment Setup [Document](https://github.com/openbmc/docs/blob/master/development/dev-environment.md)
```

So, if you want to follow _those_ instructions then yes, you need to have followed the Prerequisite document which involves setting up the SDK.  It also shows how to set up QEMU, which later instructions in this document describe how to run the code in QEMU.

Do you _have_ to use the SDK to build code?  No.  But, I don't know what you're trying to accomplish.  If you want to cross-compile an application to an ARM-based BMC then using the SDK for development and then writing a Yocto recipe when you're done is probably your best approach.  If you are just looking to compile the application for x86 for some early development, then you don't have to use an SDK, but based on the errors you showed you are not using a Linux distro with a new enough compiler in order to be successful.  You might look into using a Docker container to compiler your code if you're trying to go this route.

Please give some more details on what you're trying to accomplish if the SDK is not the route you want to go.

sdivcom-dotcom, 2022-06-09:
> The document you linked to is titled "in SDK" and the second line of it says:
>
> ```
> Prerequisites: Completed Development Environment Setup [...]

Thank you I set up the environment and I tried to run in it, but it didn't work. I am learning openbmc and just wanted to learn sdk and nothing more

sdivcom-dotcom, 2022-06-13:
> 

I would like to be able to debug without building the whole image. And I thought sdk is a good option. How can I build and debug my own modules?

williamspatrick, 2022-06-13:
@sdivcom-dotcom : Yes, SDK is a fine option for doing that.  We've tried to document how to do that with the SDK.

You've said it doesn't work but you haven't given me any details on what you're doing and what part of it doesn't work.  I can't really help you without any information.

Can you please:

1. Open a new shell.
2. Enter the SDK.
3. Attempt to build your code.

And show a log of what exactly you've done to get to step 3.  If you can show me how you get to step 3 and how it fails, we will be much more likely to be able to help you out.

sdivcom-dotcom, 2022-06-14:
> When you run step 2006 ./oecore-x86_64-arm1176jzs-toolchain-nodistro.0.sh, you should see an output similar to this:
Yes, I'm getting this kind of text. 
I did as you described.
 2022 . /home/sem/sdk/romulus-14.06.22/environment-setup-arm1176jzs-openbmc-linux-gnueabi
 2023 meson build
And I got the following error.

Could not find any valid candidates for the cross files: /home/sem/sdk/romulus-14.06.22/sysroots/x86_64-oesdk-linux/usr/share/meson/arm-openbmc-linux-gnueabi-meson.cross

ERROR: Cannot find the specified cross file: /home/sem/sdk/romulus-14.06.22/sysroots/x86_64-oesdk-linux/usr/share/meson/arm-openbmc-linux-gnueabi-meson.cross

ly8427, 2022-11-25:
Hi , I don't know if this issue is fixed or not, I have met similar issue when building phosphor-dbus-interfaces, and I found I need to generate cross file for this cross build. I referred to https://mesonbuild.com/Cross-compilation.html#cross-compilation for the cross file.
my build machine : Ubuntu 20.04.5 LTS (GNU/Linux 5.10.16.3-microsoft-standard-WSL2 **x86_64**)
my host machine:  **arm** 
I create the cross file  **/home/liu/sdk/romulus-2022-11-22/sysroots/x86_64-oesdk-linux/usr/share/meson/arm-openbmc-linux-gnueabi-meson.cross** as below

[binaries]
c = 'arm-openbmc-linux-gnueabi-gcc'
cpp = 'arm-openbmc-linux-gnueabi-g++'
ar = 'arm-openbmc-linux-gnueabi-ar'
nm = 'arm-openbmc-linux-gnueabi-nm'
strip = 'arm-openbmc-linux-gnueabi-strip'
pkgconfig = 'pkg-config'

[built-in options]
c_args = '-O2 -pipe -g -feliminate-unused-debug-types'
c_link_args = '-Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -Wl,-z,relro,-z,now -L/home/liu/sdk/romulus-2022-11-22/sysroots/arm1176jzs-openbmc-linux-gnueabi/usr/lib'
cpp_args = '-marm -mcpu=arm1176jz-s -fstack-protector-strong -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/liu/sdk/romulus-2022-11-22/sysroots/arm1176jzs-openbmc-linux-gnueabi'
cpp_link_args = '--sysroot=/home/liu/sdk/romulus-2022-11-22/sysroots/arm1176jzs-openbmc-linux-gnueabi -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -Wl,-z,relro,-z,now'

[properties]
needs_exe_wrapper = true
sys_root = '/home/liu/sdk/romulus-2022-11-22/sysroots/arm1176jzs-openbmc-linux-gnueabi'

[host_machine]
system = 'linux'
cpu_family = 'arm'
cpu = 'arm11'
endian = 'little'

[build_machine]
system = 'linux'
cpu_family = 'x86_64'
cpu = 'x86_64'
endian = 'little'

cd phosphor-dbus-interfaces , type "meson setup --wipe build" then "Ninja -C build", the build succeed.

closed 2023-09-25 by williamspatrick