openbmc/openbmc #3865: Build Error : telemetry_git.bb:do_compile

shubhabrata1979, 2022-06-09:
I am new to  OpenBMC & Yocto bitbake and am getting this Error "telemetry_git.bb:do_compile" while building "obmc-phosphor-image"  on "Ubuntu 22.04" installed with  "ninja 1.10.1". 

Build Configuration:
BB_VERSION           = "2.0.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-22.04"
TARGET_SYS           = "arm-openbmc-linux-gnueabi"
MACHINE              = "romulus"
DISTRO               = "openbmc-openpower"
DISTRO_VERSION       = "nodistro.0"
TUNE_FEATURES        = "arm thumb arm1176jzs"
TARGET_FPU           = "soft"
meta                 
meta-oe              
meta-networking      
meta-perl            
meta-python          
meta-phosphor        
meta-aspeed          
meta-openpower       
meta-romulus         = "master:ccabc569f5899abf3878d8816e78db20faacc5c8"

| /usr/src/debug/glibc/2.35-r0/git/csu/../sysdeps/arm/start.S:118: undefined reference to `main'
| collect2: error: ld returned 1 exit status
| ninja: build stopped: subcommand failed.
| WARNING: exit code 1 from a shell command.
ERROR: 
/telemetry_git.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 3433 tasks of which 3424 didn't need to be rerun and 1 failed.

Appreciate, if I get some help !!
Thanks in Advance,

williamspatrick, 2022-06-09:
> ... on "Ubuntu 22.04" installed with "ninja 1.10.1".

The version in Ninja on your host machine shouldn't really matter when you're doing bitbake.  It should use its own Ninja install.

> meta-romulus = "master:ccabc569f5899abf3878d8816e78db20faacc5c8"

This looks like you're running a pretty recent version of our repo (good), so it _shouldn't_ have any issues like this...

```
| /usr/src/debug/glibc/2.35-r0/git/csu/../sysdeps/arm/start.S:118: undefined reference to `main'
/telemetry_git.bb:do_compile) failed with exit code '1'
```

This is fairly unusual.  There is certainly a `main` defined in this repository.

Can you look at the `tmp/work/arm.../telemetry/.../temp/log.do_compile` file to see if there is additional details on why it is failing?  Maybe upload this to gist.

shubhabrata1979, 2023-01-02:
- I have tried on a different host (its a Virtual Box installed with ubuntu version is 20.04) and in that this Telemetry build is NOT resulting in any errors. When I upgraded to "22.04" and again started the build there are NO errors again.
- Found that the host on which this issue was tested previously was having an older CPU with a limited RAM size of 4G and was quiet slow in processing speed so it missed downloading many libraries and other utils , this could be one of the reasons of those errors.
- But now I am struck on another issue i.e. on Ubuntu 22.04, I am getting the following errors : 

"Failed to guess install tag for /lib/systemd/system/bmcweb.socket
Failed to guess install tag for /lib/systemd/system/bmcweb.service
Failed to guess install tag for /etc/pam.d/webserver
Failed to guess install tag for /usr/share/www/dummy"

This build is consuming all the 100% of the disk space (could be the reason of the above errors), my virtual box is installed on a 100GB logical disk partition, so now I am trying to get a larger capacity disk. 

But meanwhile is their anything else I should also suspect for fixing these latest errors ??

shubhabrata1979, 2023-01-31:
Any Comments on my last update ??

closed 2023-01-31 by williamspatrick