openbmc/openbmc #3867: Build error with glibc recipe

edineshkumar, 2022-07-01:
I have cloned and started the new build today and faced the error with glibc recipe in do_fetch / do_unpack function.

**Build Configuration Details:**

Build Configuration:
BB_VERSION           = "2.0.1"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-20.04"
TARGET_SYS           = "arm-openbmc-linux-gnueabi"
MACHINE              = "evb-ast2600"
DISTRO               = "openbmc-phosphor"
DISTRO_VERSION       = "nodistro.0"
TUNE_FEATURES        = "arm armv7a vfp vfpv4d16 callconvention-hard"
TARGET_FPU           = "hard"
meta
meta-poky
meta-oe
meta-networking
meta-python
meta-phosphor
meta-aspeed
meta-evb-ast2600     = "master:0ee49ef1a8bf8cf874b32c1897a49fcdb2faee30"

**Note on Bugs**

`DEBUG: Executing python function do_unpack
DEBUG: Executing python function base_do_unpack
DEBUG: Running 'export PSEUDO_DISABLED=1; export DBUS_SESSION_BUS_ADDRESS="unix:path=/run/user/0/bus"; `

`ERROR: Bitbake Fetcher Error: FetchError('Fetch command export PSEUDO_DISABLED=1; export DBUS_SESSION_BUS_ADDRESS="unix:path=/run/user/0/bus";`

**To Reproduce**
Steps to reproduce the behavior:
1. Cloned Latest source 'git clone https://github.com/openbmc/openbmc'
2. Specified the target machine
3. Executed bitbake obmc-phosphor-image
4. See error on glibc 

**Screenshots**

[image]

[image]

Appreciate, if I get some help..!
Thanks in Advance.

williamspatrick, 2022-07-01:
So the suggestion is that a fresh clone is broken for evb-ast2600?  @geissonator, can you glance at this?  I’m not in a place I can easily recreate for a bit.

geissonator, 2022-07-01:
Built fine for me:
```
Build Configuration:
BB_VERSION           = "2.0.1"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-20.04"
TARGET_SYS           = "arm-openbmc-linux-gnueabi"
MACHINE              = "evb-ast2600"
DISTRO               = "openbmc-phosphor"
DISTRO_VERSION       = "nodistro.0"
TUNE_FEATURES        = "arm armv7a vfp vfpv4d16 callconvention-hard"
TARGET_FPU           = "hard"
meta                 
meta-poky            
meta-oe              
meta-networking      
meta-python          
meta-phosphor        
meta-aspeed          
meta-evb-ast2600     = "master:06be640dc87caf845170f209ff5305f9911eca1a"

Initialising tasks: 100% |#############################################################################################################################| Time: 0:00:01
Checking sstate mirror object availability: 100% |#####################################################################################################| Time: 0:00:33
Sstate summary: Wanted 1640 Local 0 Mirrors 0 Missed 1640 Current 0 (0% match, 0% complete)
NOTE: Executing Tasks
Using an insecure image signing key!
NOTE: Tasks Summary: Attempted 4059 tasks of which 0 didn't need to be rerun and all succeeded.

Summary: There were 2 WARNING messages.
```

dhineshkumar256, 2022-07-12:
Adding `--no-same-owner` in the `tar -xzf` command fixed this error for me.

williamspatrick, 2022-07-13:
> 

Which tar command is this referring to?  If this is something needed, should it be perused upstream with Yocto to get it added?

closed 2023-09-25 by williamspatrick