openbmc/openbmc #3873: how to run secure image on qemu

hitesh0803, 2022-08-16:
i have two binary image (.mtd,otp-all.image).
my problem is :-
1) how to pass otp config binary  to qemu to test bmc secure boot.

amboar, 2022-08-16:
Qemu doesn't yet have a model for the secure boot and otp controllers and do does no verification of the image on the boot path. Instead, verification of the u-boot SPL against the otp image is done by the socsec tooling that generates the otp image as part of the build process. If the build didn't fail and produced an otp-all.image then the corresponding SPL image has a signature that's valid with at least one key in otp-all.image

closed 2022-08-16 by amboar

hitesh0803, 2022-08-17:
Thankyou Amboar.

we have one server board with ast2600 soc .
problem is :-
1) Is any way to program ast2600 otp memory ? 
2) how to program (otp-all.image) into otp memory ?

amboar, 2022-08-17:
You need to consult Aspeed and their documentation on that. For example, see section `9.3 OTP Utility` in https://github.com/AspeedTech-BMC/openbmc/releases/download/v08.02/SDK_User_Guide_v08.02.pdf

hitesh0803, 2022-08-17:
Thankyou Amboar

hitesh0803, 2022-08-25:
aspeed provide Secureboot_spec.pdf document for otp memory . if you have please share.

williamspatrick, 2022-08-25:
> aspeed provide Secureboot_spec.pdf document for otp memory . if you have please share.

Unless you find a spec on the Aspeed github, it is most likely only available from them under NDA.  No one should be providing links to those (or attachments) on our github issues.

hitesh0803, 2022-08-25:
Thankyou williamspatrick