openbmc/openbmc #3874: usb interface mac address mismatch issue

gayathril2703, 2022-08-16:
I had added below configs to bring up the Host NIC interface.

CONFIG_USB_U_ETHER=y
CONFIG_USB_F_ECM=y
CONFIG_USB_F_SUBSET=y
CONFIG_USB_F_RNDIS=y
CONFIG_USB_ETH=y
CONFIG_USB_ETH_RNDIS=y

Now, usb0 interface is deducted. 
But the MAC address in dmesg logs and MAC address in usb0 interface differs. They do not match.

In ip a command output, usb0 has different mac address, and 
In dmesg logs, usb0 shows different mac address.

Please suggest, how can we fix this mac mismatch.

Thanks.

Pravee207, 2023-01-03:
Please explain the reproduction steps. Did you used any specific hardware for this test?

shubhabrata1979, 2023-01-03:
Few questions:
1) Is it specific to a particular Linux Distro like Ubuntu 20/22/18 or can it be reproducible on all ?
2)  What was the kernel version when the issue occurred first time ?
3) Any specific choice hardware board that we need to try or Qemu is enough ??
Thanks,

Pravee207, 2023-02-14:
Please help me with the reproduction steps. Is any hardware is required to perform this test?.
Thanks

mdmillerii, 2023-02-14:
Also remember there are two Mac addresses one for the BMC nic and one for the host.

thangqn-ampere, 2023-02-15:
@gayathril2703 I wonder if you create USB Ethernet with fixed MAC Address. If not, random MAC address is expected.

Refer to https://github.com/openbmc/openbmc/blob/9437adff200218f5833bc9cf30fa1484c5a77699/meta-ampere/meta-common/recipes-ampere/network/ampere-usbnet/ampere_usbnet.sh#L11 for how we configured fixed MAC Address

Pravee207, 2023-03-01:
Can you please help me with reproduction steps? Without knowing them, I'm unable to move forward. Can it be done on qemu or any specific hardware required?.

Thanks.

williamspatrick, 2023-09-25:
No response from originator.

closed 2023-09-25 by williamspatrick