openbmc/openbmc #3876: wrong Image Format for bootm command ERROR: can't get kernel image!

mahendra-space, 2022-08-25:
Hi,
I have installed QEMU on my Linux PC and try to start QEMU session with romulus image "obmc-phosphor-image-romulus.static.mtd" by using command-
]./qemu-system-arm -m 256 -M romulus-bmc -nographic \
    -drive file=./obmc-phosphor-image-romulus.static.mtd,format=raw,if=mtd \
    -net nic \
    -net user,hostfwd=:127.0.0.1:2222-:22,hostfwd=:127.0.0.1:2443-:443,hostfwd=udp:127.0.0.1:2623-:623,hostname=qemu"

But it is not started- 
Issue- wrong Image Format for bootm command 
ERROR: can't get kernel image!

any solution please ??
I am following this web link "https://github.com/openbmc/docs/blob/master/development/dev-environment.md"

williamspatrick, 2022-08-25:
Where did your `obmc-phosphor-image-romulus.static.mtd` come from?  Did you build it or download it?

https://github.com/openbmc/docs/blob/master/development/dev-environment.md#download-and-start-qemu-session

The instructions here give a `wget` call that downloads an image from Jenkins.  Can you confirm this works?  If it does, and you're attempting to run an image you built yourself, can you give more details on how you built it and what your `tmp/deploy/images` tree looks like after you built.

mahendra-space, 2022-08-25:
> Where did your `obmc-phosphor-image-romulus.static.mtd` come from? Did you build it or download it? [...]

Thanks for prompt reply Williamspatrick.

Actually I downloaded from "wget https://jenkins.openbmc.org/job/latest-master/label=docker-builder,target=romulus/lastSuccessfulBuild/artifact/openbmc/build/tmp/deploy/images/romulus/obmc-phosphor-image-romulus.static.mtd".

I followed instruction as given into this link "https://github.com/openbmc/docs/blob/master/development/dev-environment.md"
for downloading and installing and start QEMU session.

 

> Where did your `obmc-phosphor-image-romulus.static.mtd` come from? Did you build it or download it? [...]

mahendra-space, 2022-08-25:
Hi Williamspatrick,

I have also build openBMC source code by using bitbake. 
QEMU session start and only once. But when I reboot my Linux pc and try to start QEMU, It got error-

:~/sdk$ ./qemu-system-arm -m 256 -M romulus-bmc -nographic     -drive file=./obmc-phosphor-image-romulus.static.mtd,format=raw,if=mtd     -net nic     -net user,hostfwd=:127.0.0.1:2222-:22,hostfwd=:127.0.0.1:2443-:443,hostfwd=udp:127.0.0.1:2623-:623,hostname=qemu
qemu-system-arm: warning: Aspeed iBT has no chardev backend
qemu-system-arm: warning: nic ftgmac100.1 has no peer

U-Boot 2019.04 (Aug 09 2022 - 05:12:55 +0000)

SOC : AST2500-A1 
RST : Power On 
2nd Boot : Enable
LPC Mode : SIO:Enable : SuperIO-2e
Eth : MAC0: RMII/NCSI, , MAC1: RMII/NCSI, 
Model: Romulus BMC
DRAM:  192 MiB (capacity:256 MiB, VGA:64 MiB, ECC:off)
MMC:   
Loading Environment from SPI Flash... SF: Detected n25q256a with page size 256 Bytes, erase size 4 KiB, total 32 MiB
OK
In:    serial@1e784000
Out:   serial@1e784000
Err:   serial@1e784000
Net:   ftgmac100_probe - NCSI detected

Warning: ethernet@1e660000 (eth0) using random MAC address - 36:62:a6:cb:88:a4
eth0: ethernet@1e660000
Hit any key to stop autoboot:  0 
Wrong Image Format for bootm command
ERROR: can't get kernel image!
ast#

williamspatrick, 2022-08-25:
Please confirm the wget is working and you are ending up with a 32 MB file. I ran this myself and it seems to work fine. I suspect your download is unsuccessful.

williamspatrick, 2022-08-25:
> ~/sdk$ ./qemu-system-arm -m 256 -M romulus-bmc -nographic -drive file=./obmc-phosphor-image-romulus.static.mtd,format=raw,if=mtd

You mentioned that you once built with bitbake and it once worked, but then you gave an invocation as example like above. This is using “./“ for the image path, which is your current directory. Since the current directory is “SDK”, this is not the location of the bitbake image. I suggest changing the file parameter to point at your bitbake build tree. It is going to have something like “tmp/deploy/images” in the full path to the image. 

Going back and forth further on GitHub is not likely to be productive. You would probably have more success asking in our Discord server for assistance so we can do more real-time investigation.

mahendra-space, 2022-08-25:
Thanks Patrick Williams,

Actually, I am using only download image from "wget https://jenkins.openbmc.org/job/latest-master/label=docker-builder,target=romulus/lastSuccessfulBuild/artifact/openbmc/build/tmp/deploy/images/romulus/obmc-phosphor-image-romulus.static.mtd" which is in my "sdk" path only . I mean current directory.

Thanks,

cometzero, 2023-06-16:
Hi, @mahendra-space and @williamspatrick,

Did you resolve this issue? Actually, I'm facing same issue on my envionment.
I followed the guide[1]. First booting was successful but second boot was same with @mahendra-space 's case.
I feel like some services or programs on the first booting will overwrite the kernel region.

```
qemu-system-arm: warning: Aspeed iBT has no chardev backend
qemu-system-arm: warning: nic ftgmac100.1 has no peer

U-Boot 2019.04 (May 30 2023 - 06:05:47 +0000)

SOC : AST2500-A1 
RST : Power On 
2nd Boot : Enable
LPC Mode : SIO:Enable : SuperIO-2e
Eth : MAC0: RMII/NCSI, , MAC1: RMII/NCSI, 
Model: Romulus BMC
DRAM:  192 MiB (capacity:256 MiB, VGA:64 MiB, ECC:off)
MMC:   
Loading Environment from SPI Flash... SF: Detected n25q256a with page size 256 Bytes, erase size 4 KiB, total 32 MiB
OK
In:    serial@1e784000
Out:   serial@1e784000
Err:   serial@1e784000
Net:   ftgmac100_probe - NCSI detected

Warning: ethernet@1e660000 (eth0) using random MAC address - de:16:60:26:de:1c
eth0: ethernet@1e660000
Hit any key to stop autoboot:  0 
Wrong Image Format for bootm command
ERROR: can't get kernel image!
```

[1]: https://github.com/openbmc/docs/blob/master/development/dev-environment.md#download-and-start-qemu-session

amboar, 2023-06-16:
Can you please paste the output of `printenv` at the u-boot prompt (stop the autoboot process by hitting a key)?

cometzero, 2023-06-16:
hi @amboar , here is my result.

```
U-Boot 2019.04 (May 30 2023 - 06:05:47 +0000)

SOC : AST2500-A1 
RST : Power On 
2nd Boot : Enable
LPC Mode : SIO:Enable : SuperIO-2e
Eth : MAC0: RMII/NCSI, , MAC1: RMII/NCSI, 
Model: Romulus BMC
DRAM:  192 MiB (capacity:256 MiB, VGA:64 MiB, ECC:off)
MMC:   
Loading Environment from SPI Flash... SF: Detected n25q256a with page size 256 Bytes, erase size 4 KiB, total 32 MiB
OK
In:    serial@1e784000
Out:   serial@1e784000
Err:   serial@1e784000
Net:   ftgmac100_probe - NCSI detected

Warning: ethernet@1e660000 (eth0) using random MAC address - 02:ef:31:36:51:ea
eth0: ethernet@1e660000
Hit any key to stop autoboot:  0 
ast# printenv
baudrate=115200
bootargs=console=ttyS4,115200n8 root=/dev/ram rw
bootcmd=bootm 20100000
bootdelay=2
bootfile=all.bin
fdtcontroladdr=89f941d8
gatewayip=192.168.0.1
ipaddr=192.168.0.45
netmask=255.255.255.0
serverip=192.168.0.81
stderr=serial@1e784000
stdin=serial@1e784000
stdout=serial@1e784000
trace_args=trace_events=fsi_slave_init,fsi_slave_invalid_cfam,fsi_dev_init,
verify=yes

Environment size: 383/131068 bytes
ast# boot
Wrong Image Format for bootm command
ERROR: can't get kernel image!
ast#
```

cometzero, 2023-06-16:
Once I revert the environment variables using below command, I was able to boot well.
```
env default -f -a
```

amboar, 2023-06-16:
What does `printenv bootcmd` say after you run `env default -f -a ; saveenv` ?

cometzero, 2023-06-16:
> What does `printenv bootcmd` say after you run `env default -f -a ; saveenv` ?

Same as before. All other variables are same as well.
```
ast# printenv bootcmd
bootcmd=bootm 20080000
```

amboar, 2023-06-16:
It's not the same as before. In your `printenv` output above where the boot failed, you have:

```
bootcmd=bootm 20100000
```

cometzero, 2023-06-16:
Oh. The address has been changed from 20080000 to 20100000 from the second boot :(

amboar, 2023-06-16:
Yes, `20100000` is wrong, `20080000` is correct. We need to understand where the `20100000` value is coming from

cometzero, 2023-06-16:
While booting, the environment variable seemed to be roll-backed...
```
Jun 16 04:00:21 romulus systemd[1]: Migrate dropbear keys from /var/lib to /etc was skipped because of an unmet condition check (ConditionPathExists=/var/lib/dropbear/dropbear_rsa_host_key).
Jun 16 04:00:21 romulus systemd[1]: Starting SSH Key Generation...
Jun 16 04:00:22 romulus systemd[1]: Starting Name Service Cache Daemon...
**Jun 16 04:00:22 romulus fw_setenv[183]: Warning: Bad CRC, using default environment**
Jun 16 04:00:22 romulus systemd[1]: Starting LDAP daemon...
Jun 16 04:00:22 romulus systemd[1]: ttyS0 Console Server was skipped because of an unmet condition check (ConditionPathExists=/etc/obmc-console/server.ttyS0.conf).
Jun 16 04:00:22 romulus systemd[1]: Started ttyVUART0 Console Server.
```

amboar, 2023-06-16:
That suggests that `/etc/fw_env.config` might be misconfigured? You'll need to check that it lines up with where u-boot expects the environment to live in the flash, after accounting for the MTD partition offsets.

cometzero, 2023-06-16:
The first booting already reported the CRC of the environment variables.
```
DRAM:  192 MiB (capacity:256 MiB, VGA:64 MiB, ECC:off)
MMC:   
Loading Environment from SPI Flash... SF: Detected n25q256a with page size 256 Bytes, erase size 4 KiB, total 32 MiB
*** Warning - bad CRC, using default environment
```

`/etc/fw_env.config` file indicates that the offset is `0x0000` and its size is `0x20000`
```
root@romulus:~# cat /etc/fw_env.config 
# MTD device name	Device offset	Env. size	Flash sector size	Number of sectors
/dev/mtd/u-boot-env	0x0000		0x20000		0x1000
```
However, u-boot is trying to find the environment from `0x60000`
```
CONFIG_BOOTCOMMAND="bootm 20080000"
CONFIG_ENV_SIZE=0x20000            
CONFIG_ENV_OFFSET=0x60000          
```

amboar, 2023-06-16:
Yep, so we need to look at the kernel devicetree:

https://github.com/openbmc/linux/blob/dev-6.1/arch/arm/boot/dts/openbmc-flash-layout.dtsi#L14

So that looks like it all aligns. Is the default environment built into `fw_setenv` different from the default environment built into the u-boot binary?

cometzero, 2023-06-16:
I'm looking the process of the flash image generation. When I checked the flash image, it was empty.
```
ast# sf read 0x80000000 0x60000 0x20000
device 0 offset 0x60000, size 0x20000
SF: 131072 bytes @ 0x60000 Read: OK
ast# md 0x80000000
80000000: ffffffff ffffffff ffffffff ffffffff    ................
80000010: ffffffff ffffffff ffffffff ffffffff    ................
80000020: ffffffff ffffffff ffffffff ffffffff    ................
80000030: ffffffff ffffffff ffffffff ffffffff    ................
80000040: ffffffff ffffffff ffffffff ffffffff    ................
80000050: ffffffff ffffffff ffffffff ffffffff    ................
```

Below is the result after execution of `env default -a -f`.
```
ast# env default -a -f; saveenv
## Resetting to default environment
Saving Environment to SPI Flash... Erasing SPI flash...Writing to SPI flash...done
OK
ast# sf read 0x80000000 0x60000 0x20000
device 0 offset 0x60000, size 0x20000
SF: 131072 bytes @ 0x60000 Read: OK
ast# md 0x80000000                     
80000000: d8e1e586 64756162 65746172 3531313d    ....baudrate=115
80000010: 00303032 746f6f62 73677261 6e6f633d    200.bootargs=con
80000020: 656c6f73 7974743d 312c3453 30323531    sole=ttyS4,11520
80000030: 20386e30 746f6f72 65642f3d 61722f76    0n8 root=/dev/ra
80000040: 7772206d 6f6f6200 646d6374 6f6f623d    m rw.bootcmd=boo
80000050: 32206d74 30383030 00303030 746f6f62    tm 20080000.boot
80000060: 616c6564 00323d79 746f6f62 656c6966    delay=2.bootfile
80000070: 6c6c613d 6e69622e 74616700 79617765    =all.bin.gateway
```

williamspatrick, 2023-06-16:
Do you expect different?

mdmillerii, 2023-06-19:
The real issue is your bootcmd doesn't do anything to load the image into ram before trying to transfer control to it.

Usually this would be part of the u-boot customization.   Most of our image recipes don't try to pre-populate the u-boot environment in the image, but instead rely on the default compiled into u-boot.

cometzero, 2023-06-19:
I've uploaded a patch to fix this issue. I verified the second booting was successful without any errors.

https://gerrit.openbmc.org/c/openbmc/openbmc/+/64287 meta-phosphor: aspeed-layer: correct u-boot fw-utils mismatch

closed 2023-06-20 by geissonator