openbmc/openbmc #3881: nscd monitor file path is incorrect

wayne-tung, 2022-09-07:
Hi all,
I used openbmc tag **2.12.0-rc1** to build my project, and found that there are some error messages about nscd.
**This service is okay when my code base is 2.11.0.**
I traced some nscd source codes from glibc and cannot figure out why nscd sets the traced file path to /passwd (expected path is /etc/passwd)
Did I miss some configuration? or it is a bug on glibc.

**Unexpected behavior you saw**
Jan 01 00:00:47 ck-enterprise nscd[219]: 219 disabled inotify-based monitoring for file `/passwd': No such file or directory
Jan 01 00:00:47 ck-enterprise nscd[219]: 219 stat failed for file `/passwd'; will try again later: No such file or directory
Jan 01 00:00:47 ck-enterprise nscd[219]: 219 disabled inotify-based monitoring for file `/group': No such file or directory
Jan 01 00:00:47 ck-enterprise nscd[219]: 219 stat failed for file `/group'; will try again later: No such file or directory
Jan 01 00:00:47 ck-enterprise nscd[219]: 219 disabled inotify-based monitoring for file `/hosts': No such file or directory
Jan 01 00:00:47 ck-enterprise nscd[219]: 219 stat failed for file `/hosts'; will try again later: No such file or directory
Jan 01 00:00:47 ck-enterprise nscd[219]: 219 disabled inotify-based monitoring for file `/resolv.conf': No such file or directory
Jan 01 00:00:47 ck-enterprise nscd[219]: 219 stat failed for file `/resolv.conf'; will try again later: No such file or directory
Jan 01 00:00:47 ck-enterprise nscd[219]: 219 disabled inotify-based monitoring for file `/services': No such file or directory
Jan 01 00:00:47 ck-enterprise nscd[219]: 219 stat failed for file `/services'; will try again later: No such file or directory
Jan 01 00:00:47 ck-enterprise nscd[219]: 219 disabled inotify-based monitoring for file `/nsswitch.conf': No such file or directory
Jan 01 00:00:47 ck-enterprise nscd[219]: 219 stat failed for file `/nsswitch.conf'; will try again later: No such file or directory
Jan 01 00:00:47 ck-enterprise nscd[219]: 219 disabled inotify-based monitoring for file `/nsswitch.conf': No such file or directory
Jan 01 00:00:47 ck-enterprise nscd[219]: 219 stat failed for file `/nsswitch.conf'; will try again later: No such file or directory
Jan 01 00:00:47 ck-enterprise nscd[219]: 219 disabled inotify-based monitoring for file `/nsswitch.conf': No such file or directory
Jan 01 00:00:47 ck-enterprise nscd[219]: 219 stat failed for file `/nsswitch.conf'; will try again later: No such file or directory
Jan 01 00:00:47 ck-enterprise nscd[219]: 219 disabled inotify-based monitoring for file `/nsswitch.conf': No such file or directory
Jan 01 00:00:47 ck-enterprise nscd[219]: 219 stat failed for file `/nsswitch.conf'; will try again later: No such file or directory
Jan 01 00:00:47 ck-enterprise nscd[219]: 219 cannot write to database file /var/db/nscd/passwd: Invalid argument
Jan 01 00:00:48 ck-enterprise nscd[219]: 219 cannot write to database file /var/db/nscd/group: Invalid argument
Jan 01 00:00:48 ck-enterprise nscd[219]: 219 cannot write to database file /var/db/nscd/hosts: Invalid argument
Jan 01 00:00:49 ck-enterprise nscd[219]: 219 cannot write to database file /var/db/nscd/services: Invalid argument
Jan 01 00:01:04 ck-enterprise nscd[219]: 219 checking for monitored file `/nsswitch.conf': No such file or directory
Jan 01 00:01:04 ck-enterprise nscd[219]: 219 checking for monitored file `/passwd': No such file or directory
Jan 01 00:01:05 ck-enterprise nscd[219]: 219 checking for monitored file `/nsswitch.conf': No such file or directory
Jan 01 00:01:05 ck-enterprise nscd[219]: 219 checking for monitored file `/group': No such file or directory
Jan 01 00:01:06 ck-enterprise nscd[219]: 219 checking for monitored file `/nsswitch.conf': No such file or directory
Jan 01 00:01:06 ck-enterprise nscd[219]: 219 checking for monitored file `/resolv.conf': No such file or directory
Jan 01 00:01:06 ck-enterprise nscd[219]: 219 checking for monitored file `/hosts': No such file or directory
Jan 01 00:01:07 ck-enterprise nscd[219]: 219 checking for monitored file `/nsswitch.conf': No such file or directory
Jan 01 00:01:07 ck-enterprise nscd[219]: 219 checking for monitored file `/services': No such file or directory

**Expected behavior**
Sep 07 07:31:19 CK-Enterprise-EF0 nscd[195]: 195 monitoring file `/etc/passwd` (1)
Sep 07 07:31:19 CK-Enterprise-EF0 nscd[195]: 195 monitoring directory `/etc` (2)
Sep 07 07:31:19 CK-Enterprise-EF0 nscd[195]: 195 monitoring file `/etc/group` (3)
Sep 07 07:31:19 CK-Enterprise-EF0 nscd[195]: 195 monitoring directory `/etc` (2)
Sep 07 07:31:19 CK-Enterprise-EF0 nscd[195]: 195 monitoring file `/etc/hosts` (4)
Sep 07 07:31:19 CK-Enterprise-EF0 nscd[195]: 195 monitoring directory `/etc` (2)
Sep 07 07:31:19 CK-Enterprise-EF0 nscd[195]: 195 disabled inotify-based monitoring for file `/etc/resolv.conf': No such file or directory
Sep 07 07:31:19 CK-Enterprise-EF0 nscd[195]: 195 stat failed for file `/etc/resolv.conf'; will try again later: No such file or directory
Sep 07 07:31:19 CK-Enterprise-EF0 nscd[195]: 195 monitoring file `/etc/services` (5)
Sep 07 07:31:19 CK-Enterprise-EF0 nscd[195]: 195 monitoring directory `/etc` (2)
Sep 07 07:31:19 CK-Enterprise-EF0 nscd[195]: 195 monitoring file `/etc/nsswitch.conf` (6)
Sep 07 07:31:19 CK-Enterprise-EF0 nscd[195]: 195 monitoring directory `/etc` (2)
Sep 07 07:31:19 CK-Enterprise-EF0 nscd[195]: 195 monitoring file `/etc/nsswitch.conf` (6)
Sep 07 07:31:19 CK-Enterprise-EF0 nscd[195]: 195 monitoring directory `/etc` (2)
Sep 07 07:31:19 CK-Enterprise-EF0 nscd[195]: 195 monitoring file `/etc/nsswitch.conf` (6)
Sep 07 07:31:19 CK-Enterprise-EF0 nscd[195]: 195 monitoring directory `/etc` (2)
Sep 07 07:31:19 CK-Enterprise-EF0 nscd[195]: 195 monitoring file `/etc/nsswitch.conf` (6)
Sep 07 07:31:19 CK-Enterprise-EF0 nscd[195]: 195 monitoring directory `/etc` (2)

**To Reproduce**
Every time nscd is running.
and "checking for monitored file" will keep printing.

**OpenBMC Information:**
TAG: 2.12.0-rc1

edtanous, 2022-10-12:
Which platform are you working on?  I have not seen this error on the S7106 platform that I use, so presumably what you're seeing is platform-specific.

closed 2022-10-12 by wayne-tung

zevweiss, 2022-10-12:
I ran into the same issue a while back and tracked it to the same underlying cause (https://discord.com/channels/775381525260664832/867820390406422538/966827096527765547) -- @wayne-tung, has it been resolved somehow?

wayne-tung, 2022-10-12:
@zevweiss 
Currently, I used 32 bits time_t for ncsd,
To make it use 32 bits time_t I modified glibc Makeconfig.
Everything works well now, but I'm not sure whether using 32 bits time_t on ncsd causes some side effects.

zevweiss, 2022-10-12:
> To make it use 32 bits time_t I modified glibc Makeconfig.

Any chance you could share the patch you used?

> Everything works well now, but I'm not sure whether using 32 bits time_t on ncsd causes some side effects.
>
> I think it is better to make all time_t to be 64 bits in the future.

Yeah, there are certainly backwards-compatibility complexities there; I don't think I know enough to judge whether a given potential fix is really "right".

wayne-tung, 2022-10-13:
@zevweiss

wayne-tung, 2022-10-13:
@zevweiss 
`From 0fda9faf757abd4f5469e6d9207499e97f79a663 Mon Sep 17 00:00:00 2001
From: Wayne Tung <wayne.tung@ui.com>
Date: Thu, 13 Oct 2022 13:10:21 +0800
Subject: [PATCH] Use 32 bits time_t for ncsd

---
 Makeconfig | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

diff --git a/Makeconfig b/Makeconfig
index 47db08d6ae..f78f7cc74a 100644
--- a/Makeconfig
+++ b/Makeconfig
@@ -869,7 +869,7 @@ endif
 +extra-math-flags = $(if $(filter libm,$(in-module)),-fno-math-errno,-fmath-errno)

 # Use 64 bit time_t support for installed programs
-installed-modules = nonlib nscd lddlibc4 ldconfig locale_programs \
+installed-modules = nonlib lddlibc4 ldconfig locale_programs \
            iconvprogs libnss_files libnss_compat libnss_db libnss_hesiod \
            libutil libpcprofile libSegFault
 +extra-time-flags = $(if $(filter $(installed-modules),\
--
`

wayne-tung, 2022-10-13:
From 0fda9faf757abd4f5469e6d9207499e97f79a663 Mon Sep 17 00:00:00 2001
From: Wayne Tung <wayne.tung@ui.com>
Date: Thu, 13 Oct 2022 13:10:21 +0800
Subject: [PATCH] Use 32 bits time_t for ncsd

---
 Makeconfig | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

diff --git a/Makeconfig b/Makeconfig
index 47db08d6ae..f78f7cc74a 100644
--- a/Makeconfig
+++ b/Makeconfig
@@ -869,7 +869,7 @@ endif
 +extra-math-flags = $(if $(filter libm,$(in-module)),-fno-math-errno,-fmath-errno)

 # Use 64 bit time_t support for installed programs
-installed-modules = nonlib nscd lddlibc4 ldconfig locale_programs \
+installed-modules = nonlib lddlibc4 ldconfig locale_programs \
            iconvprogs libnss_files libnss_compat libnss_db libnss_hesiod \
            libutil libpcprofile libSegFault
 +extra-time-flags = $(if $(filter $(installed-modules),\
--

wayne-tung, 2022-10-13:
@zevweiss 
https://pastecode.io/s/xpcgtbrz
Just remove ncsd from 64 bits time_t list.

zevweiss, 2022-10-13:
Neat, thanks @wayne-tung.  Have you posted that to the [glibc mailing list](https://sourceware.org/mailman/listinfo/libc-alpha)?  (I couldn't find anything in a quick search of the archives.)  I'm not sure if it's necessarily the correct fix in all cases, but even if it's not it might help spur the maintainers into helping fix it.  (CC-ing the [OpenBMC list](https://lists.ozlabs.org/listinfo/openbmc) on the thread would be helpful too.)

wayne-tung, 2022-10-13:
No, I haven't posted it.
Because I think this is not a good way to fix this issue.
But you're right, Could you help post this to glibc mailing list? Thanks.

zevweiss, 2022-12-09:
[Upstream bug](https://sourceware.org/bugzilla/show_bug.cgi?id=29402) now marked resolved, upcoming 2.37 release should include the fix.  Since that isn't scheduled to come out until circa February, I might look into backporting the patches into poky upstream so we can get them a bit sooner...