openbmc/openbmc #3884: Error in bitbake kernel for Raspberry PI 3-64

zhangyu827, 2022-09-19:
Hi all
I'm trying to build OpenBMC image for Raspberry PI 3-64 on Ubuntu 18.04 / 64 bits machine.

I've followed these steps:
1. cd openbmc
2. export TEMPLATECONF=meta-evb/meta-evb-raspberrypi/conf
3. . openbmc-env
4. open the conf/local.conf file in the editor and change the machine to raspberrypi3-64
   MACHINE ??= "raspberrypi3-64"
5. bitbake obmc-phosphor-image

The following error appears:
zhangyu@zhangyu-virtual-machine:~/openbmc/build$ bitbake obmc-phosphor-image
Loading cache: 100% |############################################| Time: 0:00:02
Loaded 4057 entries from dependency cache.
Parsing recipes: 100% |##########################################| Time: 0:00:01
Parsing of 2655 .bb files complete (2654 cached, 1 parsed). 4058 targets, 364 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies
ERROR: Nothing RPROVIDES 'virtual-obmc-system-mgmt' (but /home/zhangyu/openbmc/meta-phosphor/recipes-phosphor/images/obmc-phosphor-image.bb RDEPENDS on or otherwise requires it)
NOTE: Runtime target 'virtual-obmc-system-mgmt' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['virtual-obmc-system-mgmt']
ERROR: Required build target 'obmc-phosphor-image' has no buildable providers.
Missing or unbuildable dependency chain was: ['obmc-phosphor-image', 'virtual-obmc-system-mgmt']

Summary: There were 2 ERROR messages shown, returning a non-zero exit code.

Thanks

edtanous, 2022-10-12:
The raspberry pi based images have been relatively unmaintained for some time, and their use case (to boot an inexpensive SOC dev environment) have largely been superseded in most peoples workflows by things like QEMU, which give a better simulation of what a real BMC hardware environment looks like.

With that said, the fix above is likely some minor tweaks to the RPI meta layer config, which, if found, feel free to submit to code review and we will get them reintegrated.

caohj1996, 2022-11-11:
i think

> Hi all I'm trying to build OpenBMC image for Raspberry PI 3-64 on Ubuntu 18.04 / 64 bits machine.
>
> I've followed these steps:
>
> 1. cd openbmc
> 2. export TEMPLATECONF=meta-evb/meta-evb-raspberrypi/conf [...]

12212

caohj1996, 2022-11-11:
> Hi all I'm trying to build OpenBMC image for Raspberry PI 3-64 on Ubuntu 18.04 / 64 bits machine.
>
> I've followed these steps:
>
> 1. cd openbmc
> 2. export TEMPLATECONF=meta-evb/meta-evb-raspberrypi/conf [...]

i think i solve this problem and it caused by version thing. i change my version environment. and work becomes easy
here is my version to solve a series of pi4B-64 problems:
1. change Vmware to WSL2
2. change ubuntu 22.04 to 20.04
3. change openbmc version to 2.9.0
4. python 3.8
though i have no idea about which change make things easier. i think this way 
 is very helpful

closed 2023-09-25 by williamspatrick

Prajwal2930, 2026-02-25:
Run this in openbmc directore 
cmd >export TEMPLATECONF=meta-evb/meta-evb-raspberrypi/conf
          . openbmc-env
### Initializing OE build env ###
**Error:** TEMPLATECONF value (which is /home/prajwal-pt8168/yocto-project/openbmc/meta-evb/meta-evb-raspberrypi/conf) must point to meta-some-layer/conf/templates/template-name