openbmc/openbmc #3885: [quanta-q71l  build failed] on ubuntu 1804 system

hptsf, 2022-09-20:
Branch： master
using command:
      . setup quanta-q71l 
      bitbake obmc-phosphor-image
and build failed, error msg:

bitbake obmc-phosphor-image
WARNING: Host distribution "ubuntu-18.04" has not been validated with this version of the build system; you may possibly experience unexpected failures. It is recommended that you use a tested distribution.
ERROR:  OE-core's config sanity checker detected a potential misconfiguration.
    Either fix the cause of this error or at your own risk disable the checker (see sanity.conf).
    Following is the list of potential problems / advisories:

    Fetcher failure for URL: 'https://yoctoproject.org/connectivity.html'. URL https://yoctoproject.org/connectivity.html doesn't work.
    Please ensure your host's network is configured correctly.
    Please ensure CONNECTIVITY_CHECK_URIS is correct and specified URIs are available.
    If your ISP or network is blocking the above URL,
    try with another domain name, for example by setting:
    CONNECTIVITY_CHECK_URIS = "https://www.example.com/"    You could also set BB_NO_NETWORK = "1" to disable network
    access if all required sources are on local disk.

Summary: There was 1 WARNING message.
Summary: There was 1 ERROR message, returning a non-zero exit code.

amboar, 2022-09-20:
Please check that you can reach example.com from your network. If your DNS isn't resolving it or you have some other filtering preventing access, please follow the advice in the error message.

closed 2022-09-20 by amboar

hptsf, 2022-09-23:
[image]

hptsf, 2022-09-23:
the network to example.com is ok

hptsf, 2022-09-23:
[image]

and “https://yoctoproject.org/connectivity.html” is ok too.