openbmc/openbmc #3887: Not able to boot AST2600 image on Qemu

Jay-vz, 2022-09-27:
**Note on Bugs**

**Unexpected behavior you saw**
 AST2600 image built successfully on an Ubuntu host. When trying to launch on Qemu, it fails to complete the boot process. It complains about bad hash value for kernel-1 image node. 

**Expected behavior**
It should boot the image 

**To Reproduce**
Steps to reproduce the behavior:
1.  git clone https://github.com/openbmc/openbmc
2. cd openbmc/
3.. . setup evb-ast2600 
4. bitbake obmc-phosphor-image
5. Once the image is built, try launching it using the command - qemu-system-arm -m 1G -M ast2600-evb -nographic -drive file=/tmp/deploy/images/evb-ast2600/obmc-phosphor-image-evb-ast2600-20220926160243.static.mtd,format=raw,if=mtd -net nic -net

**Output**
user,hostfwd=:127.0.0.1:2222-:22,hostfwd=:127.0.0.1:2443-:443,hostname=qemu
qemu-system-arm: warning: nic ftgmac100.1 has no peer
qemu-system-arm: warning: nic ftgmac100.2 has no peer
qemu-system-arm: warning: nic ftgmac100.3 has no peer

U-Boot SPL 2019.04 (Sep 23 2022 - 08:28:55 +0000)
Trying to boot from RAM

U-Boot 2019.04 (Sep 23 2022 - 08:28:55 +0000)

SOC: AST2600-A3 
eSPI Mode: SIO:Enable : SuperIO-2e
Eth: MAC0: RGMII, MAC1: RGMII, MAC2: RGMII, MAC3: RGMII
Model: AST2600 EVB
DRAM:  already initialized, 1008 MiB (capacity:1024 MiB, VGA:64 MiB, ECC:off)
MMC:   

sdhci_slot0@100: 1, sdhci_slot1@200: 2, emmc_slot0@100: 0
Loading Environment from SPI Flash... *** Warning - spi_flash_probe_bus_cs() failed, using default environment

In:    serial@1e784000
Out:   serial@1e784000
Err:   serial@1e784000
Model: AST2600 EVB
Net:   
Warning: ftgmac@1e660000 (eth0) using random MAC address - 66:3b:d9:8d:af:be
eth0: ftgmac@1e660000
Warning: ftgmac@1e680000 (eth1) using random MAC address - ee:2c:e9:7e:f1:c9
, eth1: ftgmac@1e680000
Warning: ftgmac@1e670000 (eth2) using random MAC address - 42:84:ee:b2:01:18
, eth2: ftgmac@1e670000
Warning: ftgmac@1e690000 (eth3) using random MAC address - 32:d5:b8:6e:7e:3f
, eth3: ftgmac@1e690000
Hit any key to stop autoboot:  0 
## Loading kernel from FIT Image at 83000000 ...
   Using 'conf-aspeed-ast2600-evb.dtb' configuration
   Verifying Hash Integrity ... OK
   Trying 'kernel-1' kernel subimage
     Description:  Linux kernel
     Type:         Kernel Image
     Compression:  uncompressed
     Data Start:   0x8300012c
     Data Size:    3867128 Bytes = 3.7 MiB
     Architecture: ARM
     OS:           Linux
     Load Address: 0x80001000
     Entry Point:  0x80001000
     Hash algo:    sha256
     Hash value:   fa0e835c5b787446ebcaf3bdcdaaa9b8afc354a5abc15e297d154f0ef46a93b3
   Verifying Hash Integrity ... sha256 error!
Bad hash value for 'hash-1' hash node in 'kernel-1' image node
Bad Data Hash
ERROR: can't get kernel image!
Error loading kernel FIT image
ast# 

**OpenBMC Information:**

uname -a
Linux arc-13 5.15.0-41-generic #44-Ubuntu SMP Wed Jun 22 14:20:53 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

**Additional context**

mdmillerii, 2022-09-27:
Usually this type of error is due to the image located earlier in the flash overruns it's allocation.   It could also be the u-boot environment is configured over the kernel space.   

You should check the fit content in the flash vs the build artifact.

Jay-vz, 2022-09-30:
Do you have any pointers on how to check the fit content in the flash?

mdmillerii, 2022-09-30:
For linux dd can extract a section of a device or file and transfer it to a file.
otherwise check your platform and spi programmer for hints.
concentrate on the beginning and end of each image for the common errors.

sandeep-tcs, 2022-12-23:
We were able to reproduce this issue. By looking at the verified boot sequence on the beaglebone Black, the issue seems to be due to change in kernel. We are trying to manually build an image starting from building U-boot for the specific board, creating a key pair, signing the kernel image and putting U-boot and kernel onto qemu. Will keep you updated.

15171476828, 2023-01-05:
I have the same problem as you. It compiles, but it doesn't work. 

1. . setup evb-ast2600
2. bitbake obmc-phosphor-image
3. Once the image is built, try launching it using the command：
./qemu-system-arm -m 1024 -M ast2600-evb -nographic -drive file=/home/book/yzhtest/AST2600_openbmc/openbmc/build/tmp/deploy/images/evb-ast2600/obmc-phosphor-image-evb-ast2600-20230105053839.static.mtd,format=raw,if=mtd -net nic -net user,hostfwd=:127.0.0.1:2222-:22,hostfwd=:127.0.0.1:2443-:443,hostfwd=udp:127.0.0.1:2623-:623,hostname=qemu
--------------------------------
result: 
Hit any key to stop autoboot:  0
## Loading kernel from FIT Image at 83000000 ...
   Using 'conf-aspeed-ast2600-evb.dtb' configuration
   Verifying Hash Integrity ... OK
   Trying 'kernel-1' kernel subimage
     Description:  Linux kernel
     Type:         Kernel Image
     Compression:  uncompressed
     Data Start:   0x8300011c
     Data Size:    3985816 Bytes = 3.8 MiB
     Architecture: ARM
     OS:           Linux
     Load Address: 0x80001000
     Entry Point:  0x80001000
     Hash algo:    sha256
     Hash value:   f25aafdfd31559c25d158b06c801d6c58acdfb7b398009afd753bca8bc5a3c12
   Verifying Hash Integrity ... sha256 error!
Bad hash value for 'hash-1' hash node in 'kernel-1' image node
Bad Data Hash
ERROR: can't get kernel image!
Error loading kernel FIT image
ast#
--------------------------------

shenki, 2023-01-24:
You must first load the FIT to DRAM, and then execute. The default u-boot command line should be this:

```
bootcmd=run bootspi
bootspi=fdt addr 20100000 && fdt header get fitsize totalsize && cp.b 20100000 ${loadaddr} ${fitsize} && bootm; echo Error loading kernel FIT image
```

sandeep-tcs, 2023-02-01:
After analysing the source code for u-boot, below is our observation:

_## Loading kernel from FIT Image at 83000000 ...
   Using 'conf-aspeed-ast2600-evb.dtb' configuration
   Verifying Hash Integrity ... OK
   Trying 'kernel-1' kernel subimage
     Description:  Linux kernel
     Type:         Kernel Image
     Compression:  uncompressed
     Data Start:   0x8300012c
     Data Size:    3985760 Bytes = 3.8 MiB
     Architecture: ARM
     OS:           Linux
     Load Address: 0x80001000
     Entry Point:  0x80001000
     Hash algo:    sha256
     Hash value:   f6f414009909beebfe717fdb010af642b7915ba38cb7a2d71c874ffb522b9696
   Verifying Hash Integrity ... sha256
fit_value:f6f414009909beebfe717fdb010af642b7915ba38cb7a2d71c874ffb522b9696
value:f99af980ac03ab6f14c6e9add94c93f8e0434e6be084527f28e4bf28950e3dc4
 error!
Bad hash value for 'hash-1' hash node in 'kernel-1' image node
Bad Data Hash
ERROR: can't get kernel image!
Error loading kernel FIT image_

While verifying the has value from the fit image, we get the hash values using fit_image_hash_get_value() in image-fit.c. This value corresponds below value from the logs.

**fit_value:f6f414009909beebfe717fdb010af642b7915ba38cb7a2d71c874ffb522b9696**

To verify the hash values we do calculate the hash value from the data, size and algo provided. This corresponds to below value from the logs.

**value:f99af980ac03ab6f14c6e9add94c93f8e0434e6be084527f28e4bf28950e3dc4**

So, clearly there is a mismatch either while calculating the hash or with the fit image provided.

sandeep-tcs, 2023-02-02:
ast# md.b 20100000
20100000: d0 0d fe ed 00 4d 62 f0 00 00 00 38 00 4d 5f 20    .....Mb....8.M_
20100010: 00 00 00 28 00 00 00 11 00 00 00 10 00 00 00 00    ...(............
20100020: 00 00 00 74 00 4d 5e e8 00 00 00 00 00 00 00 00    ...t.M^.........
20100030: 00 00 00 00 00 00 00 00 00 00 00 01 00 00 00 00    ................
ast#

JianMing-SMCI, 2023-02-22:
I reproduced the same issue with using the qemu downloaded from the link below.
https://jenkins.openbmc.org/job/latest-qemu-x86/
On my build environment, u-boot will use hw_sha256 to calculate the checksum and it got the distinct result with using sha256sum. Since the qemu repository in openbmc is not up to dated so that it might be a bug from ast2600_evb machine. I pickup the ubuntu version of qemu from facebookarchive and can make the same image booted to kernel. 
https://github.com/facebookarchive/openbmc-qemu/releases

sandeep-tcs, 2023-03-14:
AST2600 image works with QEMU version 7.0.50(Earlier versions might also work but not tested) and above on a ubuntu 20 system(On ubuntu 18, you need to do a lot of installations of compatible GLIBC versions of 2.29/2.30).

Please do let us know if you want us to upload the latest version of QEMU onto the OpenBMC code base?

rhung2, 2023-04-03:
saw the same problem, how do i fix it?

ubuntuEE:~/obmc/openbmc/build$ qemu-system-arm -m 256 -M romulus-bmc -nographic \
> -drive file=tmp/deploy/images/romulus/obmc-phosphor-image-romulus-20230403090025.static.mtd,format=raw,if=mtd \
> -net nic \
> -net user,hostfwd=:127.0.0.1:2222-:22,hostfwd=:127.0.0.1:2443-:443,hostname=qemu

U-Boot 2019.04 (Mar 10 2023 - 01:54:10 +0000)

SOC : AST2500-A1
RST : Power On
2nd Boot : Enable
LPC Mode : SIO:Enable : SuperIO-2e
Eth : MAC0: RMII/NCSI, , MAC1: RMII/NCSI,
Model: Romulus BMC
DRAM:  192 MiB (capacity:256 MiB, VGA:64 MiB, ECC:off)
MMC:
Loading Environment from SPI Flash... SF: Detected n25q256a with page size 256 Bytes, erase size 4 KiB, total 32 MiB
*** Warning - bad CRC, using default environment

In:    serial@1e784000
Out:   serial@1e784000
Err:   serial@1e784000
Net:   ftgmac100_probe - NCSI detected

Warning: ethernet@1e660000 (eth0) using random MAC address - b2:66:a2:c7:bc:c3
eth0: ethernet@1e660000
Hit any key to stop autoboot:  0
## Loading kernel from FIT Image at 20080000 ...
   Using 'conf-aspeed-bmc-opp-romulus.dtb' configuration
   Trying 'kernel-1' kernel subimage
     Description:  Linux kernel
     Type:         Kernel Image
     Compression:  uncompressed
     Data Start:   0x20080128
     Data Size:    3081288 Bytes = 2.9 MiB
     Architecture: ARM
     OS:           Linux
     Load Address: 0x80001000
     Entry Point:  0x80001000
     Hash algo:    sha256
     Hash value:   3708d7ba44f722604b2e2658b0d919fa0ea62adcd10bd0925ef378fb88994570
   Verifying Hash Integrity ... sha256+ OK
## Loading ramdisk from FIT Image at 20080000 ...
   Using 'conf-aspeed-bmc-opp-romulus.dtb' configuration
   Trying 'ramdisk-1' ramdisk subimage
     Description:  obmc-phosphor-initramfs
     Type:         RAMDisk Image
     Compression:  uncompressed
     Data Start:   0x20378bf4
     Data Size:    1085208 Bytes = 1 MiB
     Architecture: ARM
     OS:           Linux
     Load Address: unavailable
     Entry Point:  unavailable
     Hash algo:    sha256
     Hash value:   0f186e7fee31717d4732546adcae09ec580865c5c190e487fcde81f7a8d4600e
   Verifying Hash Integrity ... sha256+ OK
## Loading fdt from FIT Image at 20080000 ...
   Using 'conf-aspeed-bmc-opp-romulus.dtb' configuration
   Trying 'fdt-aspeed-bmc-opp-romulus.dtb' fdt subimage
     Description:  Flattened Device Tree blob
     Type:         Flat Device Tree
     Compression:  uncompressed
     Data Start:   0x20370688
     Data Size:    33952 Bytes = 33.2 KiB
     Architecture: ARM
     Hash algo:    sha256
     Hash value:   c84591efdd02c6b02ffc10b61c795cd999ec9f1b8638921fe9c29042e7eb684d
   Verifying Hash Integrity ... sha256+ OK
   Booting using the fdt blob at 0x20370688
   Loading Kernel Image ... OK
   Loading Ramdisk to 89e8a000, end 89f92f18 ... OK
   Loading Device Tree to 89e7e000, end 89e8949f ... OK
fdt_root: FDT_ERR_BADMAGIC
ERROR: root node setup failed
 - must RESET the board to recover.

FDT creation failed! hanging...### ERROR ### Please RESET the board ###
Killed

amboar, 2023-04-26:
> Please do let us know if you want us to upload the latest version of QEMU onto the OpenBMC code base?

openbmc/qemu is now based on v8.0.0

rhung2, 2023-05-13:
Hi,
I used openbmc qemu and saw romulus was running, thanks eachone.

Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) nodistro.0 romulus ttyS4

romulus login: [  OK  ] Finished Wait for /org/openbmc/control/power0.
[  OK  ] Finished Wait for /xyz/ope…ect/control/host0/auto_reboot.
[  OK  ] Finished Wait for /xyz/ope…t/control/host0/boot/one_time.
[  OK  ] Finished Wait for /xyz/ope…mc_project/control/host0/boot.
[  OK  ] Finished Wait for /xyz/ope…ol/host0/power_restore_policy.
[  OK  ] Finished Wait for /xyz/ope…ontrol/host0/restriction_mode.
[  OK  ] Finished Wait for /xyz/openbmc_project/inventory.
[  OK  ] Finished Wait for /xyz/ope…project/led/groups/bmc_booted.
[  OK  ] Finished Wait for /xyz/openbmc_project/led/groups.
[  OK  ] Finished Wait for /xyz/openbmc_project/time/sync_method.

romulus login: root
Password:

AndyWong790924, 2023-07-07:
Hi All,

I use qemu 8.0.2 to build ast2600_openbmc_defconfig not booting with QEMU
The command is as follows:
----------------------------------------------
export ARCH=arm 
export CROSS_COMPILE=arm-linux-gnueabi-
make ast2600_openbmc_defconfig
make -j $(nproc)
dd of=test.img if=./u-boot.bin conv=notrunc bs=1K seek=64
qemu-system-arm -M ast2600-evb -nographic -drive file=test.img,if=sd,format=raw,id=sd0,index=2 -nodefaults -serial mon:stdio
----------------------------------------------
and it shows message：
qemu-system-arm: warning: nic ftgmac100.0 has no peer
qemu-system-arm: warning: nic ftgmac100.1 has no peer
qemu-system-arm: warning: nic ftgmac100.2 has no peer
qemu-system-arm: warning: nic ftgmac100.3 has no peer

it's not booting and gets stuck at the following message.

What should I do to fix this?

williamspatrick, 2023-07-07:
> I use qemu 8.0.2 to build u-boot 

This is, frankly, not an openbmc bug.  OpenBMC builds a complete image and not pieces individually.

ScoobiFreeBSD, 2023-11-03:
I see this problem when trying to boot an OpenBMC for Aspeed ast2600-evb using Qemu on a [GCP Compute Engine VM](https://cloud.google.com/compute), but I can run the same command natively on my amd64 laptop with no problem.
```
qemu-system-arm -M ast2600-evb -mtdblock ./obmc-phosphor-image-evb-ast2600.static.mtd -nographic
WARNING: Image format was not specified for './obmc-phosphor-image-evb-ast2600.static.mtd' and probing guessed raw.
         Automatically detecting the format is dangerous for raw images, write operations on block 0 will be restricted.
         Specify the 'raw' format explicitly to remove the restrictions.

U-Boot SPL 2019.04 (Mar 10 2023 - 01:54:10 +0000)
Trying to boot from RAM

U-Boot 2019.04 (Mar 10 2023 - 01:54:10 +0000)

SOC: AST2600-A3 
eSPI Mode: SIO:Enable : SuperIO-2e
Eth: MAC0: RGMII, MAC1: RGMII, MAC2: RGMII, MAC3: RGMII
Model: AST2600 EVB
DRAM:  already initialized, 1008 MiB (capacity:1024 MiB, VGA:64 MiB, ECC:off)
MMC:   

sdhci_slot0@100: 1, sdhci_slot1@200: 2, emmc_slot0@100: 0
Loading Environment from SPI Flash... *** Warning - spi_flash_probe_bus_cs() failed, using default environment

In:    serial@1e784000
Out:   serial@1e784000
Err:   serial@1e784000
Model: AST2600 EVB
Net:   
Warning: ftgmac@1e660000 (eth0) using random MAC address - 1a:3a:89:b8:fb:e4
eth0: ftgmac@1e660000
Warning: ftgmac@1e680000 (eth1) using random MAC address - 7a:9c:21:04:a9:7d
, eth1: ftgmac@1e680000
Warning: ftgmac@1e670000 (eth2) using random MAC address - de:76:d1:71:1d:20
, eth2: ftgmac@1e670000
Warning: ftgmac@1e690000 (eth3) using random MAC address - 9e:d1:bc:82:9e:69
, eth3: ftgmac@1e690000
Hit any key to stop autoboot:  0 
## Loading kernel from FIT Image at 83000000 ...
   Using 'conf-aspeed-ast2600-evb.dtb' configuration
   Verifying Hash Integrity ... OK
   Trying 'kernel-1' kernel subimage
     Description:  Linux kernel
     Type:         Kernel Image
     Compression:  uncompressed
     Data Start:   0x8300012c
     Data Size:    3904552 Bytes = 3.7 MiB
     Architecture: ARM
     OS:           Linux
     Load Address: 0x80001000
     Entry Point:  0x80001000
     Hash algo:    sha256
     Hash value:   6a8f5eb720fd9e3cf16fb6cb67660cbb9503614462e75a5cb03f6fe2faf48189
   Verifying Hash Integrity ... sha256 error!
Bad hash value for 'hash-1' hash node in 'kernel-1' image node
Bad Data Hash
ERROR: can't get kernel image!
Error loading kernel FIT image
ast#
```

I think problem is that the U-boot bootloader is not being able to find the SPI flash device in VM:

```
Loading Environment from SPI Flash... *** Warning - spi_flash_probe_bus_cs() failed, using default environment
```

Another difference between the two runs is I see `Secure Boot: Mode_2,RSA4096_SHA512` on my laptop but not on the VM:
```
U-Boot 2019.04 (Mar 10 2023 - 01:54:10 +0000)

SOC: AST2600-A3 
Secure Boot: Mode_2, RSA4096_SHA512
eSPI Mode: SIO:Enable : SuperIO-2e
Eth: MAC0: RGMII, MAC1: RGMII, MAC2: RGMII, MAC3: RGMII
Model: AST2600 EVB
DRAM:  already initialized, 1008 MiB (capacity:1024 MiB, VGA:64 MiB, ECC:off)
MMC:
```
I don't have the Qemu version handy; the GCP admin shut down the VM instance; I'll get that by Monday. But I've replicated this with Qemu-8.1.2 on both my laptop and a GCP VM.

williamspatrick, 2023-11-03:
> qemu-system-arm -M ast2600-evb -mtdblock ./obmc-phosphor-image-evb-ast2600.static.mtd -nographic

Please use the `-drive` option rather than passing the image directly as a parameter.  QEMU also warned about this.

ScoobiFreeBSD, 2023-11-03:
Agreed; I used to use `-drive` with the `raw` parameter with similar results.  I can try again.

Yes, same problem, but just in the VM.  The `sf probe` command in the bootloader complains about initializing the SPI flash at 0:0:
```
$ qemu-system-arm -M ast2600-evb -drive file="obmc-phosphor-image-evb-ast2600.static.mtd",format=raw,if=mtd -nographic

U-Boot SPL 2019.04 (Mar 10 2023 - 01:54:10 +0000)
Trying to boot from RAM

U-Boot 2019.04 (Mar 10 2023 - 01:54:10 +0000)

SOC: AST2600-A3 
eSPI Mode: SIO:Enable : SuperIO-2e
Eth: MAC0: RGMII, MAC1: RGMII, MAC2: RGMII, MAC3: RGMII
Model: AST2600 EVB
DRAM:  already initialized, 1008 MiB (capacity:1024 MiB, VGA:64 MiB, ECC:off)
MMC:   

sdhci_slot0@100: 1, sdhci_slot1@200: 2, emmc_slot0@100: 0
Loading Environment from SPI Flash... *** Warning - spi_flash_probe_bus_cs() failed, using default environment

In:    serial@1e784000
Out:   serial@1e784000
Err:   serial@1e784000
Model: AST2600 EVB
Net:   
Warning: ftgmac@1e660000 (eth0) using random MAC address - 0a:8e:45:ce:97:37
eth0: ftgmac@1e660000
Warning: ftgmac@1e680000 (eth1) using random MAC address - e6:6e:1e:c7:b5:45
, eth1: ftgmac@1e680000
Warning: ftgmac@1e670000 (eth2) using random MAC address - 72:38:0b:5e:35:f6
, eth2: ftgmac@1e670000
Warning: ftgmac@1e690000 (eth3) using random MAC address - ee:c3:d8:48:a0:23
, eth3: ftgmac@1e690000
Hit any key to stop autoboot:  0 
## Loading kernel from FIT Image at 83000000 ...
   Using 'conf-aspeed-ast2600-evb.dtb' configuration
   Verifying Hash Integrity ... OK
   Trying 'kernel-1' kernel subimage
     Description:  Linux kernel
     Type:         Kernel Image
     Compression:  uncompressed
     Data Start:   0x8300012c
     Data Size:    3904552 Bytes = 3.7 MiB
     Architecture: ARM
     OS:           Linux
     Load Address: 0x80001000
     Entry Point:  0x80001000
     Hash algo:    sha256
     Hash value:   6a8f5eb720fd9e3cf16fb6cb67660cbb9503614462e75a5cb03f6fe2faf48189
   Verifying Hash Integrity ... sha256 error!
Bad hash value for 'hash-1' hash node in 'kernel-1' image node
Bad Data Hash
ERROR: can't get kernel image!
Error loading kernel FIT image
ast# sf probe
Failed to initialize SPI flash at 0:0 (error -22)
ast#
```
On the laptop, `sf probe` succeeds:
```
ast# sf probe
SF: Detected mx66u51235f with page size 256 Bytes, erase size 4 KiB, total 64 MiB
```
The hash mismatch is because U-boot can't read flash to compute and verify the image hash and compare that hash with the value from the FIT image.

Qemu version on laptop:
```
$ qemu-system-arm --version
QEMU emulator version 7.0.50 (v7.0.0-2471-g8e3d85d36b)
Copyright (c) 2003-2022 Fabrice Bellard and the QEMU Project developers
```
Qemu version on VM:
```
$ qemu-system-arm --version
QEMU emulator version 6.2.0 (Debian 1:6.2+dfsg-2ubuntu6.14)
Copyright (c) 2003-2021 Fabrice Bellard and the QEMU Project developers
```

williamspatrick, 2023-11-03:
I'm not surprised if 6.2 doesn't work out of the box.  Does your 7.0.50 work?

When you built `obmc-phosphor-image-evb-ast2600.static.mtd` did you have to do any modification to it in order to expand it?  If not, I suspect that the default flash device in the QEMU model is _smaller_ than the size the flash image is compiled to target.

```
$ git grep "FLASH_SIZE =" meta-aspeed
meta-aspeed/conf/machine/evb-ast2500.conf:FLASH_SIZE = "32768"
meta-aspeed/conf/machine/evb-ast2600.conf:FLASH_SIZE = "65536"
```

If you didn't change anything, the openbmc image is compiling to a 64MB.  If the QEMU-6.2 doesn't have a 64MB flash chip, it is probably truncating it to 32MB, which is why u-boot can't load the FIT image.

[ If you think this is still a problem after this, let's start a new issue. ]

ScoobiFreeBSD, 2023-11-03:
@williamspatrick wrote:
> I'm not surprised if 6.2 doesn't work out of the box. Does your 7.0.50 work?

Yes, on my laptop.
> When you built obmc-phosphor-image-evb-ast2600.static.mtd did you have to do any modification to it in order to expand it? If not, I suspect that the default flash device in the QEMU model is smaller than the size the flash image is compiled to target.

No modifications to my evb-ast2600 image, which is 64MiB.

I also built [Qemu-8.1.2 from source](https://download.qemu.org/qemu-8.1.2.tar.xz) and used the `qemu-system-arm` binary on both my laptop (works) and the VM (doesn't work).

> [ If you think this is still a problem after this, let's start a new issue. ]

I do, but I'm baffled as to the difference in behavior.  Qemu can read the SPI flash device enough to read and boot the bootloader, but the bootloader can't find the SPI flash device - but _only_ in a GCP VM.

ScoobiFreeBSD, 2023-11-07:
Found my issue today - the GCP VM was missing a [setting to allow nested virtualization](https://cloud.google.com/compute/docs/instances/nested-virtualization/enabling).  So now I can run AST2600 image using Qemu-8.1.2 in a GCP VM.

williamspatrick, 2023-11-07:
Was this Arm-on-Arm?  I'm surprised it wouldn't have just fallen back to the TCG emulation rather than kvm.

ScoobiFreeBSD, 2023-11-07:
Nah, an amd64 GCP VM running an amd64 build of Qemu-8.1.2 to emulate the Aspeed AST2600 EVB board.

I dunno... the GCP Compute Engine docs say that setting typically defaults to enabled unless the admin turns it off explicitly, so when I started tinkering with the setting, I thought I was chasing another dead end.  Just happy to have something working now!

chiyanglin-AStar, 2024-02-23:
I use  latest openbmc code 

. setup evb-ast2600 build

https://github.com/shenki/qemu  to build 

QEMU emulator version 8.0.91 (v8.1.0-rc1-72-g38a6de80b9)
Copyright (c) 2003-2023 Fabrice Bellard and the QEMU Project developers   

is ok !   

qemu command is like :

                       qemu-system-arm -m 512 -M ast2600-evb -nographic -drive file=./image-bmc,format=raw,if=mtd -net nic