openbmc/openbmc #3888: yocto fails to build perf with Linux 6.0

shenki, 2022-10-04:
```
06:46:03 | In function 'parse_events_term__num',
06:46:03 |     inlined from 'parse_events_multi_pmu_add' at util/parse-events.c:1591:6:
06:46:03 | util/parse-events.c:2635:64: warning: array subscript 'YYLTYPE[0]' is partly outside array bounds of 'char[4]' [-Warray-bounds]
06:46:03 |  2635 |                 .err_term  = loc_term ? loc_term->first_column : 0,
06:46:03 |       |                              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~
06:46:03 | util/parse-events.c: In function 'parse_events_multi_pmu_add':
06:46:03 | util/parse-events.c:1576:15: note: object 'config' of size 4
06:46:03 |  1576 |         char *config;
06:46:03 |       |               ^~~~~~
```

shenki, 2022-10-04:
This change did not fix the issue:
```
--- a/poky/meta/recipes-kernel/perf/perf.bb
+++ b/poky/meta/recipes-kernel/perf/perf.bb
@@ -339,7 +339,7 @@ PACKAGES =+ "${PN}-archive ${PN}-tests ${PN}-perl ${PN}-python"
 
 RDEPENDS:${PN} += "elfutils bash"
 RDEPENDS:${PN}-archive =+ "bash"
-RDEPENDS:${PN}-python =+ "bash python3 python3-modules ${@bb.utils.contains('PACKAGECONFIG', 'audit', 'audit-python', '', d)}"
+RDEPENDS:${PN}-python =+ "bash python3 python3-modules python3-setuptools ${@bb.utils.contains('PACKAGECONFIG', 'audit', 'audit-python', '', d)}"
 RDEPENDS:${PN}-perl =+ "bash perl perl-modules"
 RDEPENDS:${PN}-tests =+ "python3 bash"
```

williamspatrick, 2022-10-04:
You likely need to add python3-setuptools to DEPENDS, or more likely need an “inherit python3-setuptools”.

williamspatrick, 2022-10-04:
We will need to contribute whatever the fix ends up being upstream to Yocto.

shenki, 2022-10-04:
Trying this change too:
```
--- a/meta/recipes-kernel/perf/perf.bb
+++ b/meta/recipes-kernel/perf/perf.bb
@@ -322,7 +322,7 @@ PACKAGES =+ "${PN}-archive ${PN}-tests ${PN}-perl ${PN}-python"
 
 RDEPENDS_${PN} += "elfutils bash"
 RDEPENDS_${PN}-archive =+ "bash"
-RDEPENDS_${PN}-python =+ "bash python3 python3-modules ${@bb.utils.contains('PACKAGECONFIG', 'audit', 'audit-python3', '', d)}"
+RDEPENDS_${PN}-python =+ "bash python3 python3-modules python3-setuptools ${@bb.utils.contains('PACKAGECONFIG', 'audit', 'audit-python3', '', d)}"
 RDEPENDS_${PN}-perl =+ "bash perl perl-modules"
 RDEPENDS_${PN}-tests =+ "python3"
```

williamspatrick, 2022-10-04:
RDEPENDS is for "runtime dependencies".  As in, what packages do you need to install with this one in order to make it work.

You appear to be experiencing a build-time issue, so something to do with DEPENDS needs to be adjusted.

In this case, I'm nearly positive that you just need to add `inherit setuptools3`.  There is a bit of content (variables and functions) that that inherit pulls in in order to make setuptools-based packages work.

shenki, 2022-10-05:
Fix sent: https://lore.kernel.org/openembedded-core/20221005025625.64000-1-joel@jms.id.au/

but someone else beat me to it: https://lore.kernel.org/openembedded-core/ce66ad3692d01831e5f41f27702c01744a8708c5.1664904217.git.bruce.ashfield@gmail.com

williamspatrick, 2022-10-05:
Resolved.

closed 2022-10-05 by williamspatrick