openbmc/openbmc #3890: while bitbake I am getting 2 error listed below

mahendra-space, 2022-10-10:
nse_openssl.cc: In function 'int l_md4(lua_State*)':
| nse_openssl.cc:283:32: error: 'MD4' was not declared in this scope; did you mean 'MD5'?
|   283 |   lua_pushlstring( L, (char *) MD4( s, len, digest ), 16 );
|       |                                ^~~
|       |                                MD5
| nse_openssl.cc: In function 'int l_ripemd160(lua_State*)':
| nse_openssl.cc:313:32: error: 'RIPEMD160' was not declared in this scope
|   313 |   lua_pushlstring( L, (char *) RIPEMD160( s, len, digest ), 20 );

williamspatrick, 2022-10-10:
I have no idea what you're doing here based on the BBLAYERS you've shown.  It looks like you're trying to compile meta-phosphor from our tree with a different poky than the one we provide?  This is not supported.  On top of that you're trying to compile against Dunfell which we haven't supported in multiple years now.

This compile error is not very helpful because you've stripped out which package is failing.  I don't know of any code we provide that supplies or uses Lua.  This is the result of either: a mismatch between whatever code is compiling and its expected OpenSSL library version, or a misconfiguration of the OpenSSL library to not include necessary algorithms for whatever you're trying to compile.

closed 2022-10-10 by williamspatrick

mahendra-space, 2022-10-13:
Thanks Williams for your response.

But I want build **OpenBmc**  with "**meta-phosphor**" for **beagleBone Black** by using **bitbake**.
So,could you please guide me.

Thanks,

williamspatrick, 2022-10-13:
You'll want to start from our latest code in master branch.  We don't really support anything else.

You can use `git subtree` to import the meta layer for beaglebone black support into this repository.  Then you'll want to create a `meta-evb-beagleboneblack` under `meta-evb`.  Take a look at what was done for `meta-evb-raspberrypi` for examples.

If you get this working, you can contribute `meta-evb-bbb` to us (via Gerrit) and I can import the appropriate meta-layer formally into our tree.

mahendra-space, 2022-10-17:
Thank Williams,

I have selected RaspberryPi 4B as new hardware and successfully build and ported openBMC for RPi 4B.
By following the link "https://blog.51cto.com/zhxianbin/4997684".

Now I am unable to access openBmc Web page. Please help on that issue.

Thanks in advance.