openbmc/openbmc #3892: openBmc Web page not accessible

mahendra-space, 2022-10-18:
Hi,
I have selected RaspberryPi 4B as new hardware which is successfully build via bitbake and ported openBMC into RPi 4B.
OpenBMC build with meta-phosphor. Here is build config -
Build Configuration:
BB_VERSION           = "1.46.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-20.04"
TARGET_SYS           = "aarch64-openbmc-linux"
MACHINE              = "raspberrypi4-64"
DISTRO               = "openbmc-phosphor"
DISTRO_VERSION       = "0.1.0"
TUNE_FEATURES        = "aarch64 cortexa72 crc crypto"
TARGET_FPU           = ""
meta                 
**meta-poky**            
**meta-oe**              
meta-networking      
meta-python          
**meta-webserver**       
**meta-phospho**r        
meta-raspberrypi     = "dunfell:ab475af3890f35980cd224ec8da7143c68834989" 
I have followed the link "https://blog.51cto.com/zhxianbin/4997684".

Issue is I am not able to access openBmc Web page. I mean when I pass URL "https://ipaddress:port" like that. 
Please help on that.

ljordan-accelsius, 2022-10-19:
Could you please share your local.conf and bblayers.conf?

edtanous, 2022-12-20:
No response from submitter