openbmc/openbmc #3894: openbmc for s2600wf build error: intel-ipmi-oem need

dponamorev1976, 2022-10-31:
Openbmc for s2600wf build error occured at intel-ipmi-oem compilation. As I understand it is necessary to add `inherit pkgconfig` in /openbmc/meta-intel-openbmc/meta-common/recipes-intel/ipmi/intel-ipmi-oem_git.bb 

`inherit cmake obmc-phosphor-ipmiprovider-symlink pkgconfig`

Without this, openbmc for s2600wf compilation fails with an error:

ERROR: intel-ipmi-oem-0.1+gitAUTOINC+097497fb7b-r0 do_configure: ExecutionError('/media/ipc/bmc/openbmc/build/s2600wf/tmp/work/arm1176jzs-openbmc-linux-gnueabi/intel-ipmi-oem/0.1+gitAUTOINC+097497fb7b-r0/temp/run.do_configure.1741972', 1, None, None)
ERROR: Logfile of failure stored in: /media/ipc/bmc/openbmc/build/s2600wf/tmp/work/arm1176jzs-openbmc-linux-gnueabi/intel-ipmi-oem/0.1+gitAUTOINC+097497fb7b-r0/temp/log.do_configure.1741972
Log data follows:
| DEBUG: Executing python function extend_recipe_sysroot
| NOTE: Direct dependencies are ['/media/ipc/bmc/openbmc/meta-openembedded/meta-oe/recipes-support/libgpiod/libgpiod_1.6.3.bb:do_populate_sysroot', '/media/ipc/bmc/openbmc/meta-phosphor/recipes-phosphor/dbus/phosphor-dbus-interfaces_git.bb:do_populate_sysroot', '/media/ipc/bmc/openbmc/meta-phosphor/recipes-phosphor/ipmi/phosphor-ipmi-host_git.bb:do_populate_sysroot', '/media/ipc/bmc/openbmc/meta-phosphor/recipes-phosphor/logging/phosphor-logging_git.bb:do_populate_sysroot', '/media/ipc/bmc/openbmc/meta/recipes-core/glibc/glibc_2.36.bb:do_populate_sysroot', '/media/ipc/bmc/openbmc/meta/recipes-core/systemd/systemd_251.4.bb:do_populate_sysroot', '/media/ipc/bmc/openbmc/meta/recipes-devtools/cmake/cmake-native_3.24.0.bb:do_populate_sysroot', '/media/ipc/bmc/openbmc/meta/recipes-devtools/gcc/gcc-cross_12.2.bb:do_populate_sysroot', '/media/ipc/bmc/openbmc/meta/recipes-devtools/gcc/gcc-runtime_12.2.bb:do_populate_sysroot', '/media/ipc/bmc/openbmc/meta/recipes-devtools/quilt/quilt-native_0.67.bb:do_populate_sysroot', '/media/ipc/bmc/openbmc/meta/recipes-support/boost/boost_1.80.0.bb:do_populate_sysroot', 'virtual:native:/media/ipc/bmc/openbmc/meta/recipes-devtools/ninja/ninja_1.11.1.bb:do_populate_sysroot', 'virtual:native:/media/ipc/bmc/openbmc/meta/recipes-devtools/patch/patch_2.7.6.bb:do_populate_sysroot', 'virtual:native:/media/ipc/bmc/openbmc/meta/recipes-devtools/pseudo/pseudo_git.bb:do_populate_sysroot']
| NOTE: Installed into sysroot: []
| NOTE: Skipping as already exists in sysroot: ['libgpiod', 'phosphor-dbus-interfaces', 'phosphor-ipmi-host', 'phosphor-logging', 'glibc', 'systemd', 'cmake-native', 'gcc-cross-arm', 'gcc-runtime', 'quilt-native', 'boost', 'ninja-native', 'patch-native', 'pseudo-native', 'ncurses-native', 'curl-native', 'bzip2-native', 'xz-native', 'zlib-native', 'zstd-native', 'binutils-cross-arm', 'gnu-config-native', 'texinfo-dummy-native', 'mpfr-native', 'gmp-native', 'linux-libc-headers', 'libtool-native', 'libmpc-native', 'flex-native', 're2c-native', 'sdbusplus', 'shadow-native', 'bash-completion', 'zstd', 'util-linux', 'base-passwd', 'libseccomp', 'opkg-utils', 'libpam', 'shadow-sysroot', 'base-files', 'libcap', 'shadow', 'libgcc', 'zlib', 'bzip2', 'attr-native', 'openssl', 'phosphor-objmgr', 'phosphor-state-manager', 'sdeventplus', 'nlohmann-json', 'libcereal', 'packagegroup-obmc-yaml-providers', 'openssl-native', 'gettext-minimal-native', 'm4-native', 'ncurses', 'libcap-ng', 'util-linux-libuuid', 'libxcrypt', 'flex', 'cracklib', 'libtinyxml2', 'cli11', 'stdplus', 'phosphor-led-manager-yaml-provider', 'phosphor-software-manager-yaml-provider', 'perl-native', 'function2', 'fmt', 'liburing', 'make-native', 'gdbm-native']
| DEBUG: Python function extend_recipe_sysroot finished
| DEBUG: Executing shell function do_configure
| -- The CXX compiler identification is GNU 12.2.0
| -- Detecting CXX compiler ABI info
| -- Detecting CXX compiler ABI info - done
| -- Check for working CXX compiler: /media/ipc/bmc/openbmc/build/s2600wf/tmp/work/arm1176jzs-openbmc-linux-gnueabi/intel-ipmi-oem/0.1+gitAUTOINC+097497fb7b-r0/recipe-sysroot-native/usr/bin/arm-openbmc-linux-gnueabi/arm-openbmc-linux-gnueabi-g++ - skipped
| -- Detecting CXX compile features
| -- Detecting CXX compile features - done
| CMake Error at /media/ipc/bmc/openbmc/build/s2600wf/tmp/work/arm1176jzs-openbmc-linux-gnueabi/intel-ipmi-oem/0.1+gitAUTOINC+097497fb7b-r0/recipe-sysroot-native/usr/share/cmake-3.24/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
|   Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE)

But in this case after adding `inherit pkgconfig`, the following error occurs:
 
fatal error: com/intel/Control/OCOTShutdownPolicy/server.hpp: No such file or directory
|    28 | #include <com/intel/Control/OCOTShutdownPolicy/server.hpp>

Please tell me what could be the problem with intel-ipmi-oem compilation? 

Log attached below:

Build Configuration:
BB_VERSION           = "2.2.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-20.04"
TARGET_SYS           = "arm-openbmc-linux-gnueabi"
MACHINE              = "s2600wf"
DISTRO               = "openbmc-phosphor"
DISTRO_VERSION       = "nodistro.0"
TUNE_FEATURES        = "arm thumb arm1176jzs"
TARGET_FPU           = "soft"
meta                 
meta-poky            
meta-oe              
meta-networking      
meta-python          
meta-phosphor        
meta-aspeed          
meta-intel-openbmc   
meta-s2600wf         = "master:623fa3ff363a3f5de664b8c7ca7f693824983b3f"

Initialising tasks: 100% |##################################################################################################################################################################| Time: 0:00:03
Sstate summary: Wanted 940 Local 0 Mirrors 0 Missed 940 Current 795 (0% match, 45% complete)
Removing 4 stale sstate objects for arch s2600wf: 100% |####################################################################################################################################| Time: 0:00:00
Removing 2 stale sstate objects for arch allarch: 100% |####################################################################################################################################| Time: 0:00:00
NOTE: Executing Tasks
ERROR: intel-ipmi-oem-0.1+gitAUTOINC+097497fb7b-r0 do_compile: ExecutionError('/media/ipc/bmc/openbmc/build/s2600wf/tmp/work/arm1176jzs-openbmc-linux-gnueabi/intel-ipmi-oem/0.1+gitAUTOINC+097497fb7b-r0/temp/run.do_compile.1803477', 1, None, None)
ERROR: Logfile of failure stored in: /media/ipc/bmc/openbmc/build/s2600wf/tmp/work/arm1176jzs-openbmc-linux-gnueabi/intel-ipmi-oem/0.1+gitAUTOINC+097497fb7b-r0/temp/log.do_compile.1803477
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: VERBOSE=1 cmake --build /media/ipc/bmc/openbmc/build/s2600wf/tmp/work/arm1176jzs-openbmc-linux-gnueabi/intel-ipmi-oem/0.1+gitAUTOINC+097497fb7b-r0/build --target all --
| [1/19] cd /media/ipc/bmc/openbmc/build/s2600wf/tmp/work/arm1176jzs-openbmc-linux-gnueabi/intel-ipmi-oem/0.1+gitAUTOINC+097497fb7b-r0/git && ./generate-whitelist.py ipmi-whitelist.conf /media/ipc/bmc/openbmc/build/s2600wf/tmp/work/arm1176jzs-openbmc-linux-gnueabi/intel-ipmi-oem/0.1+gitAUTOINC+097497fb7b-r0/build/ipmi-whitelist.hpp
| [2/19] /media/ipc/bmc/openbmc/build/s2600wf/tmp/work/arm1176jzs-openbmc-linux-gnueabi/intel-ipmi-oem/0.1+gitAUTOINC+097497fb7b-r0/recipe-sysroot-native/usr/bin/arm-openbmc-linux-gnueabi/arm-openbmc-linux-gnueabi-g++ -DBOOST_ALL_NO_LIB -DBOOST_ASIO_DISABLE_THREADS -DBOOST_COROUTINES_NO_DEPRECATION_WARNING -DBOOST_ERROR_CODE_HEADER_ONLY -DBOOST_NO_RTTI -DBOOST_NO_TYPEID -DBOOST_SYSTEM_NO_DEPRECATED -DUSING_ENTITY_MANAGER_DECORATORS -Dzinteloemcmds_EXPORTS -I/media/ipc/bmc/openbmc/build/s2600wf/tmp/work/arm1176jzs-openbmc-linux-gnueabi/intel-ipmi-oem/0.1+gitAUTOINC+097497fb7b-r0/git/include -isystem /media/ipc/bmc/openbmc/build/s2600wf/tmp/work/arm1176jzs-openbmc-linux-gnueabi/intel-ipmi-oem/0.1+gitAUTOINC+097497fb7b-r0/build -marm -mcpu=arm1176jz-s -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/media/ipc/bmc/openbmc/build/s2600wf/tmp/work/arm1176jzs-openbmc-linux-gnueabi/intel-ipmi-oem/0.1+gitAUTOINC+097497fb7b-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/media/ipc/bmc/openbmc/build/s2600wf/tmp/work/arm1176jzs-openbmc-linux-gnueabi/intel-ipmi-oem/0.1+gitAUTOINC+097497fb7b-r0/git=/usr/src/debug/intel-ipmi-oem/0.1+gitAUTOINC+097497fb7b-r0  -fdebug-prefix-map=/media/ipc/bmc/openbmc/build/s2600wf/tmp/work/arm1176jzs-openbmc-linux-gnueabi/intel-ipmi-oem/0.1+gitA [... cut ...] zs-openbmc-linux-gnueabi/intel-ipmi-oem/0.1+gitAUTOINC+097497fb7b-r0/git/src/oemcommands.cpp
| /media/ipc/bmc/openbmc/build/s2600wf/tmp/work/arm1176jzs-openbmc-linux-gnueabi/intel-ipmi-oem/0.1+gitAUTOINC+097497fb7b-r0/git/src/oemcommands.cpp:28:10: fatal error: com/intel/Control/OCOTShutdownPolicy/server.hpp: No such file or directory
|    28 | #include <com/intel/Control/OCOTShutdownPolicy/server.hpp>
|       |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
| compilation terminated.

williamspatrick, 2022-10-31:
I suspect you are missing the Yocto config variable to enable the Intel OEM dbus interfaces.

```
OBMC_ORG_YAML_SUBDIRS += "com/intel"
```

You should be able to find this variable in other places in the tree as an example (but with different values because nobody else needs the Intel OEM stuff).

Since meta-intel doesn't seem to have this it seems like nobody is actively maintaining it.  Would you mind supplying the fixes?

cc: @jmbills

dponamorev1976, 2022-11-01:
Thanks for the help! I added a line OBMC_ORG_YAML_SUBDIRS += "com/intel" to the /openbmc/meta-intel-openbmc/conf/machine/include/intel.inc file and the compilation was successful. I will try to put these changes on gerrit.openbmc.org. Only I don't quite understand which repository I should use for it?

dponamorev1976, 2022-11-01:
I think I figured it out:       https://gerrit.openbmc.org/c/openbmc/openbmc/+/58334

closed 2022-11-03 by dponamorev1976