openbmc/openbmc #3897: ifconfig not showing /etc/systemd/network/00-bmc-eth0.network details

AlexD202, 2022-12-16:
I have used the following commands to set MAC and IP Address:
busctl set-property xyz.openbmc_project.Network /xyz/openbmc_project/network/eth0 xyz.openbmc_project.Network.MACAddress MACAddress s "XX:XX:XX:XX:XX:XX"

busctl call xyz.openbmc_project.Network /xyz/openbmc_project/network/eth0 xyz.openbmc_project.Network.IP.Create IP ssys "xyz.openbmc_project.Network.IP.Protocol.IPv4" "X.X.X.X"  24 "X.X.X.X"

The file under /etc/systemd/network/00-bmc-eth0.network shows correct IP and MAC address. However when I do ifconfig eth0, it shows the following (only MAC is correct) and no IP or GW is showing:
root@evb-ast2500:~# ifconfig eth0
eth0      Link encap:Ethernet  HWaddr 00:02:C5:12:34:56
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:70 overruns:0 frame:140
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:22

When I try to pink any IP within that subnet, I get the following error:
root@evb-ast2500:~# ping 172.17.217.156
PING 172.17.217.156 (172.17.217.156): 56 data bytes
ping: sendto: Network is unreachable

Can someone please help me find out why these two show different details and why  I am not able to see the changes under ifcpnfog?

Thanks,

williamspatrick, 2022-12-16:
Why did you ignore and not provide any details from the bug report template?  What revision of code are you trying to run?  What kind of network card?  Direct attached, NC-SI, etc.

AlexD202, 2022-12-16:
Code Details:
1) BMC Code was checked out from https://github.com/openbmc/openbmc

2) Code Revision is:  svn info --show-item last-changed-revision
15156

BMC Image Details:
1. uname -a
Linux proteus.burlington.evertz.tv 5.11.0-25-generic #27~20.04.1-Ubuntu SMP Tue Jul 13 17:41:23 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

2. cat /etc/os-release
NAME="Ubuntu"
VERSION="20.04.5 LTS (Focal Fossa)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 20.04.5 LTS"
VERSION_ID="20.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=focal
UBUNTU_CODENAME=focal

Networking Details:
1. ifconfig
eth0      Link encap:Ethernet  HWaddr 00:02:C5:12:34:56
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:152 overruns:0 frame:304
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:22

eth1      Link encap:Ethernet  HWaddr 00:02:C5:12:34:57
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:142 overruns:0 frame:284
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:23

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:18 errors:0 dropped:0 overruns:0 frame:0
          TX packets:18 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:2803 (2.7 KiB)  TX bytes:2803 (2.7 KiB)

2. ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast qlen 1000
    link/ether 00:02:c5:12:34:56 brd ff:ff:ff:ff:ff:ff
3: eth1: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast qlen 1000
    link/ether 00:02:c5:12:34:57 brd ff:ff:ff:ff:ff:ff

3.  cat /etc/systemd/network/00-bmc-eth0.network
[DHCP]
SendHostname=true
UseHostname=true
UseDomains=true
UseDNS=true
UseNTP=true
ClientIdentifier=mac
[IPv6AcceptRA]
DHCPv6Client=false
[Network]
Address=172.17.217.156/24
Gateway=172.17.217.1
DHCP=false
IPv6AcceptRA=false
LinkLocalAddressing=yes
[Link]
MACAddress=00:02:c5:12:34:56
[Match]
Name=eth0

Steps to reproduce:
1. I compiled the code without making any changes for evb-ast2500
2. Loaded the code into BMC Module
3. Used following commands to set  set MAC and IP Address:
busctl set-property xyz.openbmc_project.Network /xyz/openbmc_project/network/eth0 xyz.openbmc_project.Network.MACAddress MACAddress s "XX:XX:XX:XX:XX:XX"

busctl call xyz.openbmc_project.Network /xyz/openbmc_project/network/eth0 xyz.openbmc_project.Network.IP.Create IP ssys "xyz.openbmc_project.Network.IP.Protocol.IPv4" "X.X.X.X" 24 "X.X.X.X"

Expected result: both ifconfig and 00-bmc-eth0.network should show the new ip address and I should be able to ping the newly created IP (172.17.217.156).

Actual result: 00-bmc-eth0.network shows the correct details, but ifconfig does not show any value for IP or GW. When I ping 172.17.217.156, I get the following error
root@evb-ast2500:~# ping 172.17.217.156
PING 172.17.217.156 (172.17.217.156): 56 data bytes
ping: sendto: Network is unreachable

amboar, 2022-12-16:
> Code Details:
>
>    BMC Code was checked out from https://github.com/openbmc/openbmc

Okay, but what commit ID have you built it from, because

>
>    Code Revision is: svn info --show-item last-changed-revision
>    15156

This is Github, which holds git repositories, and git is used extensively by the OpenBMC project. An SVN ID makes no sense in this context. It does not help us identify what software was built into your image.

>
> BMC Image Details:
>
>    uname -a
>    Linux proteus.burlington.evertz.tv 5.11.0-25-generic #27 20.04.1-Ubuntu SMP Tue Jul 13 17:41:23 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

I suspect this isn't `uname -a` output from a BMC.

>
> cat /etc/os-release
> NAME="Ubuntu"
> VERSION="20.04.5 LTS (Focal Fossa)"

If you're running Ubuntu on your BMC you should be filing bugs against Ubuntu, not OpenBMC.

But I suspect you're not running Ubuntu on your BMC, and are really providing us with the details of your development machine. This isn't what was asked for, and is only useful for a very narrow set of bugs (failing to build bitbake native recipes).

You need to provide us with information that actually might help us resolve your problem. This requires giving us a clear understanding of all the relevant context. There's not much point opening an issue otherwise.

AlexD202, 2022-12-16:
Sorry my bad. I had given you wrong details:

Here is the correct details:
uname -a
Linux evb-ast2500 6.0.10-115a0aa #1 Mon Nov 28 02:55:02 UTC 2022 armv6l GNU/Linux

AlexD202, 2022-12-16:
Sorry I had provided the wrong details. I updated my comment with the correct revision and image details... sorry for the confusion.

AlexD202, 2022-12-19:
Hi, do you guys have all the details to help me with my issue? thanks

williamspatrick, 2022-12-20:
> Hi, do you guys have all the details to help me with my issue? thanks

I still know nothing about what hardware you are targeting here.  Is this a direct attached PHY or NC-SI?  Have you looked at dmesg output to ensure there isn't a problem in the network driver?

You mentioned compiling an EVB image but are you loading this on an EVB?

williamspatrick, 2023-09-25:
Insufficient details given.  No follow up from originator.