openbmc/openbmc #3898: initramfs commandline parsing is fraglie

shenki, 2022-12-22:
See [meta-phosphor/recipes-phosphor/initrdscripts/phosphor-mmc-init/mmc-init.sh](https://github.com/openbmc/openbmc/blob/bc1b9a18a5505a5f997aa5af2e2672573bddeb19/meta-phosphor/recipes-phosphor/initrdscripts/phosphor-mmc-init/mmc-init.sh#L4)

It expects a command line string to be in the form:

```
console=ttyS4,115200n8 rootwait root=PARTLABEL=rofs-a
```

If booting with `root=PARTLABEL=rofs-a` at the start of the command line string (instead of the end), the parsing gets confused: 

```
[    8.241229] /dev/disk/by-partlabel/root=PARTLABEL=rofs-a: Can't open blockdev
mount: mounting /dev/disk/by-partlabel/root=PARTLABEL=rofs-a on /mnt/rofs failed: No such file or directory
```

Similarly if the partition is not set, it gets confused:
```
[    1.919816] Run /init as init process
Starting version 251.8+
[    8.219396] /dev/disk/by-partlabel/rootwait: Can't open blockdev
mount: mounting /dev/disk/by-partlabel/rootwait on /mnt/rofs failed: No such file or directory
```

We should fix `get_root` so that it checks for the existance of root=PARTLABEL, and matches on it. We could fall back on rofs-a if it can't be found?

shenki, 2022-12-22:
@anoo1 @amboar are recent contributors to this code who may be interested in working on it.

amboar, 2023-01-23:
Yeah, this seems a bit silly.

amboar, 2023-01-25:
I pushed this: https://gerrit.openbmc.org/c/openbmc/openbmc/+/60447

williamspatrick, 2023-09-25:
I think this is resolved.

closed 2023-09-25 by williamspatrick