openbmc/openbmc #3901: Why no u-boot output on serial port?

dponamorev1976, 2023-01-13:
Hi! 

I am trying to adapt openbmс firmware for Gigabyte C621-SU8 motherboard. 
It has: 
1st and 2nd Generation Intel® Xeon® Scalable Processors, 
ASPEED® AST2500 BMC chip,
1 x Realtek RTL8211E chip (10/100/1000 Mbit) (LAN1),
2 x Intel® 210AT GbE LAN chips (10/100/1000 Mbit) (LAN2, LAN3),
2 x Serial port: COM0, COM1

The openbmc s2600wf target machine used. 
I was hoping to see the u-boot loading output on the serial port, but nothing out on it.

What could it be? Is output to the serial port disabled by default in u-boot or 
serial port on AST2500 does not match ttyS4 (default for s2600wf)?
How can I change serial port on u-boot to see the output? U-boot version is v2016.07.

zevweiss, 2023-01-13:
Firstly, if you're attempting a port to a new platform, I'd recommend creating a new meta-layer for it instead of trying to reuse an existing one.

Secondly, where are you looking for serial output?  If it's COM0 and/or COM1, those are probably the host's serial ports, not the BMC's.  The BMC typically has its own independent serial port where u-boot & kernel console output go, but it's not always easily accessible unfortunately -- some boards put it on header pins, some don't really expose it at all and just have bare PCB pads or (possibly depopulated) surface-mount resistors as the only point you can really access the RX/TX lines.  Unless you have contacts with your hardware vendor or access to documentation, you might need to probe around a bit on the board with a multimeter/oscilloscope or the like to try to find it.

(Or you can just attempt to do the port "blind", i.e. without a BMC serial port -- it's certainly not ideal, but if you can get the network interface working and just log in via that you _might_ be able to get away without it.)

dponamorev1976, 2023-01-16:
Thanks for the detailed answer. 

Motherboard Gigabyte C621-SU8 is a clone of motherboard supermicro X11SPL-F.  Judging by the functional diagram (see attachment), the COM ports are connected to the BMC. One COM0 port on the back plate, the second internal COM1 header. The host uses both of them (I can configure the bios to redirect the host to them).
I tried almost all target machine versions for Aspeed AST2500 - I couldn't get a working network interface on any of them. Two adapters are connected to the ASPEED AST2500 on Gigabyte C621-SU8. The first one is Intel I210AT via RGRMII/NCSI. Second dedicated network interface Realtek RTL8211E.
As for the new meta level, I trie
[System Block Diagram.pdf](https://github.com/openbmc/openbmc/files/10426730/System.Block.Diagram.pdf)
d to make it based on "romulus" but also without success (I made my device tree with the second nic enabled and add realtek_phy driver in linux config). 
I watched the flash memory (W25Q256JV) output for AST2500 with an oscilloscope and the data was flow, but at what stage the failure occurs is not clear. That's why I needed a serial console. 

Maybe there will be some other tips on raising network interfaces in my configuration?

mdmillerii, 2023-01-16:
You need to check your device tree.   For the NCSI the port needs use-ncsi property and for the other you should specify the phy.   You might need to check the strapping of the phy mii id.  There are other systems using i210 so that shouldn't be a problem properly configured.

dponamorev1976, 2023-01-18:
Board x11spi is very similar to my motherboard. 
(Instead of 2x intel i210AT it intel x722, but dedicated lan same: realtek rtl8211e).
The compiled image for x11spi failed to connect over the network.

Device tree in aspeed-g5.dtsi:

```
&mac0 {
    status = "okay";

    pinctrl-names = "default";
    pinctrl-0 = <&pinctrl_rmii1_default>;
    clocks = <&syscon ASPEED_CLK_GATE_MAC1CLK>,
           <&syscon ASPEED_CLK_MAC1RCLK>;
    clock-names = "MACCLK", "RCLK";
    use-ncsi;
};

&mac1 {
	status = "okay";

	pinctrl-names = "default";
	pinctrl-0 = <&pinctrl_rgmii2_default &pinctrl_mdio2_default>;
};
```

Device tree in aspeed-g5.dtsi:

```
ethernet@1e660000 {
	compatible = "aspeed,ast2500-mac\0faraday,ftgmac100";
	reg = <0x1e660000 0x180>;
	interrupts = <0x02>;
	clocks = <0x02 0x11 0x02 0x24>;
	status = "okay";
	pinctrl-names = "default";
	pinctrl-0 = <0x04>;
	clock-names = "MACCLK\0RCLK";
	use-ncsi;
};

ethernet@1e680000 {
	compatible = "aspeed,ast2500-mac\0faraday,ftgmac100";
	reg = <0x1e680000 0x180>;
	interrupts = <0x03>;
	clocks = <0x02 0x12>;
	status = "okay";
	pinctrl-names = "default";
	pinctrl-0 = <0x05 0x06>;
};

```
Resistors on RTL8211E pins:
PHY_AD0[34]=0, (no resistor)
PHY_AD1[35]=1,
PHY_AD2[13]=1

Accordingly PHY_ADDR = 0x06? How to display it in devicetree? 
Or am I missing something about strapping of the phy mii id? Is that not it?

dponamorev1976, 2023-01-19:
Thanks for the link! That helped!
I was able to find the output of the console port TX pin. I bring it picture it location (red circle). 

[image]

Just opposite the landfill "K" 

[image]

Judging by the log, Linux loading hangs. 

[putty_ga_c621_su8_bmc_19.01.2023.log](https://github.com/openbmc/openbmc/files/10456894/putty_ga_c621_su8_bmc_19.01.2023.log)

```
[    6.306161] systemd[1]: Hostname set to <x11spi>.
[    8.407295] systemd[1]: Failed to put bus name to hashmap: File exists
[    8.414208] systemd[1]: xyz.openbmc_project.State.Host@0.service: Two services allocated for the same bus name xyz.openbmc_project.State.Host0, refusing operation.
[   10.872177] systemd[1]: /lib/systemd/system/phosphor-ipmi-net@.socket:3: Invalid interface name, ignoring: sys-subsystem-net-devices-%i.device
[   11.004523] systemd[1]: Failed to isolate default target: Unit xyz.openbmc_project.State.Host@0.service failed to load properly, please adjust/correct and reload service manager: File exists
[!!!!!!] Failed to isolate default target.
[   11.308690] systemd[1]: Freezing execution.
```
Maybe temporarily exclude something when compiling the image so that the booting goes through and it is possible to work on the network?

dponamorev1976, 2023-01-19:
Found a similar issue while booting...
https://github.com/openbmc/openbmc/issues/3853

zevweiss, 2023-01-20:
Yeah, looks like the same issue.  AFAIK the `VIRTUAL-RUNTIME_obmc-{host,chassis}-state-manager` switch should select one package or the other (phosphor-state-manager or x86-power-control), but it seems we're somehow ending up with both.  I'll try to see if I can figure anything out about what's going on there, though I'm not great at yocto debugging...

In the meantime, if you can also locate an RX contact for the serial console and you want to just get a (bare-bones) shell to poke around with, you can hit a key to interrupt u-boot's default boot sequence and then use the u-boot command prompt to add some kernel command-line flags that will drop you into a root shell instead of starting the usual systemd boot process:

```
$ setenv bootargs $bootargs debug-init-sh enable-initrd-debug-sh
$ boot
```

dponamorev1976, 2023-01-26:
The suggested solution didn't work. I mean adding lines: 
VIRTUAL-RUNTIME_obmc-chassis-state-manager = "x86-power-control"
VIRTUAL-RUNTIME_obmc-discover-system-state = "x86-power-control"
to meta-intel-openbmc/conf/machine/include/intel.inc file. 
Maybe it was necessary to execute the "devtool modify" command for some module?

Images for x11spi and s2600wf - the bootin is stop with the same errors.
The image for evb-ast2500 works and I was able to connect over the network via ssh. 
I did not notice this right away, since the indicators on the realtek RJ45 connector did not work correctly.
I would like to clarify what needs to be done so that the x11spi image work correctly? 
Should edit meta-phosphor recipes to exclude phosphor-state-manager interface?

dponamorev1976, 2023-01-30:
With this change in the code of the file openbmc/meta-supermicro/conf/machine/include/supermicro.inc 
```
VIRTUAL-RUNTIME_obmc-host-state-manager = "x86-power-control"
VIRTUAL-RUNTIME_obmc-chassis-state-manager = "x86-power-control"
VIRTUAL-RUNTIME_obmc-discover-system-state = "x86-power-control"
```
there is no error. 

Thank you all, I'll look into it further.

```
MACHINE_FEATURES += "\
        obmc-bmc-state-mgmt \
        obmc-chassis-state-mgmt \
        obmc-host-state-mgmt \
        obmc-host-ipmi \
        obmc-phosphor-chassis-mgmt \
        obmc-phosphor-fan-mgmt \
        obmc-phosphor-flash-mgmt \
        "

VIRTUAL-RUNTIME_skeleton_workbook = "${MACHINE}-config"
# since we use x86-power-control service, we don't need below three services
VIRTUAL-RUNTIME_obmc-host-state-manager = "x86-power-control"
VIRTUAL-RUNTIME_obmc-chassis-state-manager = "x86-power-control"
VIRTUAL-RUNTIME_obmc-discover-system-state = "x86-power-control"

PREFERRED_PROVIDER_virtual/obmc-chassis-mgmt = "packagegroup-supermicro-apps"
PREFERRED_PROVIDER_virtual/obmc-fan-mgmt = "packagegroup-supermicro-apps"
PREFERRED_PROVIDER_virtual/obmc-flash-mgmt = "packagegroup-supermicro-apps"
PREFERRED_PROVIDER_virtual/obmc-system-mgmt = "packagegroup-supermicro-apps"
PREFERRED_PROVIDER_virtual/obmc-inventory-data ?= "${VIRTUAL-RUNTIME_skeleton_workbook}"
PREFERRED_PROVIDER_virtual/obmc-host-ipmi-hw ?= "phosphor-ipmi-kcs"

OVERRIDES .= ":supermicro"
```

closed 2023-01-30 by dponamorev1976