openbmc/openbmc #3903: No rotation of runtime journal

ofaurax, 2023-01-31:
**Unexpected behavior you saw**
After about 24h of uptime, the partition containing the journal is full (~40Mo).
We see error messages on console : `systemd-journald[153]: Failed to open runtime journal: No space left on device`
This bug has been seen due to #3902 (mDNS is verbose in the journal)

**Expected behavior**
Partition not full, and the journal is rotated (like other logs)

**To Reproduce**
Steps to reproduce the behavior:
1. Compile and run openbmc on qemu
2. Wait (~24h)

**OpenBMC Information:**
 - /etc/os-release:
```
ID=openbmc-phosphor
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="2.14.0-dev"
VERSION_ID=2.14.0-dev-367-gf0dca1402
VERSION_CODENAME="langdale"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 2.14.0-dev"
BUILD_ID="20230112104511"
OPENBMC_TARGET_MACHINE="qemuarm"
EXTENDED_VERSION="2.14.0-dev-367-gf0dca1402"
```
 - uname -a: `Linux qemuarm 5.19.14-yocto-s-dirty-8150698 #1 SMP PREEMPT Thu Oct 6 18:17:00 UTC 2022 armv7l GNU/Linux`

ofaurax, 2023-02-21:
Perhaps we could limit the size by using this tip: https://askubuntu.com/questions/1238214/big-var-log-journal

ofaurax, 2023-03-01:
For the record, I managed to prevent the `No space left` error with this seetings in `/etc/systemd/journald.conf`:
```
RuntimeMaxFileSize=6M
RuntimeMaxFiles=5
```

zevweiss, 2023-03-03:
@ofaurax, want to prepare a patch for that and post it to gerrit?  I expect there'll be some details to work out about exactly where/how it gets done, but that's probably a better place to have that discussion.

zevweiss, 2023-03-03:
(As one small suggestion I think it'd probably be better to add this as a single drop-in file in `/usr/lib/systemd/journald.conf.d` instead of changing `/etc/systemd/journald.conf`.)

ofaurax, 2023-05-11:
Patch submitted here: https://gerrit.openbmc.org/c/openbmc/openbmc/+/63409

matthewfischer, 2025-01-12:
The original PR seems to be abandoned (and perhaps considered fixed?) and I cannot reproduce the mDNS issue. Can you note if it's still happening?