openbmc/openbmc #3904: Build full image for Supermicro X11SBA-LN4F with ASPEED AST2400 ?

CpServiceSpb, 2023-02-13:
I am the beginner at the time 
There is Supermicro X11SBA-LN4F wih on-board Aspeed AST2400.
Is it possible to build full Openbmc image with Web Ui, remote Html5 kvm, virtual media support and sensors support as default BMC/IPmi firmware replacement within Openbmc project ?

That is:
1. is there AST2400, not AST2500 definitions in Openbmc ?
2. does Openmbc have WebUI ? 
3. does Openbmc support Html5 not Jave kvm and virtual media ?
4. somehow to get sensors ID/addresses/types is necessary, is it ?

williamspatrick, 2023-02-14:
As mentioned in the issue template(s), we request you don't use issues for support / inquiries.

Someone started porting to X11SPI but not SBA.  I don't know what state of the port is or if it is sufficient.

```
meta-supermicro/meta-x11spi/conf/layer.conf
```

We do support AST2400 thru AST2600.  There is a WebUI with non-Java KVM.  I don't know what you're asking about sensor support but there are certainly multiple ways to read sensor readings.

Porting to new motherboard is a non-trivial effort.  Generally you need some schematics available to figure out what devices are connected to the BMC and how.

closed 2023-02-14 by williamspatrick

CpServiceSpb, 2024-06-06:
Hi.
Regarding Supermicro X11SBA-LN4F based on AST2400, what is exactly necessary to add t the project that is to build flash image for the motherboard ?
I meant sensors list, type it addresses or something else ?