openbmc/openbmc #3905: ipmid and netipmid exiting with segmentation fault || openbmc version 2.12

akraj19, 2023-02-23:
Hi all,
We are facing issue with ipmid and netipmid . Both are exiting with segmenation fault. We are using version 2.12.

**Please see  error logs for reference:** 
root@aaaa:/etc/eip# journalctl | grep ipmi
Jan 01 00:01:36 aaaa systemd[1]: /lib/systemd/system/phosphor-ipmi-net@.socket:3: Invalid interface name, ignoring: sys-subsystem-net-devices-%i.device
Jan 01 00:01:36 aaaa systemd[1]: Created slice Slice /system/phosphor-ipmi-kcs.
Jan 01 00:01:36 aaaa systemd[1]: Created slice Slice /system/phosphor-ipmi-net.
Jan 01 00:02:05 aaaa systemd[1]: Listening on phosphor-ipmi-net@eip0.socket.
Jan 01 00:02:37 aaaa systemd[1]: phosphor-ipmi-host.service: Unexpected error response from GetNameOwner(): Connection terminated
Jan 01 00:03:08 aaaa systemd[1]: phosphor-ipmi-net@eip0.service: Job phosphor-ipmi-net@eip0.service/start failed with result 'dependency'.
Jan 01 00:03:17 aaaa ipmid[191]: Channel not configured so loading default.
Jan 01 00:03:17 aaaa ipmid[191]: Channel not configured so loading default.
Jan 01 00:03:17 aaaa ipmid[191]: JSON file not found
Jan 01 00:03:19 aaaa ipmid[191]: Registering OEM:[0X00C2CF], Cmd:[0X04] for Manual Zone Control
Jan 01 00:03:18 aaaa ipmid[191]: JSON file not found
Jan 01 00:03:23 aaaa kcsbridged-ipmi-kcs3[201]: FAILED: open `/dev/ipmi-kcs3`: No such file or directory
Jan 01 00:03:25 aaaa systemd[1]: phosphor-ipmi-kcs@ipmi-kcs3.service: Main process exited, code=exited, status=1/FAILURE
Jan 01 00:03:25 aaaa systemd[1]: phosphor-ipmi-kcs@ipmi-kcs3.service: Failed with result 'exit-code'.
Jan 01 00:03:28 aaaa systemd[1]: phosphor-ipmi-kcs@ipmi-kcs3.service: Scheduled restart job, restart counter is at 1.
Jan 01 00:03:32 aaaa kcsbridged-ipmi-kcs3[209]: FAILED: open `/dev/ipmi-kcs3`: No such file or directory
Jan 01 00:03:33 aaaa systemd[1]: phosphor-ipmi-kcs@ipmi-kcs3.service: Main process exited, code=exited, status=1/FAILURE
Jan 01 00:03:33 aaaa systemd[1]: phosphor-ipmi-kcs@ipmi-kcs3.service: Failed with result 'exit-code'.
Jan 01 00:03:35 aaaa systemd[1]: phosphor-ipmi-kcs@ipmi-kcs3.service: Scheduled restart job, restart counter is at 2.
Jan 01 00:03:36 aaaa systemd[1]: phosphor-ipmi-kcs@ipmi-kcs3.service: Start request repeated too quickly.
Jan 01 00:03:36 aaaa systemd[1]: phosphor-ipmi-kcs@ipmi-kcs3.service: Failed with result 'exit-code'.
Jan 01 00:05:01 aaaa systemd-coredump[210]: Process 191 (ipmid) of user 0 dumped core.
Jan 01 00:05:02 aaaa systemd[1]: phosphor-ipmi-host.service: Main process exited, code=dumped, status=11/SEGV
Jan 01 00:05:02 aaaa systemd[1]: phosphor-ipmi-host.service: Failed with result 'core-dump'.
Jan 01 00:05:03 aaaa systemd[1]: phosphor-ipmi-host.service: Scheduled restart job, restart counter is at 1.
Jan 01 00:05:05 aaaa ipmid[243]: Channel not configured so loading default.
Jan 01 00:05:05 aaaa ipmid[243]: Channel not configured so loading default.
Jan 01 00:05:05 aaaa ipmid[243]: Channel not configured so loading default.
Jan 01 00:05:05 aaaa ipmid[243]: Channel not configured so loading default.
Jan 01 00:05:06 aaaa ipmid[243]: Registering OEM:[0X00C2CF], Cmd:[0X04] for Manual Zone Control

root@aaaa:/etc/eip# systemctl status phosphor-ipmi-net@eip0.service | cat
* phosphor-ipmi-net@eip0.service - Network IPMI daemon
     Loaded: loaded (/lib/systemd/system/phosphor-ipmi-net@.service; enabled; vendor preset: disabled)
     Active: deactivating (stop-sigterm) since Thu 1970-01-01 00:15:44 UTC; 34s ago
TriggeredBy: * phosphor-ipmi-net@eip0.socket
   Main PID: 601 (netipmid)
      Tasks: 1 (limit: 2382)
     Memory: 1.1M
     CGroup: /system.slice/system-phosphor\x2dipmi\x2dnet.slice/phosphor-ipmi-net@eip0.service
             `- 601 /usr/bin/netipmid -c eip0
Jan 01 00:15:40 aaaa systemd[1]: Started Network IPMI daemon.
Jan 01 00:15:43 aaaa systemd[1]: phosphor-ipmi-net@eip0.service: Main process exited, code=dumped, status=11/SEGV
Jan 01 00:15:43 aaaa systemd[1]: phosphor-ipmi-net@eip0.service: Failed with result 'core-dump'.
Jan 01 00:15:44 aaaa systemd[1]: Stopping Network IPMI daemon...

root@aaaa:/etc/eip# systemctl status phosphor-ipmi-host
* phosphor-ipmi-host.service - Phosphor Inband IPMI
     Loaded: loaded (/lib/systemd/system/phosphor-ipmi-host.service; enabled; vendor preset: enabled)
    Drop-In: /lib/systemd/system/phosphor-ipmi-host.service.d
             `-10-override.conf
     Active: active (running) since Thu 1970-01-01 00:16:36 UTC; 1min 1s ago
   Main PID: 631 (ipmid)
      Tasks: 1 (limit: 2382)
     Memory: 1.7M
     CGroup: /system.slice/phosphor-ipmi-host.service
             `- 631 ipmid
Jan 01 00:16:35 aaaa ipmid[631]: Channel not configured so loading default.
Jan 01 00:16:35 aaaa ipmid[631]: Registering OEM:[0X00C2CF], Cmd:[0X04] for Manual Zone Control
Jan 01 00:16:36 aaaa systemd[1]: Started Phosphor Inband IPMI.

root@aaaa:/etc/eip# systemctl status phosphor-ipmi-host
* phosphor-ipmi-host.service - Phosphor Inband IPMI
     Loaded: loaded (/lib/systemd/system/phosphor-ipmi-host.service; enabled; vendor preset: enabled)
    Drop-In: /lib/systemd/system/phosphor-ipmi-host.service.d
             `-10-override.conf
     Active: activating (auto-restart) (Result: core-dump) since Thu 1970-01-01 00:17:43 UTC; 1s ago
    Process: 631 ExecStart=/usr/bin/env ipmid (code=dumped, signal=SEGV)
   Main PID: 631 (code=dumped, signal=SEGV)

vmauery, 2023-06-15:
Looks like you don't have the system configured correctly. ipmid is loading the default channel config which does not match your system config. This probably should not cause a segmentation fault/core dump, but you should make sure that your channel config file matches the channels that are available in your BMC system.

akraj19, 2023-06-29:
@vmauery  I have configured channel_config.json. Anything else we need to do. Also if i disabled net-ipmi package then ipmid is working fine

akraj19, 2023-07-05:
@vmauery See below logs after adding the channel config:
root@wer:~#
root@wer:~# journalctl | grep ipmi
July 01 00:00:22 wer systemd[1]: /lib/systemd/system/phosphor-ipmi-net@.socket:3: Invalid interface name, ignoring: sys-subsystem-net-devices-%i.device
July 01 00:00:22 wer systemd[1]: Created slice Slice /system/phosphor-ipmi-kcs.
July 01 00:00:22 wer systemd[1]: Created slice Slice /system/phosphor-ipmi-net.
July 01 00:00:23 wer systemd[1]: Listening on phosphor-ipmi-net@eip1.socket.
July 01 00:00:54 wer systemd[1]: phosphor-ipmi-host.service: Unexpected error response from GetNameOwner(): Connection terminated
July 01 00:01:27 wer ipmid[238]: JSON file not found
July 01 00:01:27 wer ipmid[238]: JSON file not found
July 01 00:01:27 wer ipmid[238]: Registering OEM:[0X00C2CF], Cmd:[0X04] for Manual Zone Control
July 01 00:01:27 wer kcsbridged-ipmi-kcs3[258]: FAILED: open `/dev/ipmi-kcs3`: No such file or directory
July 01 00:01:27 wer systemd[1]: phosphor-ipmi-kcs@ipmi-kcs3.service: Main process exited, code=exited, status=1/FAILURE
July 01 00:01:27 wer systemd[1]: phosphor-ipmi-kcs@ipmi-kcs3.service: Failed with result 'exit-code'.
July 01 00:01:28 wer systemd[1]: phosphor-ipmi-kcs@ipmi-kcs3.service: Scheduled restart job, restart counter is at 1.
July 01 00:01:28 wer kcsbridged-ipmi-kcs3[267]: FAILED: open `/dev/ipmi-kcs3`: No such file or directory
July 01 00:01:28 wer systemd[1]: phosphor-ipmi-kcs@ipmi-kcs3.service: Main process exited, code=exited, status=1/FAILURE
July 01 00:01:28 wer systemd[1]: phosphor-ipmi-kcs@ipmi-kcs3.service: Failed with result 'exit-code'.
July 01 00:01:30 wer systemd[1]: phosphor-ipmi-kcs@ipmi-kcs3.service: Scheduled restart job, restart counter is at 2.
July 01 00:01:30 wer systemd[1]: phosphor-ipmi-kcs@ipmi-kcs3.service: Start request repeated too quickly.
July 01 00:01:30 wer systemd[1]: phosphor-ipmi-kcs@ipmi-kcs3.service: Failed with result 'exit-code'.
July 01 00:01:31 wer systemd-coredump[263]: Process 238 (ipmid) of user 0 dumped core.
July 01 00:01:31 wer systemd-coredump[265]: Process 259 (netipmid) of user 0 dumped core.
July 01 00:01:31 wer systemd[1]: phosphor-ipmi-host.service: Main process exited, code=dumped, status=11/SEGV
July 01 00:01:31 wer systemd[1]: phosphor-ipmi-host.service: Failed with result 'core-dump'.
July 01 00:01:31 wer systemd[1]: phosphor-ipmi-net@eip1.service: Main process exited, code=dumped, status=11/SEGV
July 01 00:01:31 wer systemd[1]: phosphor-ipmi-net@eip1.service: Failed with result 'core-dump'.
July 01 00:01:32 wer systemd[1]: phosphor-ipmi-host.service: Scheduled restart job, restart counter is at 1.
July 01 00:01:32 wer systemd[1]: phosphor-ipmi-net@eip1.service: Scheduled restart job, restart counter is at 1.
July 01 00:01:32 wer ipmid[283]: Registering OEM:[0X00C2CF], Cmd:[0X04] for Manual Zone Control
July 01 00:01:35 wer systemd-coredump[288]: Process 284 (netipmid) of user 0 dumped core.
July 01 00:01:35 wer systemd[1]: phosphor-ipmi-net@eip1.service: Main process exited, code=dumped, status=11/SEGV
July 01 00:01:35 wer systemd[1]: phosphor-ipmi-net@eip1.service: Failed with result 'core-dump'.
July 01 00:01:35 wer systemd-coredump[286]: Process 283 (ipmid) of user 0 dumped core.
July 01 00:01:35 wer systemd[1]: phosphor-ipmi-host.service: Main process exited, code=dumped, status=11/SEGV
July 01 00:01:35 wer systemd[1]: phosphor-ipmi-host.service: Failed with result 'core-dump'.
July 01 00:01:36 wer systemd[1]: phosphor-ipmi-net@eip1.service: Scheduled restart job, restart counter is at 2.
July 01 00:01:36 wer systemd[1]: phosphor-ipmi-host.service: Start request repeated too quickly.
July 01 00:01:36 wer systemd[1]: phosphor-ipmi-host.service: Failed with result 'core-dump'.
July 01 00:01:36 wer systemd[1]: phosphor-ipmi-net@eip1.service: Job phosphor-ipmi-net@eip1.service/start failed with result 'dependency'.

siricillasridhar, 2023-12-09:
I'm too facing the same issue with 2.12.0 version. May I know how is this resolved?