openbmc/openbmc #3906: Openbmc not showing sd card

aamirrad, 2023-03-02:
**Note on Bugs**
I am running bmc image (details below) and trying to access the sd card attached to the board. However when i type in fdisk -l , it shows nothing. If i look into the dmesg, i see nothing about the sd, however i do see an error message about the usb which i am not sure if its related. Can you guys direct me if there is any configuration i need to do for the kernal to see the sd card?
root@BMC:~# dmesg | grep usb
[    2.086493] aspeed_vhub 1e6a0000.usb-vhub: Initialized virtual hub in USB2 mode
[    2.107681] udc 1e6a0000.usb-vhub:p1: failed to start g_mass_storage: -22
[    2.134496] udc 1e6a0000.usb-vhub:p2: failed to start g_mass_storage: -22
[    2.161222] udc 1e6a0000.usb-vhub:p3: failed to start g_mass_storage: -22
[    2.187821] udc 1e6a0000.usb-vhub:p4: failed to start g_mass_storage: -22
[    2.214421] udc 1e6a0000.usb-vhub:p5: failed to start g_mass_storage: -22

**Unexpected behavior you saw**
I expect the sd card to show up when i type in fdisk -l on the command line, however the output shows empty for this command.

**Expected behavior**
I expect the sd card to show up when i type in fdisk -l on the command line, however the output shows empty for this command.

**To Reproduce**
Steps to reproduce the behavior:
1. Login to the bmc module
2. Type in fdisk -l
3. You should see a list of the external sd card
4. It shows nothing.

**OpenBMC Information:**
 
root@BMC:~# uname -a
Linux BMC 6.0.10-115a0aa #1 Mon Nov 28 02:55:02 UTC 2022 armv6l GNU/Linux

root@BMC:~# cat /etc/os-release
ID=openbmc-phosphor
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="None"
VERSION_ID=none
VERSION_CODENAME="${DISTRO_CODENAME}"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) None"
BUILD_ID="20221206174257"
OPENBMC_TARGET_MACHINE="evb-ast2500"
EXTENDED_VERSION="None"

alandvbh, 2023-03-08:
Is there someone here to assist?

williamspatrick, 2023-03-09:
What kernel config did you end up with?  `zcat /proc/config.gz`

alandvbh, 2023-03-09:
#
# Automatically generated file; DO NOT EDIT.
# Linux/arm 6.0.8 Kernel Configuration
#
CONFIG_CC_VERSION_TEXT="arm-openbmc-linux-gnueabi-gcc (GCC) 12.2.0"
CONFIG_CC_IS_GCC=y
CONFIG_GCC_VERSION=120200
CONFIG_CLANG_VERSION=0
CONFIG_AS_IS_GNU=y
CONFIG_AS_VERSION=23900
CONFIG_LD_IS_BFD=y
CONFIG_LD_VERSION=23900
CONFIG_LLD_VERSION=0
CONFIG_CC_HAS_ASM_GOTO_OUTPUT=y
CONFIG_CC_HAS_ASM_INLINE=y
CONFIG_CC_HAS_NO_PROFILE_FN_ATTR=y
CONFIG_PAHOLE_VERSION=0
CONFIG_IRQ_WORK=y
CONFIG_BUILDTIME_TABLE_SORT=y
CONFIG_THREAD_INFO_IN_TASK=y

#
# General setup
#
CONFIG_BROKEN_ON_SMP=y
CONFIG_INIT_ENV_ARG_LIMIT=32
# CONFIG_COMPILE_TEST is not set
# CONFIG_WERROR is not set
CONFIG_LOCALVERSION="-1b16243"
CONFIG_LOCALVERSION_AUTO=y
CONFIG_BUILD_SALT=""
CONFIG_HAVE_KERNEL_GZIP=y
CONFIG_HAVE_KERNEL_LZMA=y
CONFIG_HAVE_KERNEL_XZ=y
CONFIG_HAVE_KERNEL_LZO=y
CONFIG_HAVE_KERNEL_LZ4=y
# CONFIG_KERNEL_GZIP is not set
# CONFIG_KERNEL_LZMA is not set
CONFIG_KERNEL_XZ=y
# CONFIG_KERNEL_LZO is not set
# CONFIG_KERNEL_LZ4 is not set
CONFIG_DEFAULT_INIT=""
CONFIG_DEFAULT_HOSTNAME="(none)"
CONFIG_SYSVIPC=y
CONFIG_SYSVIPC_SYSCTL=y
# CONFIG_POSIX_MQUEUE is not set
# CONFIG_WATCH_QUEUE is not set
CONFIG_CROSS_MEMORY_ATTACH=y
# CONFIG_USELIB is not set
# CONFIG_AUDIT is not set
CONFIG_HAVE_ARCH_AUDITSYSCALL=y

#
# IRQ subsystem
#
CONFIG_GENERIC_IRQ_PROBE=y
CONFIG_GENERIC_IRQ_SHOW=y
CONFIG_GENERIC_IRQ_SHOW_LEVEL=y
CONFIG_HARDIRQS_SW_RESEND=y
CONFIG_IRQ_DOMAIN=y
CONFIG_IRQ_FORCED_THREADING=y
CONFIG_SPARSE_IRQ=y
# CONFIG_GENERIC_IRQ_DEBUGFS is not set
# end of IRQ subsystem

CONFIG_GENERIC_IRQ_MULTI_HANDLER=y
CONFIG_GENERIC_CLOCKEVENTS=y

#
# Timers subsystem
[... 2673 lines cut ...]
#
# CONFIG_LEDS_AAT1290 is not set
# CONFIG_LEDS_AS3645A is not set
# CONFIG_LEDS_KTD2692 is not set
# CONFIG_LEDS_LM3601X is not set
# CONFIG_LEDS_RT4505 is not set
# CONFIG_LEDS_RT8515 is not set
# CONFIG_LEDS_SGM3140 is not set

#
# RGB LED drivers
#

#
# LED Triggers
#
CONFIG_LEDS_TRIGGERS=y
CONFIG_LEDS_TRIGGER_TIMER=y
# CONFIG_LEDS_TRIGGER_ONESHOT is not set
# CONFIG_LEDS_TRIGGER_MTD is not set
CONFIG_LEDS_TRIGGER_HEARTBEAT=y
# CONFIG_LEDS_TRIGGER_BACKLIGHT is not set
# CONFIG_LEDS_TRIGGER_CPU is not set
# CONFIG_LEDS_TRIGGER_ACTIVITY is not set
# CONFIG_LEDS_TRIGGER_GPIO is no

williamspatrick, 2023-03-09:
Next time please use a gist.github.com upload or at least escape your file with triple-backticks.  This was pretty impossible to look through.

```
# CONFIG_MMC is not set
```

At a minimum you're missing this.

williamspatrick, 2023-03-09:
You should look at the aspeed-g6/defconfig as a potential reference.

```
meta-aspeed/recipes-kernel/linux/linux-aspeed/aspeed-g6/defconfig:CONFIG_MMC=y
meta-aspeed/recipes-kernel/linux/linux-aspeed/aspeed-g6/defconfig:CONFIG_MMC_SDHCI=y
meta-aspeed/recipes-kernel/linux/linux-aspeed/aspeed-g6/defconfig:CONFIG_MMC_SDHCI_PLTFM=y
meta-aspeed/recipes-kernel/linux/linux-aspeed/aspeed-g6/defconfig:CONFIG_MMC_SDHCI_OF_ASPEED=y
```

closed 2023-03-09 by williamspatrick

alandvbh, 2023-03-09:
I added the following 4 lines into aspeed-g5/defconfig
CONFIG_MMC=y
CONFIG_MMC_SDHCI_OF_ASPEED=y
CONFIG_MMC_SDHCI_PLTFM=y
CONFIG_MMC_SDHCI=y

However the build fails due to the image being too large. Do i need to increase the image size somewhere for it to not fail?

(No title)
DEBUG: Executing python function do_generate_static
DEBUG: Executing shell function do_mk_static_nor_image
32768+0 records in
32768+0 records out
33554432 bytes (34 MB, 32 MiB) copied, 0.0345804 s, 970 MB/s
DEBUG: Shell function do_mk_static_nor_image finished
DEBUG: Considering file size=386977 name=/home/xyz/openbmc.git/trunk/build/evb-ast2500/tmp/deploy/images/evb-ast2500/u-boot.bin
DEBUG: Spanning start=0K end=512K
DEBUG: Compare needed=386977 available=524288 margin=137311
377+1 records in
377+1 records out
386977 bytes (387 kB, 378 KiB) copied, 0.000685095 s, 565 MB/s
DEBUG: Considering file size=4468996 name=/home/xyz/openbmc.git/trunk/build/evb-ast2500/tmp/deploy/images/evb-ast2500/fitImage-obmc-phosphor-initramfs-evb-ast2500-evb-ast2500
DEBUG: Spanning start=512K end=4864K
DEBUG: Compare needed=4468996 available=4456448 margin=-12548
ERROR: Image '/home/xyz/openbmc.git/trunk/build/evb-ast2500/tmp/deploy/images/evb-ast2500/fitImage-obmc-phosphor-initramfs-evb-ast2500-evb-ast2500' is too large!
DEBUG: Python function do_generate_static finished

williamspatrick, 2023-03-09:
You probably need to find something else to turn off. Changing the kernel partition size is not simple because you need to change DTS files also.

alandvbh, 2023-03-10:
Hi William. Thanks for the help. I can now see something in my dmesg, but still nothing in my fdisk -l. Can you please let me know if there is anything else I need to do before I can access this sd card? Sorry I am new to embeded  and not very familiar with these.

root@BMC:~# dmesg | grep -i sd
[    8.489566] sdhci: Secure Digital Host Controller Interface driver
[    8.495894] sdhci: Copyright(c) Pierre Ossman
[    8.508109] sdhci-pltfm: SDHCI platform and OF driver helper
[    8.518897] sdhci-aspeed 1e740100.sdhci: Phase control not supported for slot 0
[    8.526371] sdhci-aspeed 1e740100.sdhci: Configured for slot 0
[    8.573784] mmc0: SDHCI controller on 1e740100.sdhci [1e740100.sdhci] using ADMA

Below is output of zcat /proc/config.gz

 `<script src="https://gist.github.com/alandvbh/d36f52a5f6f4dc3bbe8dbc125ddcdc9e.js"></script>`

williamspatrick, 2023-03-10:
This is what a working AST2600 looks like:

```
[    5.506800] sdhci: Secure Digital Host Controller Interface driver
[    5.507132] sdhci: Copyright(c) Pierre Ossman
[    5.507380] sdhci-pltfm: SDHCI platform and OF driver helper
[    5.514812] sdhci-aspeed 1e750100.sdhci: Configured for slot 0
[    5.548977] mmc0: SDHCI controller on 1e750100.sdhci [1e750100.sdhci] using ADMA
[    5.588480] mmc0: new high speed SD card at address 4567
```

```
root@bletchley:~# ls /dev/mmc*
/dev/mmcblk0
root@bletchley:~# fdisk -l
Disk /dev/mmcblk0: 1024 MB, 1073741824 bytes, 2097152 sectors
32768 cylinders, 4 heads, 16 sectors/track
Units: sectors of 1 * 512 = 512 bytes

Disk /dev/mmcblk0 doesn't contain a valid partition table
```

This is an eMMC device and not an SD card, but they use the same technology.

I don't know what "Phase control not supported for slot 0" means.  I suggest you check out the kernel for that.  It is possible that the AST2500 code isn't 100% complete.

The aspeed-g6.dts has the following:
```
			emmc_controller: sdc@1e750000 {
				compatible = "aspeed,ast2600-sd-controller";
				reg = <0x1e750000 0x100>;
				#address-cells = <1>;
				#size-cells = <1>;
				ranges = <0 0x1e750000 0x10000>;
				clocks = <&syscon ASPEED_CLK_GATE_EMMCCLK>;
				status = "disabled";

				emmc: sdhci@1e750100 {
					compatible = "aspeed,ast2600-sdhci";
					reg = <0x100 0x100>;
					sdhci,auto-cmd12;
					interrupts = <GIC_SPI 15 IRQ_TYPE_LEVEL_HIGH>;
					clocks = <&syscon ASPEED_CLK_EMMC>;
					pinctrl-names = "default";
					pinctrl-0 = <&pinctrl_emmc_default>;
				};
			};
```

Then in my system dts I have the following:

```
&emmc_controller {
	status = "okay";
};

&emmc {
	status = "okay";
};
````

The aspeed-g5.dtsi specifies them different, but similarly "default-disabled":

```
			sdmmc: sd-controller@1e740000 {
				compatible = "aspeed,ast2500-sd-controller";
				reg = <0x1e740000 0x100>;
				#address-cells = <1>;
				#size-cells = <1>;
				ranges = <0 0x1e740000 0x10000>;
				clocks = <&syscon ASPEED_CLK_GATE_SDCLK>;
				status = "disabled";

				sdhci0: sdhci@100 {
					compatible = "aspeed,ast2500-sdhci";
					reg = <0x100 0x100>;
					interrupts = <26>;
					sdhci,auto-cmd12;
					clocks = <&syscon ASPEED_CLK_SDIO>;
					status = "disabled";
				};

				sdhci1: sdhci@200 {
					compatible = "aspeed,ast2500-sdhci";
					reg = <0x200 0x100>;
					interrupts = <26>;
					sdhci,auto-cmd12;
					clocks = <&syscon ASPEED_CLK_SDIO>;
					status = "disabled";
				};
			};
```

Looking at `aspeed-ast2500-evb.dst`, it looks like `sdmmc` and `sdhci0` are enabled by default.  This might be the on-board mmc and not the SD slot.  You might try enabling `sdhci1`.

alandvbh, 2023-03-13:
I did enable sdhci1, but nothing changed. fdisk -l still shows blank and dmesg | grep -i sd still shows the same messages as when sdhci1 was disabled. Anything else i should try?

root@BMC:~# dmesg | grep -i sd
[ 8.489566] sdhci: Secure Digital Host Controller Interface driver
[ 8.495894] sdhci: Copyright(c) Pierre Ossman
[ 8.508109] sdhci-pltfm: SDHCI platform and OF driver helper
[ 8.518897] sdhci-aspeed 1e740100.sdhci: Phase control not supported for slot 0
[ 8.526371] sdhci-aspeed 1e740100.sdhci: Configured for slot 0
[ 8.573784] mmc0: SDHCI controller on 1e740100.sdhci [1e740100.sdhci] using ADMA

williamspatrick, 2023-03-13:
`[ 8.526371] sdhci-aspeed 1e740100.sdhci: Configured for slot 0`  <--- pretty sure you need to figure out how to get that to say "slot 1".

alandvbh, 2023-03-21:
I have both slots (0 and 1) showing in dmesg now, but still nothing under fdisk -l and dev/mmc*. Anything else you I need to do to get access to this sd card?

root@BMC:~# dmesg | grep -i sdhci
[    8.496251] sdhci: Secure Digital Host Controller Interface driver
[    8.502572] sdhci: Copyright(c) Pierre Ossman
[    8.515510] sdhci-pltfm: SDHCI platform and OF driver helper
[    8.535967] sdhci-aspeed 1e740100.sdhci: Configured for slot 0
[    8.550735] sdhci-aspeed 1e740200.sdhci: Configured for slot 1
[    8.631650] mmc1: SDHCI controller on 1e740200.sdhci [1e740200.sdhci] using ADMA
[    8.639425] mmc0: SDHCI controller on 1e740100.sdhci [1e740100.sdhci] using ADMA

root@BMC:~# df -h
Filesystem                Size      Used Available Use% Mounted on
dev                     477.6M         0    477.6M   0% /dev
tmpfs                   494.7M     11.5M    483.2M   2% /run
/dev/mtdblock4           17.1M     17.1M         0 100% /run/initramfs/ro
/dev/mtdblock5            4.0M      1.3M      2.7M  32% /run/initramfs/rw
cow                       4.0M      1.3M      2.7M  32% /
tmpfs                   494.7M      8.0K    494.7M   0% /dev/shm
tmpfs                   494.7M    108.0K    494.6M   0% /tmp
tmpfs                   494.7M      8.0K    494.7M   0% /var/volatile

williamspatrick, 2023-03-21:
@shenki Do you have any pointers on debugging mmc further?

shenki, 2023-03-22:
The SD support on ast2500 is not regularly tested.

Can you share your device tree?

Are you able to test the a similar setup on an AST2500 EVB?

alandvbh, 2023-03-27:
got it to work. had to make some changes in the dts file in addition to what was suggested. thanks for the help.