openbmc/openbmc #3907: Build error on Ubuntu 22.0.4 LTS

david6076, 2023-03-03:
I observe python script failures when running this cmd:  bitbake core-minimal-image

**To Reproduce**
Steps to reproduce the behavior:
1. git clone https://github.com/openbmc/openbmc  / install all required libs per docs
2. . setup evb-ast2600
3. bitbake core-minimal-image
4. See error output below
I am running on Ubuntu 22.0.4 LTS

dj@Stimpy:~/obmc/openbmc$ . setup evb-ast2600 /media/dj/openBMC/build
Machine evb-ast2600 found in meta-evb/meta-evb-aspeed/meta-evb-ast2600
Common targets are:

     obmc-phosphor-image: Includes OpenBMC Phosphor userspace and Web UI

     core-image-minimal: A small image just capable of allowing a device to boot

     core-image-full-cmdline: A small image with more Linux functionality
                              installed, including a ssh server.
dj@Stimpy:/media/dj/openBMC/build$ bitbake core-image-minimal
Loading cache: 100% |##########################################################################################| Time: 0:00:07
Loaded 4171 entries from dependency cache.
Parsing recipes: 100% |########################################################################################| Time: 0:00:02
Parsing of 2682 .bb files complete (2681 cached, 1 parsed). 4172 targets, 398 skipped, 0 masked, 0 errors.
WARNING: No bb files in default matched BBFILE_PATTERN_meta-evb-ast2600 '^/home/dj/obmc/openbmc/meta-evb/meta-evb-aspeed/meta-evb-ast2600/'
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "2.2.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-22.04"
TARGET_SYS           = "arm-openbmc-linux-gnueabi"
MACHINE              = "evb-ast2600"
DISTRO               = "openbmc-phosphor"
DISTRO_VERSION       = "nodistro.0"
TUNE_FEATURES        = "arm armv7a vfp vfpv4d16 callconvention-hard"
TARGET_FPU           = "hard"
meta                 
meta-poky            
meta-oe              
meta-networking      
meta-python          
meta-phosphor        
meta-aspeed          
meta-evb-ast2600     = "master:ae3048554faba38f5df5c1b17df741f90ff0eeab"

Initialising tasks: 100% |#####################################################################################| Time: 0:00:06
Sstate summary: Wanted 822 Local 0 Mirrors 0 Missed 822 Current 0 (0% match, 0% complete)
NOTE: Executing Tasks
ERROR: texinfo-dummy-native-1.0-r0 do_fetch: Execution of event handler 'defaultsstate_eventhandler' failed
Traceback (most recent call last):
  File "/home/dj/obmc/openbmc/meta/lib/oe/sstatesig.py", line 222, in SignatureGeneratorOEBasicHash.dump_sigtask(fn='/home/dj/obmc/openbmc/meta/recipes-extended/texinfo-dummy-native/texinfo-dummy-native.bb', task='do_fetch', stampbase='/media/dj/openBMC/build/sstate-cache/69/85/sstate:texinfo-dummy-native::1.0:r0::10:698522da0d79b44a918223330480e440edf60b5e7f7604feda3574060f3671ac_fetch.tar.zst.siginfo', runtime='customfile:/media/dj/openBMC/build/tmp/stamps/x86_64-linux/texinfo-dummy-native/1.0-r0'):
                 return
    >        super(bb.siggen.SignatureGeneratorBasicHash, self).dump_sigtask(fn, task, stampbase, runtime)
     
  File "/home/dj/obmc/openbmc/poky/bitbake/lib/bb/siggen.py", line 459, in SignatureGeneratorOEBasicHash.dump_sigtask(mcfn='/home/dj/obmc/openbmc/meta/recipes-extended/texinfo-dummy-native/texinfo-dummy-native.bb', task='do_fetch', stampbase='/media/dj/openBMC/build/sstate-cache/69/85/sstate:texinfo-dummy-native::1.0:r0::10:698522da0d79b44a918223330480e440edf60b5e7f7604feda3574060f3671ac_fetch.tar.zst.siginfo', runtime='customfile:/media/dj/openBMC/build/tmp/stamps/x86_64-linux/texinfo-dummy-native/1.0-r0'):
                     pass
    >            raise err
     
  File "/home/dj/obmc/openbmc/poky/bitbake/lib/bb/siggen.py", line 453, in SignatureGeneratorOEBasicHash.dump_sigtask(mcfn='/home/dj/obmc/openbmc/meta/recipes-extended/texinfo-dummy-native/texinfo-dummy-native.bb', task='do_fetch', stampbase='/media/dj/openBMC/build/sstate-cache/69/85/sstate:texinfo-dummy-native::1.0:r0::10:698522da0d79b44a918223330480e440edf60b5e7f7604feda3574060f3671ac_fetch.tar.zst.siginfo', runtime='customfile:/media/dj/openBMC/build/tmp/stamps/x86_64-linux/texinfo-dummy-native/1.0-r0'):
                 os.chmod(tmpfile, 0o664)
    >            bb.utils.rename(tmpfile, sigfile)
             except (OSError, IOError) as err:
  File "/home/dj/obmc/openbmc/poky/bitbake/lib/bb/utils.py", line 1789, in rename(src='/media/dj/openBMC/build/sstate-cache/69/85/sigtask.FaUymt24ETsaGEn1cg2K6hdomxiz', dst='/media/dj/openBMC/build/sstate-cache/69/85/sstate:texinfo-dummy-native::1.0:r0::10:698522da0d79b44a918223330480e440edf60b5e7f7604feda3574060f3671ac_fetch.tar.zst.siginfo'):
             else:
    >            raise err
     
  File "/home/dj/obmc/openbmc/poky/bitbake/lib/bb/utils.py", line 1783, in rename(src='/media/dj/openBMC/build/sstate-cache/69/85/sigtask.FaUymt24ETsaGEn1cg2K6hdomxiz', dst='/media/dj/openBMC/build/sstate-cache/69/85/sstate:texinfo-dummy-native::1.0:r0::10:698522da0d79b44a918223330480e440edf60b5e7f7604feda3574060f3671ac_fetch.tar.zst.siginfo'):
         try:
[... 231 lines cut ...]
    bb.parse.siggen.dump_sigtask(mcfn, task, outfile, "customfile:" + referencestamp)
  File "/home/dj/obmc/openbmc/meta/lib/oe/sstatesig.py", line 222, in dump_sigtask
    super(bb.siggen.SignatureGeneratorBasicHash, self).dump_sigtask(fn, task, stampbase, runtime)
  File "/home/dj/obmc/openbmc/poky/bitbake/lib/bb/siggen.py", line 459, in dump_sigtask
    raise err
  File "/home/dj/obmc/openbmc/poky/bitbake/lib/bb/siggen.py", line 453, in dump_sigtask
    bb.utils.rename(tmpfile, sigfile)
  File "/home/dj/obmc/openbmc/poky/bitbake/lib/bb/utils.py", line 1789, in rename
    raise err
  File "/home/dj/obmc/openbmc/poky/bitbake/lib/bb/utils.py", line 1783, in rename
    os.rename(src, dst)
OSError: [Errno 22] Invalid argument: '/media/dj/openBMC/build/sstate-cache/8d/22/sigtask.IYcX9mpVjOsbvBVP7P7ev8_hbip_' -> '/media/dj/openBMC/build/sstate-cache/8d/22/sstate:m4-native::1.4.19:r0::10:8d2279fc67560b643203776535546ad7f9a8490c739a6ceceee9078c9afcc306_fetch.tar.zst.siginfo'

ERROR: Task (/home/dj/obmc/openbmc/meta/recipes-devtools/m4/m4-native_1.4.19.bb:do_fetch) failed with exit code '1'
NOTE: Tasks Summary: Attempted 5 tasks of which 0 didn't need to be rerun and 5 failed.

Summary: 5 tasks failed:
  /home/dj/obmc/openbmc/meta/recipes-extended/texinfo-dummy-native/texinfo-dummy-native.bb:do_fetch
  /home/dj/obmc/openbmc/meta/recipes-core/gettext/gettext-minimal-native_0.21.1.bb:do_fetch
  /home/dj/obmc/openbmc/meta/recipes-devtools/quilt/quilt-native_0.67.bb:do_fetch
  virtual:native:/home/dj/obmc/openbmc/meta/recipes-devtools/gnu-config/gnu-config_git.bb:do_fetch
  /home/dj/obmc/openbmc/meta/recipes-devtools/m4/m4-native_1.4.19.bb:do_fetch
Summary: There was 1 WARNING message.
Summary: There were 15 ERROR messages, returning a non-zero exit code.
dj@Stimpy:/media/dj/openBMC/build$

williamspatrick, 2023-03-03:
> /media/dj/openBMC/build

What kind of file system is this?  You wouldn't think that `rename` syscalls would fail, but ... are you trying to use something like FAT or NTFS which doesn't like `:` in file names?

david6076, 2023-03-03:
Crap! It is ntfs.

closed 2023-03-03 by williamspatrick