openbmc/openbmc #3909: OpenBMC no longer builds on ppc64le - obmc-phosphor-image.bb:do_image_wic

geissonator, 2023-03-20:
**Unexpected behavior you saw**
We used to have a CI node that was ppc64le and made sure OpenBMC master continued to build on that architecture. Unfortunately we lost that node last year. I'd like to use some of our ppc64le hardware (POWER9) to build OpenBMC but alas it no longer can build ppc64le. One issue was with QEMU (https://gitlab.com/qemu-project/qemu/-/issues/1528). While we wait for that fix to make it into QEMU, I did a workaround to avoid needing it. After that I can get all the way to the end of the build, but it fails here:

```
| Traceback (most recent call last):
|   File "/home/fspcibld/284637575/openbmc/scripts/wic", line 547, in <module>
|     sys.exit(main(sys.argv[1:]))
|              ^^^^^^^^^^^^^^^^^^
|   File "/home/fspcibld/284637575/openbmc/scripts/wic", line 542, in main
|     return hlp.invoke_subcommand(args, parser, hlp.wic_help_usage, subcommands)
|            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
|   File "/home/fspcibld/284637575/openbmc/poky/scripts/lib/wic/help.py", line 83, in invoke_subcommand
|     subcmd[0](args, usage)
|   File "/home/fspcibld/284637575/openbmc/scripts/wic", line 222, in wic_create_subcommand
|     engine.wic_create(wks_file, rootfs_dir, bootimg_dir, kernel_dir,
|   File "/home/fspcibld/284637575/openbmc/poky/scripts/lib/wic/engine.py", line 190, in wic_create
|     plugin.do_create()
|   File "/home/fspcibld/284637575/openbmc/poky/scripts/lib/wic/plugins/imager/direct.py", line 99, in do_create
|     self.assemble()
|   File "/home/fspcibld/284637575/openbmc/poky/scripts/lib/wic/plugins/imager/direct.py", line 194, in assemble
|     self._image.assemble()
|   File "/home/fspcibld/284637575/openbmc/poky/scripts/lib/wic/plugins/imager/direct.py", line 624, in assemble
|     sparse_copy(source, self.path, seek=part.start * self.sector_size)
|   File "/home/fspcibld/284637575/openbmc/poky/scripts/lib/wic/filemap.py", line 540, in sparse_copy
|     for first, last in fmap.get_mapped_ranges(0, fmap.blocks_cnt):
|   File "/home/fspcibld/284637575/openbmc/poky/scripts/lib/wic/filemap.py", line 458, in get_mapped_ranges
|     first_prev, last_prev = next(iterator)
|                             ^^^^^^^^^^^^^^
|   File "/home/fspcibld/284637575/openbmc/poky/scripts/lib/wic/filemap.py", line 440, in _do_get_mapped_ranges
|     assert extent_len % self.block_size == 0
|            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
| AssertionError
| WARNING: exit code 1 from a shell command.
ERROR: Task (/home/fspcibld/284637575/openbmc/meta-phosphor/recipes-phosphor/images/obmc-phosphor-image.bb:do_image_wic) failed with exit code '1'
NOTE: Tasks Summary: Attempted 5214 tasks of which 5213 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/fspcibld/284637575/openbmc/meta-phosphor/recipes-phosphor/images/obmc-phosphor-image.bb:do_image_wic

```

**Expected behavior**
Expected that OpenBMC will build on ppc64le

**To Reproduce**
Steps to reproduce the behavior:
1. Install RH8 on your P9 ppc64le machine
2. Utilize the build-setup.sh script to generate a build docker container
3. Do a bitbake of the p10bmc image within the docker container

**OpenBMC Information:**
 - Building most recent master, hash 6dce5383f8fb50619796e72ef3cf5e71fe4519c6

**Additional context**
I added a bit of debug to the failing python script, I don't really understand what it's checking but the assert is obviously catching something here:

```
| DEBUG: Python function extend_recipe_sysroot finished
| DEBUG: Executing python function set_image_size
| DEBUG: 197444.000000 = 151880 * 1.300000
| DEBUG: 197444.000000 = max(197444.000000, 65536)[197444.000000] + 0
| DEBUG: 197444.000000 = int(197444.000000)
| DEBUG: 197444 = aligned(197444)
| DEBUG: returning 197444
| DEBUG: Python function set_image_size finished
| Var <EXTRA_USERS_PARAMS:pn-obmc-phosphor-image>:1: DeprecationWarning: invalid escape sequence '\$'
| Var <EXTRA_USERS_PARAMS>:1: DeprecationWarning: invalid escape sequence '\$'
| DEBUG: Executing shell function do_image_wic
| INFO: Creating image(s)...
| 
| extend_start and block_size(0, 65536)
| extend_len and block_size(65536, 65536)
| extend_start and block_size(0, 65536)
| extend_len and block_size(65536, 65536)
| extend_start and block_size(0, 65536)
| extend_len and block_size(274432, 65536)
<ASSERT>

```

williamspatrick, 2023-03-20:
My first guess would be something related to 64K page size on Power when most other architectures use 4K by default.

geissonator, 2023-03-20:
> My first guess would be something related to 64K page size on Power when most other architectures use 4K by default.

Good guess! Hard coding the script to make the block size 4096 allowed the build to complete.

geissonator, 2023-03-21:
Submitted an upstream bug via https://bugzilla.yoctoproject.org/show_bug.cgi?id=15075

geissonator, 2023-03-28:
Fixed in upstream via https://lists.openembedded.org/g/openembedded-core/message/179079

closed 2023-03-28 by geissonator