openbmc/openbmc #3910: Invalid username or password

toravishivraj, 2023-03-21:
Hi all,
I'm build OpenBMC image for our custom h/w on Ubuntu 18.04 machine, minimal image in working.
i added in local config 
 IMAGE_INSTALL_append +=" bmcweb"
CORE_IMAGE_EXTRA_INSTALL +="webui-vue"
CORE_IMAGE_EXTRA_INSTALL +="systemd"
VIRTUAL_RUNTIME_init_manager="systemd"
VIRTUAL_RUNTIME_init_manager="systemd-compat-units"
 
in obmc-phosphor-Image.bb
OBMC_IMAGE_EXTRA_INSTALL_${MACHINE}+="webui-vue"

in packagegroup-obmc-apps.bb
RDEPENDS_${PN}-inventory_${MACHINE}+="webui-vue"

**### _Problem_**
1) it doesn't generate /var/log/redfish directory,why?(after adding /var/log/redfish bmcweb start)
2) i am using username 'root' and password '0penBmc' for Ui log in but it shows Invalid username or password 
response 'unauthorized' Error : Request failed with status code 401.?
3) usr/share/www/ contain css, img, js, favicon, index in host image but target image contain redfish, styles, DMTF_Redfish_logo_2017, highlight.pack why? (when i add manually  css, img, js, favicon, index in target image than login page opens but with 2) problem arrive)
4) also tryed with $curl -k -H 'X-Auth-Token: $token" https://${bmc}/xyz/opebmc_project/list---> unauthorized?

please Help.

Regard's
Shivraj T.

williamspatrick, 2023-03-21:
Please do not ignore the issue templates. I have no idea what you are building, what version of OpenBMC you are building from, etc. 

`IMAGE_INSTALL_append +=" bmcweb"` this is an older Yocto syntax which hasn’t worked for over a year now. Yocto complains about this, so the fact you didn’t get a warning tells me you are trying to build from an old release which is not supported. 

> minimal image in working

What do you mean by “minimal image”?  What did you attempt to build?  Depending on what you are attempting to build, this documentation might be helpful. 

https://github.com/openbmc/openbmc/commit/ec4935823527d2db4a279ac2f17c184a793f6e16

toravishivraj, 2023-03-21:
Hi,
Thank you for reply
I will complete issue template
For openBmc version 
 #cat /etc/os-release
ID= poky
Name= poky
VERSION= yocto-3.1.23
VERSION _ID=poky yocto -3.1.23
DISTRO_CODENAME=dunfell

Please tell new yocto syntax for bmcweb
Minimal image means console base image (basic image) i am building image for beagle bone black.

toravishivraj, 2023-03-21:
[image]

williamspatrick, 2023-03-21:
> Hi, Thank you for reply I will complete issue template For openBmc version #cat /etc/os-release ID= poky Name= poky VERSION= yocto-3.1.23 VERSION _ID=poky yocto -3.1.23 DISTRO_CODENAME=dunfell [...]

This is not OpenBMC and it is not recent code. You are somehow building a raw poky image. You’ve still not told me what you are trying to build.

closed 2023-03-21 by williamspatrick

toravishivraj, 2023-03-28:
Hi william,

Thank you for your replay,
my questions are, 
 1) Any documents to Build a openBMC image for Beagle Bone Black?
 2) How to select BBB hardware means e.g. like ". setup romulus"?
 3) Which branch will select for BBB like master or dunfell ?

Regard's
Shivraj T.

mdmillerii, 2023-03-28:
Like the lightly tested rpi it's probably best to start with a working beaglebone yocto image build and add the phosphor layers on top.   I don't expect that the bbb has dedicated host interfaces like lpc, bt, espi slave but you can learn how the stack is put together and might have i2c slave or USB device to expose media or host network device to explore those.

mdmillerii, 2023-03-28:
This community creates branches when we merge past upstream releases but generally only develop on the latest upstream.   Occasionally there will be security backports but you should not rely on them.

williamspatrick, 2023-03-29:
> Any documents to Build a openBMC image for Beagle Bone Black?

No one has attempted this and documented it, as far as I know.

> How to select BBB hardware means e.g. like ". setup romulus"?

We would need to pull in the meta-layer from upstream that supports this hardware.  Similar to how we have meta-aspeed for their BMC chips and meta-raspberrypi.

Once that is done you'd probably want to look at meta-evb and see how RPi4 is supported there.  You would need to mimic that.

> Which branch will select for BBB like master or dunfell ?

Most developers on the project work off the master branch and not older releases.

---

What is it you're trying to accomplish by putting OpenBMC onto a BBB?  Most of the code is intended to be useful for managing a connected server (which is what a BMC is) and a BBB doesn't have the appropriate hardware to be able to do that.  The RPi4 "port" is mostly as a playground for people that want to play with the code on some real hardware (vs QEMU or buying an EVB from Aspeed/Nuvoton), but it is currently not much more useful than that.