openbmc/openbmc #3911: openbmc# . setup evb-ast2600 Your RubyGems version (2.7.6) has a bug that prevents `required_ruby_version` from working for Bundler. Any scripts that use `gem install bundler` will break as soon as Bundler drops support for your Ruby version. Please upgrade RubyGems to avoid future breakage and silence this warning by running `gem update --system 3.2.3` Don't run Bundler as root. Bundler can ask for sudo if it is needed, and installing your bundle as root will break this application for all non-root users on this machine. The Gemfile specifies no dependencies Bundle complete! 0 Gemfile dependencies, 1 gem now installed. Use `bundle info [gemname]` to see where a bundled gem is installed. Your RubyGems version (2.7.6) has a bug that prevents `required_ruby_version` from working for Bundler. Any scripts that use `gem install bundler` will break as soon as Bundler drops support for your Ruby version. Please upgrade RubyGems to avoid future breakage and silence this warning by running `gem update --system 3.2.3`

max0604, 2023-03-22:
# STOP! READ BELOW.

We do not field questions or feature requests by Github Issues!

Questions or feature requests should be discussed on the mailing list or
Discord. Please see https://github.com/openbmc/openbmc#contact for info
on how to contact us.

Github Issues in openbmc/openbmc are for bugs only!

max0604, 2023-03-22:
openbmc$ gem --version
3.3.3

edtanous, 2023-04-27:
Retaining original title text for historical purposes, but making it smaller to be less annoying in the bug list.
```
 Any scripts that use `gem install bundler` will break as soon as Bundler drops support for your Ruby version. Please upgrade RubyGems to avoid future breakage and silence this warning by running `gem update --system 3.2.3` Don't run Bundler as root. Bundler can ask for sudo if it is needed, and installing your bundle as root will break this application for all non-root users on this machine. The Gemfile specifies no dependencies Bundle complete! 0 Gemfile dependencies, 1 gem now installed. Use `bundle info [gemname]` to see where a bundled gem is installed. Your RubyGems version (2.7.6) has a bug that prevents `required_ruby_version` from working for Bundler. Any scripts that use `gem install bundler` will break as soon as Bundler drops support for your Ruby version. Please upgrade RubyGems to avoid future breakage and silence this warning by running `gem update --system 3.2.3`
 ```
 
 Openbmc doesn't use ruby so far as I'm aware, so this appears to be an automated bug that's a false positive.  As was commented, please start discussions about these kinds of thing on the mailing list.  To my knowledge, original commented hasn't responded.  Closing.

closed 2023-04-27 by edtanous