openbmc/openbmc #3913: Issue replacing /etc/fstab file

AlexD202, 2023-03-28:
**Unexpected behavior you saw**
I am trying to replace the default /etc/fstab file with a new one. I have tried many different methods, but all of them fail to replace the file with my new file

**Expected behavior**
the new fstab should replace the default fstab under /etc/ directory.

**To Reproduce**
Steps to reproduce the behavior:
1. Check out the code from the git
2. modify poky/meta/recipes-core/base-files/base-files/fstab **OR** meta-phosphor/recipes-core/base-files/base-files_%.bbappend to include a new fstab file.
3. Run . setup evb-ast2500
4. Run bitbake -c clean base-files -f
5. Run bitbake -c compile base-files -f
6. Run  bitbake obmc-phosphor-image
7. Go to the ./build/evb-ast2500/tmp/work/evb_ast2500-openbmc-linux-gnueabi/obmc-phosphor-image/1.0-r0/rootfs/etc/ folder. The fstab there has been updated with the new content
8. scp the obmc-phosphor-image-evb-ast2500-*.static.mtd.tar to the module and use the following command to load the image busctl set-property xyz.openbmc_project.Software.BMC.Updater /xyz/openbmc_project/software/<ID> xyz.openbmc_project.Software.Activation RequestedActivation s  xyz.openbmc_project.Software.Activation.RequestedActivations.Active
9. Reboot the module
10. After module comes back up, I expect the /etc/fstab to be the new file, however it is still the old one. I see that the fstab under /run/initramfs/ro/etc/fstab is changed to the new one.

**OpenBMC Information:**
root@BMC:~# cat /etc/os-release
ID=openbmc-phosphor
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="2.14.0-dev"
VERSION_ID=2.14.0-dev-477-gf395a82c6-dirty
VERSION_CODENAME="langdale"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 2.14.0-dev"
BUILD_ID="20230316200521"
OPENBMC_TARGET_MACHINE="evb-ast2500"
EXTENDED_VERSION="2.14.0-dev-477-gf395a82c6-dirty"

root@BMC:~# uname -a
Linux BMC 6.0.19-67c9407 #1 Wed Jan 18 02:06:18 UTC 2023 armv6l GNU/Linux

williamspatrick, 2023-03-29:
> After module comes back up, I expect the /etc/fstab to be the new file, however it is still the old one. I see that the fstab under /run/initramfs/ro/etc/fstab is changed to the new one.

Is there a copy in `/run/initramfs/rw`?  It sounds like the overlay likely has a copy which is overriding the base version.  That is how Linux overlays work.

AlexD202, 2023-03-29:
I see 3 copies of the file 
/etc/fstab and /run/initramfs/rw/cow/etc/fstab are the same (default one)

/run/initramfs/ro/etc/fstab is my new file...

How do I make the /etc/fstab to take the new one when I load the image?

williamspatrick, 2023-03-29:
> How do I make the /etc/fstab to take the new one when I load the image?

https://stackoverflow.com/questions/57865670/overlayfs-restore-removed-or-changed-file-from-lower

closed 2023-03-29 by williamspatrick