openbmc/openbmc #3915: Issue with ipmi

alandvbh, 2023-03-29:
**Unexpected behavior you saw**
I have included ipmi drivers into the kernel and included ipmitool inside the image to configure the ipmi. the ipmi does not seem to be working and i am not getting correct details/status

**Expected behavior**
when I use ipmi, i expect to be getting a correct details/status

**To Reproduce**
Steps to reproduce the behavior:
1. Check out the code and modify meta-aspeed/recipes-kernel/linux/linux-aspeed/aspeed-g5/defconfig file to include ipmi drivers
2. Modify local.conf file to include ipmitool IMAGE_INSTALL:append = " ipmitool"
3. load the image on the module and start using ipmi.

**OpenBMC Information:**
root@BMC:~# cat /etc/os-release
ID=openbmc-phosphor
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="2.14.0-dev"
VERSION_ID=2.14.0-dev-477-gf395a82c6-dirty
VERSION_CODENAME="langdale"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 2.14.0-dev"
BUILD_ID="20230316200521"
OPENBMC_TARGET_MACHINE="evb-ast2500"
EXTENDED_VERSION="2.14.0-dev-477-gf395a82c6-dirty"

root@BMC:~# uname -a
Linux BMC 6.0.19-67c9407 https://github.com/openbmc/openbmc/pull/1 Wed Jan 18 02:06:18 UTC 2023 armv6l GNU/Linux

Here are some commands that might be helpful: Not sure why power shows off. Do I need to install a different driver?

**config.gz output:**

https://gist.github.com/alandvbh/2f71012e33461006b5baf9da14377032

**File /etc/modules is missing**

**root@BMC:~# ipmitool chassis status**
System Power         : off
Power Overload       : false
Power Interlock      : inactive
Main Power Fault     : false
Power Control Fault  : false
Power Restore Policy : always-off
Last Power Event     :
Chassis Intrusion    : inactive
Front-Panel Lockout  : inactive
Drive Fault          : false
Cooling/Fan Fault    : false
Front Panel Control  : none

**root@BMC:~# ipmitool power status**
Chassis Power is off

**root@BMC:~# dmesg | grep -i ipmi**
[    0.275872] IPMI message handler: version 39.2
[    0.276040] ipmi device interface
[    0.276587] ipmi_si: IPMI System Interface driver
[    0.277002] ipmi_si: Unable to find any System Interface(s)
[    0.277030] ipmi_ssif: IPMI SSIF Interface driver
[    0.277345] IPMI Watchdog: driver initialized
[    0.277372] IPMI poweroff: Copyright (C) 2004 MontaVista Software - IPMI Powerdown via sys_reboot
[   18.122798] systemd[1]: /lib/systemd/system/phosphor-ipmi-net@.socket:3: Invalid interface name, ignoring: sys-subsystem-net-devices-%i.device
[   18.734557] systemd[1]: Created slice Slice /system/phosphor-ipmi-net.

alandvbh, 2023-04-04:
Can someone please suggest a solution for my problem. I am not able to work with the ipmi.

williamspatrick, 2023-04-04:
I have no idea what you're trying to accomplish.

> I have included ipmi drivers into the kernel and included ipmitool inside the image to configure the ipmi. the ipmi does not seem to be working and i am not getting correct details/status

Why would you install the ipmi drivers into the BMC's kernel?  The ipmi drivers are for a host to talk its local BMC and not for the BMC to ... talk to itself?

I believe by default the `ipmitool` running on the BMC is able to talk to the local IPMI daemon.  The fact you are getting some status (and not errors) seems to corroborate that.

> when I use ipmi, i expect to be getting a correct details/status

```
root@BMC:~# ipmitool power status
Chassis Power is off
```

For all we know this might be the correct status.  What system is the BMC managing?  What kind of system did you install this image into?  What BMC image did you install into that system?

williamspatrick, 2023-04-04:
> `OPENBMC_TARGET_MACHINE="evb-ast2500"`

The EVB is not a target that can manage a server on its own.  There is a lot of code and configuration missing.

You're going to need to add some configuration to the whole IPMI subsystem.  Is your BIOS expecting KCS, IPMB, BT, something else?  You need to enable the appropriate IPMI bridge.

closed 2023-09-25 by williamspatrick