openbmc/openbmc #3917: Kirkstone fails to compile bmcweb with "-DBMCWEB_ENABLE_VM_NBDPROXY"

atlibtd, 2023-04-07:
**Unexpected behavior you saw**
bmcweb fails to build.

> | arm-openbmc-linux-gnueabi-g++ -marm -mcpu=arm1176jz-s -fstack-protector-strong -Os -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/mnt/work_storage/work/troym/yocto.poky/openbmc/build/romulus/tmp/work/arm1176jzs-openbmc-linux-gnueabi/bmcweb/1.0+gitAUTOINC+3d768a1651-r0/recipe-sysroot -Ibmcweb.p -I. -I../git -I../git/include -I../git/redfish-core/include -I../git/redfish-core/lib -I../git/http -flto=auto -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wnon-virtual-dtor -Wextra -Wpedantic -Werror -std=c++20 -fno-rtti -Os -g -fdata-sections -ffunction-sections -DNDEBUG -DBMCWEB_ENABLE_BASIC_AUTHENTICATION -DBMCWEB_ENABLE_COOKIE_AUTHENTICATION -DBMCWEB_ENABLE_HOST_SERIAL_WEBSOCKET -DBMCWEB_ENABLE_SSL -DBMCWEB_ENABLE_KVM -DBMCWEB_ENABLE_MUTUAL_TLS_AUTHENTICATION -DBMCWEB_ALLOW_DEPRECATED_POWER_THERMAL -DBMCWEB_ENABLE_REDFISH_DBUS_LOG_ENTRIES -DBMCWEB_ENABLE_REDFISH_HOST_LOGGER -DBMCWEB_ENABLE_REDFISH -DBMCWEB_ENABLE_SESSION_AUTHENTICATION -DBMCWEB_ENABLE_STATIC_HOSTING -DBMCWEB_ENABLE_VM_WEBSOCKET -DBMCWEB_ENABLE_XTOKEN_AUTHENTICATION -Wcast-align -Wconversion -Wformat=2 -Wold-style-cast -Woverloaded-virtual -Wsign-conversion -Wunused -Wno-attributes -Wduplicated-cond -Wduplicated-branches -Wlogical-op -Wunused-parameter -Wnull-dereference -Wdouble-promotion -fno-fat-lto-objects -fvisibility=hidden -fvisibility-inlines-hidden -fstack-protector-strong -fPIE -fPIC -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -DBOOST_ALLOW_DEPRECATED_HEADERS -DBOOST_ASIO_NO_DEPRECATED -DBOOST_ASIO_SEPARATE_COMPILATION -DBOOST_BEAST_SEPARATE_COMPILATION -DBOOST_BEAST_USE_STD_STRING_VIEW -DBOOST_EXCEPTION_DISABLE -DJSON_NOEXCEPTION -DBMCWEB_ENABLE_VM_NBDPROXY -Os -fvisibility-inlines-hidden -DBOOST_ASIO_DISABLE_THREADS -DBOOST_ALL_NO_LIB -DBOOST_SYSTEM_NO_DEPRECATED -DBOOST_ERROR_CODE_HEADER_ONLY -DBOOST_COROUTINES_NO_DEPRECATION_WARNING -MD -MQ bmcweb.p/src_webserver_main.cpp.o -MF bmcweb.p/src_webserver_main.cpp.o.d -o bmcweb.p/src_webserver_main.cpp.o -c ../git/src/webserver_main.cpp
> | In file included from ../git/src/webserver_main.cpp:31:
> | ../git/include/nbd_proxy.hpp: In lambda function:
> | ../git/include/nbd_proxy.hpp:261:47: error: 'asyncResp' was not declared in this scope
> |   261 |             auto getUserInfoHandler = [&conn, asyncResp](
> |       |                                               ^~~~~~~~~
> | ../git/include/nbd_proxy.hpp:263:55: error: 'utilities' in namespace 'dbus' does not name a type; did you mean 'utility'?
> |   263 |                                           const dbus::utilities::
> |       |                                                       ^~~~~~~~~
> 
> 

**Expected behavior**
Expected bmcweb to build and work

**To Reproduce**
Steps to reproduce the behavior:
1. git checkout kirkstone
2. Apply this diff

> $ git diff
> diff --git a/meta-phosphor/recipes-phosphor/interfaces/bmcweb_git.bb b/meta-phosphor/recipes-phosphor/interfaces/bmcweb_git.bb
> index b461a2e6b..ccfd82a83 100644
> --- a/meta-phosphor/recipes-phosphor/interfaces/bmcweb_git.bb
> +++ b/meta-phosphor/recipes-phosphor/interfaces/bmcweb_git.bb
> @@ -59,3 +59,5 @@ EXTRA_OEMESON = " \
>  SYSTEMD_SERVICE:${PN} += "bmcweb.service bmcweb.socket"
> 
>  FULL_OPTIMIZATION = "-Os "
> +
> +TARGET_CPPFLAGS += "-DBMCWEB_ENABLE_VM_NBDPROXY"

3.     . setup romulus
4.     bitbake obmc-phosphor-image

**OpenBMC Information:**

$ cat /etc/os-release
NAME="Ubuntu"
VERSION="20.04.5 LTS (Focal Fossa)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 20.04.5 LTS"
VERSION_ID="20.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=focal
UBUNTU_CODENAME=focal

$ uname -a
Linux crossfield 5.4.0-139-generic #156-Ubuntu SMP Fri Jan 20 17:27:18 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux

atlibtd, 2023-04-27:
Any update on this problem? This prevents upgrading from Honister to Kirkstone.
Thanks in advance.

williamspatrick, 2023-04-27:
I don't think anyone working on the project cares about supporting various repositories on past distro versions.  Please upgrade to master.

edtanous, 2023-04-27:
> Expected bmcweb to build and work
> +TARGET_CPPFLAGS += "-DBMCWEB_ENABLE_VM_NBDPROXY"

The backend for nbd_proxy was never upstreamed, which is why this option is commented out in the meson_options.txt with a note that we're looking for people to maintain it.  bmcweb does "Build and work" as configured by default.
https://github.com/openbmc/bmcweb/blob/02e01b5108d46720a0b438c0d79952464320d954/meson_options.txt#L32

You've bypassed that commented out option, so if you're interested in maintaining that, please let one of the various mailing lists know, participate, debug, and review patchsets.  That comment goes into more details about why that option is commented out.  @williamspatrick  is also correct that there is no maintainer support for support mixing various lineages of components.  If that's something you're interested in maintaining, please start participating with test results and patchsets to fix up the old builds.

FWIW, the code behind that option is slated for removal from bmcweb, and arguably should've been deleted already given it's been 2 years since being commented out, and the backends for it are still not present.

closed 2023-04-27 by edtanous

atlibtd, 2023-05-01:
Thank you for replies. I know that work on upstream of GXP support is in progress, but I don't know the exact time line. I'm not on the team doing that work so I don't have insights on the plans or progress. I'll look into the upstream plans. Note, the code has been fixed on master.

Since there isn't support for various lineages, I'll just add the Kirkstone diff here for bmcweb in case someone else wants it:

```
diff --git a/include/nbd_proxy.hpp b/include/nbd_proxy.hpp
index 9f46c031..fd572f37 100644
--- a/include/nbd_proxy.hpp
+++ b/include/nbd_proxy.hpp
@@ -258,9 +258,9 @@ inline void requestRoutes(App& app)
         .onopen([](crow::websocket::Connection& conn) {
             BMCWEB_LOG_DEBUG << "nbd-proxy.onopen(" << &conn << ")";

-            auto getUserInfoHandler = [&conn, asyncResp](
+            auto getUserInfoHandler = [&conn](
                                           const boost::system::error_code ec,
-                                          const dbus::utilities::
+                                          const dbus::utility::
                                               DBusPropertiesMap& userInfo) {
                 if (ec)
                 {
@@ -270,7 +270,8 @@ inline void requestRoutes(App& app)
                 }

                 const std::string* userRolePtr = nullptr;
-                auto userInfoIter = userInfo.find("UserPrivilege");
+                auto userInfoIter = std::find_if(userInfo.begin(), userInfo.end(),
+                            [](auto x) { return x.first == "UserPrivilege"; });
                 if (userInfoIter != userInfo.end())
                 {
                     userRolePtr =
@@ -300,7 +301,7 @@ inline void requestRoutes(App& app)
                     return;
                 }

-                auto openHandler = [&conn, asyncResp](
+                auto openHandler = [&conn](
                                        const boost::system::error_code ec,
                                        const dbus::utility::ManagedObjectType&
                                            objects) {

```

williamspatrick, 2023-05-01:
You’re welcome to add this as a patch in the repo within meta-phosphor on the Kirkstone patch.