openbmc/openbmc #3920: do_fetch failure of openbmc/linux consistently on 2.12.0

dan-bryant, 2023-04-13:
**Unexpected behavior you saw**
Followed the instructions for a build of 2.12.0 using the following:
```
sudo apt-get install -y git build-essential libsdl1.2-dev texinfo gawk chrpath diffstat zstd pigz
git clone https://github.com/openbmc/openbmc.git -b 2.12.0 openbmc
bash -c "source setup romulus build && bitbake obmc-phosphor-image"
```

And make it 65% of the way through the build but consistantly fail on step 3076 of 4720 with the `do_fetch` of `git://github.com/openbmc/linux`.  All other fetches work, no idea every build fails at that fetch.

```
Currently  3 running tasks (3076 of 4720)  65% 
...
ERROR: linux-aspeed-5.15.36+gitAUTOINC+49caedb668-r0 do_fetch: Fetcher failure: Fetch command export PSEUDO_DISABLED=1; export PATH="/home/builder/openbmc/scripts:/home/builder/openbmc/build/tmp/work/romulus-openbmc-linux-gnueabi/linux-aspeed/5.15.36+gitAUTOINC+49caedb668-r0/recipe-sysroot-native/usr/bin/arm-openbmc-linux-gnueabi:/home/builder/openbmc/build/tmp/work/romulus-openbmc-linux-gnueabi/linux-aspeed/5.15.36+gitAUTOINC+49caedb668-r0/recipe-sysroot/usr/bin/crossscripts:/home/builder/openbmc/build/tmp/work/romulus-openbmc-linux-gnueabi/linux-aspeed/5.15.36+gitAUTOINC+49caedb668-r0/recipe-sysroot-native/usr/sbin:/home/builder/openbmc/build/tmp/work/romulus-openbmc-linux-gnueabi/linux-aspeed/5.15.36+gitAUTOINC+49caedb668-r0/recipe-sysroot-native/usr/bin:/home/builder/openbmc/build/tmp/work/romulus-openbmc-linux-gnueabi/linux-aspeed/5.15.36+gitAUTOINC+49caedb668-r0/recipe-sysroot-native/sbin:/home/builder/openbmc/build/tmp/work/romulus-openbmc-linux-gnueabi/linux-aspeed/5.15.36+gitAUTOINC+49caedb668-r0/recipe-sysroot-native/bin:/home/builder/openbmc/poky/bitbake/bin:/home/builder/openbmc/build/tmp/hosttools"; export HOME="/home/builder"; LANG=C git -c core.fsyncobjectfiles=0 -c gc.autoDetach=false -c core.pager=cat clone --bare --mirror https://github.com/openbmc/linux /home/builder/openbmc/build/downloads/git2/github.com.openbmc.linux --progress failed with exit code 128, no output

ERROR: linux-aspeed-5.15.36+gitAUTOINC+49caedb668-r0 do_fetch: Bitbake Fetcher Error: FetchError('Unable to fetch URL from any source.', 'git://github.com/openbmc/linux;protocol=https;branch=dev-5.15')
```

**Expected behavior**
Expect build to complete

**To Reproduce**
I'm attempting the build under Docker, below is my Dockerfile
```
FROM ubuntu:20.04
ENV DEBIAN_FRONTEND=noninteractive
RUN apt-get update \
    && ln -fs /usr/share/zoneinfo/America/Chicago /etc/localtime \
    && apt-get install -y locales git build-essential libsdl1.2-dev texinfo gawk chrpath diffstat zstd pigz cpio lz4 wget \
    && dpkg-reconfigure --frontend noninteractive tzdata
RUN sed -i '/en_US.UTF-8/s/^# //g' /etc/locale.gen && locale-gen
ENV LANG en_US.UTF-8  
ENV LANGUAGE en_US:en  
ENV LC_ALL en_US.UTF-8   
RUN useradd -ms /bin/bash builder
USER builder
WORKDIR /home/builder/openbmc
RUN git clone https://github.com/openbmc/openbmc.git -b 2.12.0 .
RUN bash -c "source setup romulus build && bitbake obmc-phosphor-image"
```

williamspatrick, 2023-04-13:
This sounds like a network issue.  I regularly have connection drops like this to the GH archive CDN.  This download is a few hundred MB and the GH CDN seems to give me about 100kbps on this package.

I'd suggest running `bitbake linux-aspeed -c fetch` a few times until it passes.

williamspatrick, 2023-04-13:
`bitbake <image> —runall=fetch`

williamspatrick, 2023-04-13:
> `bitbake linux-aspeed -c fetch` worked well. I think when I'm in the middle of the build my system gets so pegged that it begins to hinder my network connectivity.
> 
> > `bitbake <image> —runall=fetch`
> 
> I tried `bitbake <image> —runall=fetch` and `bitbake obmc-phosphor-image -c fetchall`. Received similar errors on both:
> 
> ```
> ERROR: Task do_fetchall does not exist for target obmc-phosphor-image (/home/builder/openbmc/meta-phosphor/recipes-phosphor/images/obmc-phosphor-image.bb:do_fetchall
> ```

Did you `s/<image>/obmc-phosphor-image/` in your case?

dan-bryant, 2023-04-13:
> Did you `s/<image>/obmc-phosphor-image/` in your case?

LOL... my bad... yes, `bitbake obmc-phosphor-image —runall=fetch` returns `ERROR: Nothing PROVIDES '—runall=fetch'`

williamspatrick, 2023-04-14:
It is suppose to be dash-dash but thanks to my Apple device it became something else. Dash-dash-runall.

dan-bryant, 2023-04-14:
LOL... too funny.  Yes, seems to return now saying all 415 objects are already downloaded.

Thx for your patience, I'll move on to the build errors in `function2`, `fmt`, and `cli11` now.

closed 2023-04-14 by dan-bryant