openbmc/openbmc #3922: Issue with BMC Upgrade

AlexD202, 2023-04-27:
**Note on Bugs**
I checked out the latest code from the github and created an image and upgraded my AST2500 module with this image by transferring the image under /tmp/images and using the following command and then rebooting the module
busctl set-property xyz.openbmc_project.Software.BMC.Updater /xyz/openbmc_project/software/{ID_NUMBE} xyz.openbmc_project.Software.Activation RequestedActivation s  xyz.openbmc_project.Software.Activation.RequestedActivations.Active

the module came back normal and there was no issue. 

Then I used an older version of the code and followed the same procedures to upgrade. It got stuck in some part of the upgrade and after that, the module does not come back up. It keeps rebooting and shows kernel panic. I Could only go to the u-boot environment, but if I reset, it keeps doing the same thing (kernel panic).

I am not able to get into the module to upgrade the image to something else.
I also tried socflash.sh which is a script given by the AST2500 compnay for upgrading from the host, but I get the following message:
The BMC is proteced, skip to update firmware image!!

I tried 2 different modules with same steps and both got into this issue. Is this a known issue where downgrading fails?
What do we do when we get into this situation? Is there a way to forcefully upgrade?

**Unexpected behavior you saw**
The module keeps rebooting with a kernel panic message

**Expected behavior**
After upgrade, I should be able to use the module

**To Reproduce**
Steps to reproduce the behavior:
1. 

**OpenBMC Information:**
Latest Image (I only have the below details as I am not able to log into it anymore):
 - /etc/os-release:
ID=openbmc-phosphor
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="2.14.0-dev"
VERSION_ID=2.14.0-dev-987-g0ee147759-dirty
VERSION_CODENAME="langdale"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 2.14.0-dev"
BUILD_ID="20230425033157"
OPENBMC_TARGET_MACHINE="evb-ast2500"
EXTENDED_VERSION="2.14.0-dev-987-g0ee147759-dirty"

Older Image:
oot@BMC:~# cat /etc/os-release
ID=openbmc-phosphor
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="2.14.0-dev"
VERSION_ID=2.14.0-dev-477-gf395a82c6-dirty
VERSION_CODENAME="langdale"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 2.14.0-dev"
BUILD_ID="20230316200521"
OPENBMC_TARGET_MACHINE="evb-ast2500"
EXTENDED_VERSION="2.14.0-dev-477-gf395a82c6-dirty"

root@BMC:~# uname -a
Linux BMC 6.0.19-67c9407 https://github.com/openbmc/openbmc/pull/1 Wed Jan 18 02:06:18 UTC 2023 armv6l GNU/Linux

williamspatrick, 2023-04-27:
What is the kernel panic?

AlexD202, 2023-04-27:
[    0.124628] clocksource: Switched to clocksource FTTMR010-TIMER2
[    0.201704] NET: Registered PF_INET protocol family
[    0.202341] IP idents hash table entries: 16384 (order: 5, 131072 bytes, linear)
[    0.206433] tcp_listen_portaddr_hash hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.206528] Table-perturb hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.206582] TCP established hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.206720] TCP bind hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.206899] TCP: Hash tables configured (established 8192 bind 8192)
[    0.207196] UDP hash table entries: 512 (order: 1, 8192 bytes, linear)
[    0.207287] UDP-Lite hash table entries: 512 (order: 1, 8192 bytes, linear)
[    0.207852] NET: Registered PF_UNIX/PF_LOCAL protocol family
[    0.211833] Unpacking initramfs...
[    0.227561] workingset: timestamp_bits=30 max_order=18 bucket_order=0
[    0.238713] squashfs: version 4.0 (2009/01/31) Phillip Lougher
[    0.239457] jffs2: version 2.2. (SUMMARY)  © 2001-2006 Red Hat, Inc.
[    0.257894] gpio gpiochip0: (1e780000.gpio): not an immutable chip, please consider fixing it!
[    0.276013] IPMI message handler: version 39.2
[    0.276177] ipmi device interface
[    0.276687] ipmi_si: IPMI System Interface driver
[    0.277113] ipmi_si: Unable to find any System Interface(s)
[    0.277145] ipmi_ssif: IPMI SSIF Interface driver
[    0.277451] IPMI Watchdog: driver initialized
[    0.277475] IPMI poweroff: Copyright (C) 2004 MontaVista Software - IPMI Powerdown via sys_reboot
[    0.282197] Serial: 8250/16550 driver, 6 ports, IRQ sharing enabled
[    0.299452] printk: console [ttyS4] disabled
[    0.299651] 1e784000.serial: ttyS4 at MMIO 0x1e784000 (irq = 20, base_baud = 1500000) is a 16550A
[    1.274158] printk: console [ttyS4] enabled
[    1.295396] timeriomem_rng 1e6e2078.hwrng: 32bits from 0x(ptrval) @ 1us
[    1.303014] aspeed_gfx 1e6e6000.display: assigned reserved memory node framebuffer
[    1.325820] random: crng init done
[    1.330695] [drm] Initialized aspeed-gfx-drm 1.0.0 20180319 for 1e6e6000.display on minor 0
[    1.359926] aspeed_gfx 1e6e6000.display: [drm] fb0: aspeed-gfx-drmd frame buffer device
[    1.392406] loop: module loaded
[    1.460615] spi-nor spi0.0: mx66l51235f (65536 Kbytes)
[    1.594958] spi-aspeed-smc 1e620000.spi: CE0 read buswidth:2 [0x203c0641]
[    1.747357] 5 fixed-partitions partitions found on MTD device bmc
[    1.753520] Creating 5 MTD partitions on "bmc":
[    1.758189] 0x000000000000-0x000000060000 : "u-boot"
[    1.787611] 0x000000060000-0x000000080000 : "u-boot-env"
[    1.807776] 0x000000080000-0x0000004c0000 : "kernel"
[    1.815176] 0x0000004c0000-0x000001c00000 : "rofs"
[    1.828115] 0x000001c00000-0x000002000000 : "rwfs"
[    1.850210] spi-nor spi1.0: unrecognized JEDEC id bytes: 00 03 00 00 00 00
[    1.863887] ftgmac100 1e660000.ethernet: Read MAC address ca:b0:5d:00:dd:3e from chip
[    1.893297] Generic PHY 1e660000.ethernet--1:00: attached PHY driver (mii_bus:phy_addr=1e660000.ethernet--1:00, irq=POLL)
[    1.905635] ftgmac100 1e660000.ethernet eth0: irq 22, mapped at 34f17853
[    1.913238] ftgmac100 1e680000.ethernet: Read MAC address b6:52:ec:f3:7f:a0 from chip
[    1.942606] Generic PHY 1e680000.ethernet--1:00: attached PHY driver (mii_bus:phy_addr=1e680000.ethernet--1:00, irq=POLL)
[    1.954899] ftgmac100 1e680000.ethernet eth1: irq 23, mapped at 97afd6dc
[    8.578697] mmc0: SDHCI controller on 1e740100.sdhci [1e740100.sdhci] using ADMA
[    8.589611] Freeing unused kernel image (initmem) memory: 1024K
[    8.596218] Run /init as init process
[    8.676483] mmc0: new high speed SD card at address 43bd
[    8.689440] mmcblk0: mmc0:43bd R02GS 1.85 GiB
[    8.711931]  mmcblk0: p1
rofs = mtd4 squashfs rwfs = mtd5 jffs2
mount: mounting /dev/mtdblock4 on run/initramfs/ro failed: Invalid argument
[   10.040080] jffs2: notice: (91) jffs2_build_xattr_subsystem: complete building xattr subsystem, 5351 of xdatum (25 unchecked, 5326 orphan) and 6135 of xref (6099 dead, 2 orphan) found.
find: run/initramfs/ro/etc/os-release: No such file or directory
Fri Mar  3 18:52:26 UTC 2023
[   10.264377] overlayfs: upper fs does not support tmpfile.
[   10.270795] overlayfs: upper fs does not support RENAME_WHITEOUT.
rootfs on / type rootfs (rw)
dev on /dev type devtmpfs (rw,relatime,size=489104k,nr_inodes=122276,mode=755)
sys on /sys type sysfs (rw,relatime)
proc on /proc type proc (rw,relatime)
tmpfs on /run type tmpfs (rw,nodev,relatime,mode=755)
/dev/mtdblock5 on /run/initramfs/rw type jffs2 (rw,relatime)
cow on /root type overlay (rw,relatime,lowerdir=run/initramfs/ro,upperdir=run/initramfs/rw/cow,workdir=run/initramfs/rw/work)
chroot: can't execute '/bin/sh': No such file or directory
[... 40 lines cut ...]
     Compression:  uncompressed
     Data Start:   0x2008012c
     Data Size:    3053400 Bytes = 2.9 MiB
     Architecture: ARM
     OS:           Linux
     Load Address: 0x80001000
     Entry Point:  0x80001000
     Hash algo:    sha256
     Hash value:   3c86609f8fc2c3833cc0dcedf2ee0a7e8aea4ce5220391b94f995ae3fb49a3d8
   Verifying Hash Integrity ... sha256+ OK
## Loading ramdisk from FIT Image at 20080000 ...
   Using 'conf-aspeed-ast2500-evb.dtb' configuration
   Trying 'ramdisk-1' ramdisk subimage
     Description:  obmc-phosphor-initramfs
     Type:         RAMDisk Image
     Compression:  uncompressed
     Data Start:   0x20370490
     Data Size:    1086388 Bytes = 1 MiB
     Architecture: ARM
     OS:           Linux
     Load Address: unavailable
     Entry Point:  unavailable
     Hash algo:    sha256
     Hash value:   cbf86a48d335f60a61a0d2dcb6be67409cbfef9fa42764d0aa869356de3ca49c
   Verifying Hash Integrity ... sha256

AlexD202, 2023-04-27:
It just stays in this state and keep resetting and printing the same thing

amboar, 2023-04-27:
```
[ 8.596218] Run /init as init process
[ 8.676483] mmc0: new high speed SD card at address 43bd
[ 8.689440] mmcblk0: mmc0:43bd R02GS 1.85 GiB
[ 8.711931] mmcblk0: p1
rofs = mtd4 squashfs rwfs = mtd5 jffs2
mount: mounting /dev/mtdblock4 on run/initramfs/ro failed: Invalid argument
```

Probably a bug in the init script. Can you try booting the system with `rdinit=/bin/sh` in `bootargs` and then run `sh -x /init`?

williamspatrick, 2023-04-28:
I suspect these are fixed now by either 1bcd53d4a or 321a95f8e

```
$ git rev-list origin/master "^0ee147759" | xargs -n1 git log --oneline -n1
376078392 (origin/master, origin/HEAD) meta-inventec: fix shellcheck warnings
8c2262344 meta-quanta: fix shellcheck issues
bd58e9548 host-error-monitor: srcrev bump ee00ccc006..03ed41b37b
857f086eb meta-ibm: Using append in override syntax
c36a02126 meta-google: rng-tools: Add package suffix
8af674e68 meta-phosphor: add initrdscripts reviewer
2ea84ebfb meta-phosphor: mmc-init: Fix creation of critical directories
bb3935f2b openpower-vpd-parser: srcrev bump f457a3ef9b..636391010b
70f37e1ab openpower-debug-collector: srcrev bump 1485d7cd77..85c1ebf5ff
9ac168bf0 dbus-sensors: srcrev bump e456925b73..9a472e8ed1
bd99ba42c phosphor-networkd: srcrev bump b03a760fa2..f78a415e15
68ac86c49 bmcweb: srcrev bump 72e2137737..02e01b5108
32ec58f5f pldm: srcrev bump 6c39c7a7af..87083f266c
1bcd53d4a meta-phosphor: initrdscripts: fix multi-word variable expansion
321a95f8e meta-phosphor: initrdscripts: fix flash update
```

AlexD202, 2023-04-28:
Thanks. I pulled the latest changes and it works now, issue can be closed now.. It was getting stuck in the spot below just for your reference:
+ IFS= read -r f
+ umount /oldroot/dev/pts
+ IFS= read -r f
+ umount /oldroot/dev
+ IFS= read -r f
+ umount /oldroot
+ IFS= read -r f
+ umount /mnt/initramfs/rw
+ IFS= read -r f
+ umount /mnt/initramfs/ro
+ IFS= read -r f
+ umount /mnt
+ IFS= read -r f
+ set +x
Pinging watchdog with args -t 1 -T 5
update: --clean-saved-files
watchdog: invalid number ' 1 -T 5'
[  529.677088] jffs2: notice: (374) jffs2_build_xattr_subsystem: complete building xattr subsystem, 31 of xdatum (17 unchecked, 7 orphan) and 40 of xref (5 dead, 2 orphan) found.
Updating kernel...
Erasing block: 66/66 (100%)
Writing kb: 4182/4182 (100%)
Verifying kb: 4182/4182 (100%)
Updating rofs...
Erasing block: 224/272 (82%)

williamspatrick, 2023-04-28:
```
Updating kernel...
Erasing block: 66/66 (100%)
Writing kb: 4182/4182 (100%)
Verifying kb: 4182/4182 (100%)
Updating rofs...
Erasing block: 224/272 (82%)
```

Looks like it simply hit a watchdog timeout after X amount of time because we didn't update the watchdog time properly prior to starting the update code.  Makes sense based on the bug / fix that was delivered.

closed 2023-04-28 by williamspatrick

AlexD202, 2023-05-09:
Hey guys. This issue is fixed, but can you please let me know how I can forcefully load a new image on the BMC module when current image fails to come back up.

I have a few module that are in this state (kernel panic) where module does not come back up and i am not able to transfer any image to it to upgrade it. What is an alternative for upgrading the module when this happens?

williamspatrick, 2023-05-09:
If you look at the initscripts, there are some kernel cmdline options that will drop you into an emergency shell which you could likely use to put a new image on. 

Do you have two flash banks in your server design?  In most of my companies designs we have two flash banks and in this state you could restore an older version from the alternative flash bank by copying in u-boot.

amboar, 2023-05-09:
The easiest thing to do is to netboot a kernel that isn't broken and use `flashcp`. If you can't netboot the board then an alternative is a flash programmer, or to try reflash it over the debug UART with either `socflash` or [culvert](https://github.com/amboar/culvert)