openbmc/openbmc #3923: Unable to boot AST2500 image on Qemu

LingfengPang, 2023-04-28:
**The image is built,  I try launching it using the command**

```
qemu-system-arm -m 256 -M romulus-bmc -nographic -drive file=/home/onetree/openbmc/build/tmp/deploy/images/romulus/obmc-phosphor-image-romulus.static.mtd,format=raw,if=mtd -net nic -net user,hostfwd=:127.0.0.1:2222-:22,hostfwd=:127.0.0.1:2443-:443,hostname=qemu
```

**Qemu log:**
```
U-Boot 2019.04 (Feb 21 2023 - 23:19:44 +0000)

SOC : AST2500-A1 
RST : Power On 
2nd Boot : Enable
LPC Mode : SIO:Enable : SuperIO-2e
Eth : MAC0: RMII/NCSI, , MAC1: RMII/NCSI, 
Model: Romulus BMC
DRAM:  192 MiB (capacity:256 MiB, VGA:64 MiB, ECC:off)
MMC:   
Loading Environment from SPI Flash... SF: Detected n25q256a with page size 256 Bytes, erase size 4 KiB, total 32 MiB
*** Warning - bad CRC, using default environment

In:    serial@1e784000
Out:   serial@1e784000
Err:   serial@1e784000
Net:   ftgmac100_probe - NCSI detected

Warning: ethernet@1e660000 (eth0) using random MAC address - aa:3c:9a:28:84:1b
eth0: ethernet@1e660000
Hit any key to stop autoboot:  0 
## Loading kernel from FIT Image at 20080000 ...
   Using 'conf-aspeed-bmc-opp-romulus.dtb' configuration
   Trying 'kernel-1' kernel subimage
     Description:  Linux kernel
     Type:         Kernel Image
     Compression:  uncompressed
     Data Start:   0x20080128
     Data Size:    3087936 Bytes = 2.9 MiB
     Architecture: ARM
     OS:           Linux
     Load Address: 0x80001000
     Entry Point:  0x80001000
     Hash algo:    sha256
     Hash value:   ef3a7c73f18b9627f8e855293970c56ab6d223785ba5d62d5f7c72dc2085293f
   Verifying Hash Integrity ... sha256+ OK
## Loading ramdisk from FIT Image at 20080000 ...
   Using 'conf-aspeed-bmc-opp-romulus.dtb' configuration
   Trying 'ramdisk-1' ramdisk subimage
     Description:  obmc-phosphor-initramfs
     Type:         RAMDisk Image
     Compression:  uncompressed
     Data Start:   0x2037a5ec
     Data Size:    1084060 Bytes = 1 MiB
     Architecture: ARM
     OS:           Linux
     Load Address: unavailable
     Entry Point:  unavailable
     Hash algo:    sha256
     Hash value:   46a0e777f5695684cc654443961c8bbe1886926db4825008b46c4c9ab3fd24f7
   Verifying Hash Integrity ... sha256+ OK
## Loading fdt from FIT Image at 20080000 ...
   Using 'conf-aspeed-bmc-opp-romulus.dtb' configuration
   Trying 'fdt-aspeed-bmc-opp-romulus.dtb' fdt subimage
     Description:  Flattened Device Tree blob
     Type:         Flat Device Tree
     Compression:  uncompressed
     Data Start:   0x20372080
     Data Size:    33952 Bytes = 33.2 KiB
     Architecture: ARM
     Hash algo:    sha256
     Hash value:   c84591efdd02c6b02ffc10b61c795cd999ec9f1b8638921fe9c29042e7eb684d
   Verifying Hash Integrity ... sha256+ OK
   Booting using the fdt blob at 0x20372080
   Loading Kernel Image ... OK
   Loading Ramdisk to 89e8a000, end 89f92a9c ... OK
   Loading Device Tree to 89e7e000, end 89e8949f ... OK
fdt_root: FDT_ERR_BADMAGIC
ERROR: root node setup failed
 - must RESET the board to recover.

FDT creation failed! hanging...### ERROR ### Please RESET the board ###
```

geissonator, 2023-04-28:
I'm not sure what versions you're using but this works fine for me:
```
wget https://jenkins.openbmc.org/job/latest-qemu-x86/lastSuccessfulBuild/artifact/qemu/build/qemu-system-arm

chmod u+rwx qemu-system-arm

wget  https://jenkins.openbmc.org/job/latest-master/label=docker-builder,target=romulus/lastSuccessfulBuild/artifact/openbmc/build/tmp/deploy/images/romulus/obmc-phosphor-image-romulus.static.mtd

./qemu-system-arm -m 256 -M romulus-bmc -nographic -drive file=./obmc-phosphor-image-romulus.static.mtd,format=raw,if=mtd -net nic -net user,hostfwd=:127.0.0.1:2222-:22,hostfwd=:127.0.0.1:2443-:443,hostname=qemu
```

LingfengPang, 2023-04-30:
> I'm not sure what versions you're using but this works fine for me:
>
> ```
> wget https://jenkins.openbmc.org/job/latest-qemu-x86/lastSuccessfulBuild/artifact/qemu/build/qemu-system-arm
>
> chmod u+rwx [...]

Thank you for your suggestion. But I use these commands, Qemu log is same as me.

LingfengPang, 2023-05-01:
I find it is the problem of ubuntu version.
Upgrade the ubuntu to 20.04 and try to boot
Qemu logs:
```
U-Boot 2019.04 (Mar 10 2023 - 01:54:10 +0000)

SOC : AST2500-A1 
RST : Power On 
2nd Boot : Enable
LPC Mode : SIO:Enable : SuperIO-2e
Eth : MAC0: RMII/NCSI, , MAC1: RMII/NCSI, 
Model: Romulus BMC
DRAM:  960 MiB (capacity:1024 MiB, VGA:64 MiB, ECC:off)
MMC:   
Loading Environment from SPI Flash... SF: Detected n25q256a with page size 256 Bytes, erase size 4 KiB, total 32 MiB
*** Warning - bad CRC, using default environment

In:    serial@1e784000
Out:   serial@1e784000
Err:   serial@1e784000
Net:   ftgmac100_probe - NCSI detected

Warning: ethernet@1e660000 (eth0) using random MAC address - a2:34:b3:52:08:17
eth0: ethernet@1e660000
Hit any key to stop autoboot:  0 
## Loading kernel from FIT Image at 20080000 ...
   Using 'conf-aspeed-bmc-opp-romulus.dtb' configuration
   Trying 'kernel-1' kernel subimage
     Description:  Linux kernel
     Type:         Kernel Image
     Compression:  uncompressed
     Data Start:   0x20080128
     Data Size:    3081352 Bytes = 2.9 MiB
     Architecture: ARM
     OS:           Linux
     Load Address: 0x80001000
     Entry Point:  0x80001000
     Hash algo:    sha256
     Hash value:   d1d96902a9f4d8cd71d35cb030c13d935dd6ed8773aa45e0a34ea356e347b83f
   Verifying Hash Integrity ... sha256+ OK
## Loading ramdisk from FIT Image at 20080000 ...
   Using 'conf-aspeed-bmc-opp-romulus.dtb' configuration
   Trying 'ramdisk-1' ramdisk subimage
     Description:  obmc-phosphor-initramfs
     Type:         RAMDisk Image
     Compression:  uncompressed
     Data Start:   0x20378c34
     Data Size:    1085128 Bytes = 1 MiB
     Architecture: ARM
     OS:           Linux
     Load Address: unavailable
     Entry Point:  unavailable
     Hash algo:    sha256
     Hash value:   ff013347076b4df315659b3c9877c955edf66d30213f11aa9a4f2da9510be528
   Verifying Hash Integrity ... sha256+ OK
## Loading fdt from FIT Image at 20080000 ...
   Using 'conf-aspeed-bmc-opp-romulus.dtb' configuration
   Trying 'fdt-aspeed-bmc-opp-romulus.dtb' fdt subimage
     Description:  Flattened Device Tree blob
     Type:         Flat Device Tree
     Compression:  uncompressed
     Data Start:   0x203706c8
     Data Size:    33952 Bytes = 33.2 KiB
     Architecture: ARM
     Hash algo:    sha256
     Hash value:   c84591efdd02c6b02ffc10b61c795cd999ec9f1b8638921fe9c29042e7eb684d
   Verifying Hash Integrity ... sha256+ OK
   Booting using the fdt blob at 0x203706c8
   Loading Kernel Image ... OK
   Loading Ramdisk to 8fef7000, end 8ffffec8 ... OK
[... 263 lines cut ...]
[    2.750608] In-situ OAM (IOAM) with IPv6
[    2.753365] NET: Registered PF_PACKET protocol family
[    2.754441] 8021q: 802.1Q VLAN Support v1.8
[    2.771798] printk: console [netcon0] enabled
[    2.772198] netconsole: network logging started
[    2.778514] input: gpio-keys as /devices/platform/gpio-keys/input/input0
[    2.809482] Freeing unused kernel image (initmem) memory: 1024K
[    2.817947] Checked W+X mappings: passed, no W+X pages found
[    2.818447] Run /init as init process
rofs = mtd4 squashfs   rwfs = mtd5 jffs2
[    4.650923] jffs2: notice: (85) jffs2_build_xattr_subsystem: complete building xattr subsystem, 0 of xdatum (0 unchecked, 0 orphan) and 0 of xref (0 dead, 0 orphan) found.
find: run/initramfs/rw/cow/var/lib/systemd/random-seed: No such file or directory
[    4.997348] overlayfs: upper fs does not support tmpfile.
[    5.011736] overlayfs: upper fs does not support RENAME_WHITEOUT.
[    8.847687] systemd[1]: systemd 253.1^ running in system mode (+PAM -AUDIT -SELINUX -APPARMOR -IMA -SMACK +SECCOMP -GCRYPT -GNUTLS -OPENSSL -ACL +BLKID -CURL -ELFUTILS -FIDO2 -IDN2 -IDN -IPTC -KMOD -LIBCRYPTSETUP +LIBFDISK -PCRE2 -PWQUALITY -P11KIT -QRENCODE -TPM2 -BZIP2 -LZ4 -XZ -ZLIB +ZSTD -BPF_FRAMEWORK -XKBCOMMON -UTMP -SYSVINIT default-hierarchy=unified)
[    8.854171] systemd[1]: Detected architecture arm.

Welcome to Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 2.14.0-dev!

[    9.002156] systemd[1]: Hostname set to <romulus>.
[    9.017238] systemd[1]: Initializing machine ID from random generator.
[    9.636136] systemd[1]: Using hardware watchdog 'aspeed_wdt', version 0, device /dev/watchdog0
[    9.670717] systemd[1]: Watchdog running with a timeout of 2min.
```
Still boot failed.

LingfengPang, 2023-05-06:
I sloved this problem, maybe I have ever installed other qemu makes the version of glibc too old?

closed 2023-05-06 by LingfengPang