openbmc/openbmc #3924: Can't get out of emergency mode due to bad fstab file

AlexD202, 2023-05-04:
**Note on Bugs**
I will explain my issue here but basically I need to find a way to forcefully re-image the BMC module.
I have an issue in my fstab (content is below) and as a result, my init is failing and taking me to an emergency mode where I have limited functionality (sshd, scp, and nothing else works, so I am not able to transfer a new image with a fixed init/fstab file to get out of this mode). I usually transfer the image into the /tmp/images and then install it from there. If module does not have any image on it for the first time, I use sockflash that was given by the vender from another module to install the image on it. But once I install the image for the first time, I can no longer use that script as I get the following error
bmc id write protected, skip upgrade.

sh-5.2# cat /etc/fstab
# stock fstab - you probably want to override this with a machine specific one

/dev/root            /                    auto       defaults              1  1
proc                 /proc                proc       defaults              0  0
devpts               /dev/pts             devpts     mode=0620,ptmxmode=0666,gid=5      0  0
tmpfs                /run                 tmpfs      mode=0755,nodev,nosuid,strictatime 0  0
tmpfs                /var/volatile        tmpfs      defaults              0  0

# uncomment this if your device has a SD/MMC/Transflash slot
/dev/mmcblk0p1       /media          auto       defaults,sync,auto  0  0

If I just comment out the last line and try to reboot, the fstab I have under /run/initramfs/ro/etc/fstab (which is a read-only file and I can't change it) overrides the /etc/fstab. I also tried removing the lines that removes /etc/fstab and remounts that I have inside my /run/initramfs/init, but after reboot this file also gets overridden

Any help to get me out of this issue?
Is there a way to forcefully delete the current image so the module goes back to the factory default?

**Unexpected behavior you saw**
BMC Module goes into an emergency mode if init script is bad and there is no way to get out of that mode.

**Expected behavior**
There should be a way such that if init script fails for any reason, user should be able to either delete the image completely or find a way to get into the image to upgrade.

**To Reproduce**
1. Load a new image with an issue in the fstab
2. init fails and takes you to an emergency mode.

**OpenBMC Information:**
sh-5.2# uname -a
Linux evb-ast2500 6.1.15-580639a #1 Thu Apr  6 00:55:09 UTC 2023 armv6l GNU/Linux

sh-5.2# cat /etc/os-release
ID=openbmc-phosphor
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="2.14.0-dev"
VERSION_ID=2.14.0-dev-1020-g18caa9759-dirty
VERSION_CODENAME="langdale"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 2.14.0-dev"
BUILD_ID="20230428153944"
OPENBMC_TARGET_MACHINE="evb-ast2500"
EXTENDED_VERSION="2.14.0-dev-1020-g18caa9759-dirty"

**Additional context**
sh-5.2# df -h
Filesystem                Size      Used Available Use% Mounted on
dev                     477.6M         0    477.6M   0% /dev
tmpfs                   494.7M     11.3M    483.4M   2% /run
/dev/mtdblock4           17.3M     17.3M         0 100% /run/initramfs/ro
/dev/mtdblock5            4.0M    376.0K      3.6M   9% /run/initramfs/rw
cow                       4.0M    376.0K      3.6M   9% /
tmpfs                   494.7M         0    494.7M   0% /dev/shm
tmpfs                   494.7M         0    494.7M   0% /tmp
tmpfs                   494.7M         0    494.7M   0% /var/volatile

This is where it fails after a reboot:

[ TIME ] Timed out waiting for device /dev/mmcblk0p1.
[DEPEND] Dependency failed for /media.

You are in emergency mode. After logging in, type "journalctl -xb" to view
system logs, "systemctl reboot" to reboot, "systemctl default" or "exit"
to boot into default mode.
Give root password for maintenance
(or press Control-D to continue):

mdmillerii, 2023-05-15:
The phosphor init script does support access to a boot shell with the appropriate kernel command line options.
You can configure ip using the kernel command line and then transfer files using http, ftp, or tftp.   You can unpack a firmware update tarball and apply the updates.

Note that the readonly partition usually has a read-only file system on it; squashfs never supports updates other than by replacement.

lingyaof, 2023-11-08:
Hi all , i have a question that ro and rw filesystem.
In yocto openbmc project , most source packages are generated in ro (squashfs),
How to configure it to compilation and be generated in rw(jffs)?

williamspatrick, 2023-11-08:
> Hi all , i have a question that ro and rw filesystem.
> In yocto openbmc project , most source packages are generated in ro (squashfs),
> How to configure it to compilation and be generated in rw(jffs)?

Why are you hijacking an unrelated issue?

williamspatrick, 2023-11-08:
@AlexD202 Do you still have an issue here?  I can’t understand the concern.

mdmillerii, 2023-11-08:
> Why are you hijacking an unrelated issue? 

Also, @lingyaof it's unsupported.    I can discuss why and alternatives in [discord or Mail](https://github.com/openbmc/openbmc#questions)