openbmc/openbmc #3926: Put output of ". setup" on front page

sporkman, 2023-05-13:
I understand you don't want questions here, but seeing as a very common one is "what is supported?" and the answer currently is:

> This is a common question, particularly regarding boards from popular COTS (commercial off-the-shelf) vendors such as Supermicro and ASRock. You can see the list of supported boards by running . setup (with no further arguments) in the root of the OpenBMC source tree. Most of the platforms supported by OpenBMC are specialized servers operated by companies running large datacenters, but some more generic COTS servers are supported to varying degrees.

...perhaps we can call it a documentation bug?

Asking folks to download and have a basic dev environment setup just to see a list of supported hardware is probably part of why the question is "frequently asked". Any chance you can work this into your CI chain and have that output posted somewhere on GitHub?

williamspatrick, 2023-05-13:
What does “is supported” mean to you?

sporkman, 2023-05-13:
Is able to be used on a particular piece of hardware? Not sure if I'm replying to sarcasm or not, sorry.

zevweiss, 2023-05-13:
I think that was alluding to the fact that

> not all ports are equally mature -- some platforms are better supported than others, and functionality on some "supported" boards may be fairly limited.

So what exactly "supported" means can easily vary.  It might mean "something can be compiled/built/installed on this platform"; it might mean "this platform is feature-complete and production-ready"; it might mean "there's someone actively working on and maintaining this platform who will be responsive to bug reports if I encounter problems".

Note that the output of `. setup` _is_ included in the README ([here](https://github.com/openbmc/openbmc/#3-target-your-hardware)), though it's not always 100% up to date.  Setting up some sort of automation to have an always-up-to-date version of that posted somewhere easily found online would be nice; perhaps we could hack up some sort of CI check to enforce that that section of the README gets updated when platforms get added (or removed)...

williamspatrick, 2023-05-13:
> Is able to be used on a particular piece of hardware? Not sure if I'm replying to sarcasm or not, sorry.

It was not sarcasm. We regularly have people that think this project is something like OpenWRT where there is a big list of “supported hardware” that they can go buy somewhere. None of the primary platforms for this project are currently like that, as was written in the Q/A, so I’m trying to figure out what information would really satisfy what you’re looking for. 

In many ways we make people download the code because if they’re not willing to do that there is a 99% chance this isn’t going to work for their hardware. There are only a few piece of hardware for which this project has been reverse engineered in some way and none of them have top-tier support. 

As stated, almost everyone working on this project is associated in some way with a team doing the hardware design for which they are adding support.

edtanous, 2024-09-04:
Over a year old and no replies.  As Andrew points out, the setup information is available in the readme, and as Patrick points, out, it's unlikely that anything further will be of use to the common developer/engineer if they can't build an image.

I suspect this can be closed?

sporkman, 2024-09-04:
My original point still stands I think. I have a bunch of supermicro boards. They have onboard BMCs. The list of supported platforms seems to relate to code names or similar (example: "romulus" - is that Dell? Lenovo? Supermicro? None of the above? If there's a knowledge gap between normal hardware folks that manage a rack or two of stuff and say, Facebook, perhaps addressing that fact and noting "none of this is commodity hardware" would be helpful to again, save the project and the seeker of information and 3rd party BMC firmware a little bit of time.

edtanous, 2024-09-04:
I think the distinction is that _some_ of the buildable platforms are commodity hardware (s7106 for example), so we can't just make the blanket statement that "none of this is commodity hardware".

I'm not really sure how we can close this bug as you're asking it.  I understand that you wish Supermicro boards didn't require development effort, but that's not really something we can solve without folks contributing code to support them.  With some luck in the future, folks will get more and more support working.  For what it's worth, X11SPI listed on the main readme _is_ a Supermicro board, but presumably that wasn't the board you were looking for.

The title of "Put output of ". setup" on front page" has been accomplished.  You're right, the non-retail platforms are codenames, but that's because they're only ever known by their codenames, so I don't know what we can fix there.  The retail platforms are generally called out by their retail product name, so that should be searchable.  If there's specific platform codename that has a retail name that needs fixed, we certainly can talk about renaming those specific boards, but it's not clear (to me at least) what more can be done here?

Given the above, in terms of resolving this, what do you think can be done?

williamspatrick, 2024-09-04:
> remote:   https://gerrit.openbmc.org/c/openbmc/openbmc/+/74302 setup: allow simple execution for machine list [NEW]
> remote:   https://gerrit.openbmc.org/c/openbmc/openbmc/+/74303 setup: add subcommand for machine list [NEW]
> remote:   https://gerrit.openbmc.org/c/openbmc/openbmc/+/74304 meta-phosphor: generate supported machine list [NEW]

closed 2024-09-05 by geissonator