openbmc/openbmc #3927: openbmc version 2.13.0, 2.14.0, 2.15.0-dev does not work on qemu for qemux86-64 machine

sunzc, 2023-05-15:
**Unexpected behavior you saw**
I checked out openbmc version 2.13.0, 2.14.0, 2.14.0-rc, 2.15.0-dev and compile them for qemux86-64 and run on qemu. I saw the following instruction traps reported by QEMU
```
[  352.676568] traps: systemd-coredum[1673] trap invalid opcode ip:7f7f4ff784ed sp:7ffc47587f30 error:0 in libzstd.so.1.5.2[7f]
[  378.018155] systemd-journald[1510]: Data hash table of /run/log/journal/eb55e79c21014a1da79bd8e0c4edf4da/system.journal has.
[  378.026246] systemd-journald[1510]: /run/log/journal/eb55e79c21014a1da79bd8e0c4edf4da/system.journal: Journal header limits.
[  381.911006] traps: phosphor-log-ma[1686] trap invalid opcode ip:7f689a05671b sp:7ffe420fe660 error:0 in libstdc++.so.6.0.30]
[  382.654153] traps: systemd-coredum[1688] trap invalid opcode ip:7f581a9d24ed sp:7ffe8d9341e0 error:0 in libzstd.so.1.5.2[7f]
[  411.918393] traps: phosphor-log-ma[1700] trap invalid opcode ip:7f697197571b sp:7fffdca36880 error:0 in libstdc++.so.6.0.30]
[  412.662398] traps: systemd-coredum[1702] trap invalid opcode ip:7f8f992fe4ed sp:7ffdd516a970 error:0 in libzstd.so.1.5.2[7f]
[  418.315925] traps: xz[1698] trap invalid opcode ip:7efd273db376 sp:7ffea8b4e320 error:0 in liblzma.so.5.4.1[7efd273ca000+1d]
[  420.224369] traps: systemd-journal[1510] trap invalid opcode ip:7f56bc9934ed sp:7ffcec759bb0 error:0 in libzstd.so.1.5.2[7f]
```

I also can't log in as the instruction trap keep happening while typing.
```
See 'systemctl status xyz.openbmc_project.Logging.service' for details.

Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) nodistro.0 qemux86-64 ttyS0

qemux86-64 login: [  853.720190] systemd-journald[2189]: Data hash table of /run/log/journal/eb55e79c21014a1da79bd8e0c4edf4da/.
[  853.723745] systemd-journald[2189]: /run/log/journal/eb55e79c21014a1da79bd8e0c4edf4da/system.journal: Journal header limits.
[  862.472501] traps: phosphor-log-ma[2375] trap invalid opcode ip:7f786308971b sp:7ffdef103820 error:0 in libstdc++.so.6.0.30]
[  863.204325] traps: systemd-coredum[2377] trap invalid opcode ip:7f47e5aa74ed sp:7ffe7fbc8650 error:0 in libzstd.so.1.5.2[7f]
[  892.489277] traps: phosphor-log-ma[2477] trap invalid opcode ip:7f32f06c471b sp:7ffedea75bf0 error:0 in libstdc++.so.6.0.30]
[  893.248955] traps: systemd-coredum[2479] trap invalid opcode ip:7f277310e4ed sp:7fff653f5a30 error:0 in libzstd.so.1.5.2[7f]
```

**Expected behavior**
openbmc version 2.12.0 does not have the above problem. I can boot and log in on qemu.

**To Reproduce**
Steps to reproduce the behavior:
1. clone the openbmc repo: `git clone`
2. check out the listed version: `git br v2.15 2.15.0-dev; git co v2.15`
3. setup the machine type: `source setup qemux86-64 /your-path-to/openbmc-builds/qemux86-64-v2.15`
4. build: `umask 022` `bitbake obmc-phosphor-image`
5. run qemu (I tried qemu 7.2 and qemu 8.0)

williamspatrick, 2023-05-15:
I would suggest using one of the EVB models or at least qemuarm for test purposes. BMCs are not x86-64 chips so I don’t know of anyone that regularly tests this way. We’re just getting the kernel/qemu support from upstream Yocto the way you are building. 

Are you attempting to use QEMU-KVM or is this QEMU-TCG?  It could be that the qemux86-64 image is now compiling with opcodes your processor doesn’t support but I would have expected KVM to emulate those?

sunzc, 2023-05-15:
Thanks for the quick reply! Unfortunately, I need to use qemux86-64 to run openbmc image. I believe it is QEMU-TCG. Openbmc v2.12.0 works just fine. I'm wondering any change from 2.12.0 to 2.13.0, like compilation flags? I was search for a comprehensive changelog, but no luck yet. Any pointers? It seems it is the user library (libzstd.so) that is causing the problem.

williamspatrick, 2023-05-15:
Yes, there are tons of changes.  Every version is a new upstream Yocto version.  They're regularly refreshing all sorts of packages.  The compiler, libc, everything has likely been updated between those two.

williamspatrick, 2023-05-15:
How are you running qemu?  I just built a qemux86-64 image and then ran `runqemu slirp nographic serial` and it came up fine.  If I added `kvm` it tells me "You have no read or write permission on /dev/kvm", so it must be defaulting to TCG.

Can you try the `runqemu` that comes with openbmc / Yocto?

sunzc, 2023-05-16:
It works! that's unbelievable, thanks a lot! 

But why it matters ?

Before that I used a script to run qemu like below
```
./local/bmc-run-qemu.sh tmp/deploy/images/qemux86-64/obmc-phosphor-image-qemux86-64.ext4 tmp/deploy/images/qemux86-64/bzImage
```

`/local/bmc-run-qemu.sh`:
```
  1 #!/bin/bash
  2 CPU="-smp 4"
  3 
  4 DISK="-drive id=disk0,file=$1,if=none,format=raw -device virtio-blk-pci,drive=disk0"
  5 
  6 MEM="-m 8G"
  7 NET="-nic user,ipv6=off,hostfwd=:127.0.0.1:2222-:22,hostfwd=:127.0.0.1:2443-:443,hostfwd=udp:127.0.0.1:2623-:623"
  8 
  9 USB="-usb -device usb-mouse"
 10 
 11 GPU="-nographic"
 12 
 13 KERNEL="-kernel $2"
 14 KERNEL_ARGS='-append "root=/dev/vda console=ttyS0"'
 15 eval qemu-system-x86_64   ${CPU} ${MEM} ${DISK} ${NET} ${GPU} ${USB}  ${KERNEL} ${KERNEL_ARGS}
```

sunzc, 2023-05-16:
Update, my colleague helped me pinpoint the root cause of the issue: the `runqemu` specify cpu version with `-cpu IvyBridge` when run qemu while my script does not. The instruction trap was caused by that the zlib was confused about the cpu it is running on and choose the wrong instruction at runtime.

closed 2023-05-16 by williamspatrick