openbmc/openbmc #3928: Support For adding LM75 sensor in ipmitool sensor list

Manikandanrama, 2023-05-16:
@Kostr , @gtmills , @gilbert225 

I am trying to list the lm75 sensor value in ipmitool, and also verified using  **cat /sys/bus/i2c/devices/8-004d/hwmon/hwmon0/temp1_input**   command, it showed the value of lm75 .
In order to list in ipmitool sensor list, i have configured in entity manager json files, but it didn't get updated.
Then i have configured the config.yaml file in phosphor-ipmi-sensor-inventory directory, still no improvement, 
Can you please suggest me the steps to proceed further if i have missed.

Kindly guide on this.

Thanks
Manikandan

Kostr, 2023-05-16:
First to step is to ask your question in Discord and not in a GitHub issue
https://github.com/openbmc/openbmc#contact

Second step, don't tag random people in your questions :)

As for the question, there is not enough information.
- Do you use "entity-manager + dbus-sensors" ?
- Do you see your sensor on DBus? Do you see it in webui?

As an example of a working config you can check https://github.com/openbmc/openbmc/blob/master/meta-amd/meta-ethanolx/recipes-phosphor/configuration/ethanolx-yaml-config.bb which installs `ethanolx-ipmi-sensors.yaml`  (https://github.com/openbmc/openbmc/blob/master/meta-amd/meta-ethanolx/recipes-phosphor/configuration/ethanolx-yaml-config/ethanolx-ipmi-sensors.yaml)

Also the necessary configuration for the YAML enablement is in the https://github.com/openbmc/openbmc/blob/master/meta-amd/meta-ethanolx/recipes-phosphor/ipmi/phosphor-ipmi-host_%25.bbappend

Manikandanrama, 2023-05-16:
Thanks for the Update, @Kostr  Sorry for querying here

closed 2023-06-07 by Manikandanrama