openbmc/openbmc #3930: Issue with web and REST API

AlexD202, 2023-05-17:
**Note on Bugs**
I am trying to access the BMC module using the web and REST API, but both of them don't seem to be working for me.
I checked out the code and followed steps to build the image for BMC. Loaded the image onto BMC. I can ping and ssh into the BMC from the host, but I am not able to open its page using https://BMC_IP  (I get unauthorized message). I am also unable to use curl to connect to loginto the BMC:
HOST# export bmc=x.x.x.x
HOST# curl -c cjar -b cjar -k -H "Content-Type: application/json" -X POST https://${bmc}/login -d "{\"data\": [ \"root\", \"0penBmc\" ] }"
curl: (35) OpenSSL SSL_connect: SSL_ERROR_ZERO_RETURN in connection to x.x.x.x:443

**Unexpected behavior you saw**
webpage shows as unauthorized and nothing shows up.
curl shows SSL_ERROR_ZERO_RETURN when trying to use REST API calls.

**Expected behavior**
To be able to use web and REST API to configure the module

**To Reproduce**
1. Checkout the code and follow the steps to build an image for ast2500
2. Load the image on to the BMC module
3. access the webpage by going to https://BMC_IP 

**OpenBMC Information:**
root@evb-ast2500:~# cat /etc/os-release
ID=openbmc-phosphor
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="2.14.0-dev"
VERSION_ID=2.14.0-dev-1020-g18caa9759-dirty
VERSION_CODENAME="langdale"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 2.14.0-dev"
BUILD_ID="20230510214217"
OPENBMC_TARGET_MACHINE="evb-ast2500"
EXTENDED_VERSION="2.14.0-dev-1020-g18caa9759-dirty"

root@evb-ast2500:~# uname -a
Linux evb-ast2500 6.1.15-580639a-dirty-914a4fb #1 Thu Apr  6 00:55:09 UTC 2023 armv6l GNU/Linux

**Additional context**
root@evb-ast2500:~# ps | grep bmcweb
  177 root     13008 S    /usr/bin/bmcweb
  184 root     12148 S    /usr/bin/phosphor-certificate-manager --endpoint ldap --path /etc/ssl/certs/authority --type authority --unit bmcweb.service
  185 root     12392 S    /usr/bin/phosphor-certificate-manager --endpoint https --path /etc/ssl/certs/https/server.pem --type server --unit bmcweb.serv
 2645 root      2924 S    grep bmcweb

root@evb-ast2500:~# systemctl status bmcweb
* bmcweb.service - Start bmcweb server
     Loaded: loaded (/lib/systemd/system/bmcweb.service; enabled; preset: enabled)
     Active: active (running) since Thu 2023-03-09 08:29:15 UTC; 1 day 23h ago
TriggeredBy: * bmcweb.socket
    Process: 1504 ExecReload=kill -s HUP $MAINPID (code=exited, status=0/SUCCESS)
   Main PID: 177 (bmcweb)
        CPU: 6.325s
     CGroup: /system.slice/bmcweb.service
             `-177 /usr/bin/bmcweb

Mar 10 09:12:20 BMC bmcweb[177]: Generating EC key
Mar 10 09:12:20 BMC bmcweb[177]: Generating x509 Certificate
Mar 10 09:12:20 BMC systemd[1]: Reloading Start bmcweb server...
Mar 10 09:12:20 BMC systemd[1]: Reloaded Start bmcweb server.
Mar 10 09:13:38 evb-ast2500 bmcweb[177]: Checking certs in file /etc/ssl/certs/https/server.pem
Mar 10 09:13:38 evb-ast2500 bmcweb[177]: Generating new keys
Mar 10 09:13:38 evb-ast2500 bmcweb[177]: Generating EC key
Mar 10 09:13:38 evb-ast2500 bmcweb[177]: Generating x509 Certificate
Mar 10 09:13:38 evb-ast2500 systemd[1]: Reloading Start bmcweb server...
Mar 10 09:13:38 evb-ast2500 systemd[1]: Reloaded Start bmcweb server.

root@evb-ast2500:~# journalctl -u bmcweb
Mar 09 08:29:15 evb-ast2500 systemd[1]: Started Start bmcweb server.
Mar 10 04:40:30 evb-ast2500 bmcweb[177]: pam_tally2(webserver:auth): pam_get_uid; no such user
Mar 10 04:40:30 evb-ast2500 bmcweb[177]: pam_unix(webserver:auth): check pass; user unknown
Mar 10 04:40:30 evb-ast2500 bmcweb[177]: pam_unix(webserver:auth): authentication failure; logname= uid=0 euid=0 tty= ruser= rhost=
Mar 10 04:40:30 evb-ast2500 bmcweb[177]: pam_ldap(webserver:auth): error opening connection to nslcd: No such file or directory
Mar 10 04:41:24 evb-ast2500 bmcweb[177]: pam_succeed_if(webserver:auth): requirement "user ingroup redfish" was met by user "root"
Mar 10 04:42:05 evb-ast2500 bmcweb[177]: pam_succeed_if(webserver:auth): requirement "user ingroup redfish" was met by user "root"
Mar 10 04:42:46 evb-ast2500 bmcweb[177]: pam_succeed_if(webserver:auth): requirement "user ingroup redfish" was met by user "root"
Mar 10 05:05:49 evb-ast2500 bmcweb[177]: pam_succeed_if(webserver:auth): requirement "user ingroup redfish" was met by user "root"
Mar 10 05:06:45 evb-ast2500 bmcweb[177]: pam_succeed_if(webserver:auth): requirement "user ingroup redfish" was met by user "root"
Mar 10 09:12:20 BMC bmcweb[177]: Checking certs in file /etc/ssl/certs/https/server.pem
Mar 10 09:12:20 BMC bmcweb[177]: Generating new keys
Mar 10 09:12:20 BMC bmcweb[177]: Generating EC key
Mar 10 09:12:20 BMC bmcweb[177]: Generating x509 Certificate
Mar 10 09:12:20 BMC systemd[1]: Reloading Start bmcweb server...
Mar 10 09:12:20 BMC systemd[1]: Reloaded Start bmcweb server.
Mar 10 09:13:38 evb-ast2500 bmcweb[177]: Checking certs in file /etc/ssl/certs/https/server.pem
Mar 10 09:13:38 evb-ast2500 bmcweb[177]: Generating new keys
Mar 10 09:13:38 evb-ast2500 bmcweb[177]: Generating EC key
Mar 10 09:13:38 evb-ast2500 bmcweb[177]: Generating x509 Certificate
Mar 10 09:13:38 evb-ast2500 systemd[1]: Reloading Start bmcweb server...
Mar 10 09:13:38 evb-ast2500 systemd[1]: Reloaded Start bmcweb server.

AlexD202, 2023-05-18:
Curl command with -v option .
HOST# curl -v -c cjar -k -X POST -H "Content-Type: application/json" -d '{"data": [ "root","0penBmc" ] }' https://169.254.10.134/login        
Note: Unnecessary use of -X or --request, POST is already inferred.
* TLSv1.2 (OUT), TLS handshake, Client hello (1):
* TLSv1.2 (IN), TLS alert, handshake failure (552):
* OpenSSL SSL_connect: SSL_ERROR_ZERO_RETURN in connection to 169.254.10.134:443
curl: (35) OpenSSL SSL_connect: SSL_ERROR_ZERO_RETURN in connection to 169.254.10.134:443

edtanous, 2023-05-19:
TLSv1.2 (IN), TLS alert, handshake failure (552):

This would imply that:
1. You don't have certificates that are valid
2. You have certificates that aren't valid, but haven't ignored them.
3. The version of openssl on the system you're running on is incompatible with the version being run on the BMC.

Have you been able to make this workflow work in the past?

AlexD202, 2023-05-23:
Do i need to generate my own certificate file or the one that comes with the image under /etc/ssl/https/certs should work?

edtanous, 2023-05-24:
Can you try to replicate the test harness results under qemu?  They run as part of every recipe bump, so I would be surprised if we had a systemic issue here.  Instead, I suspect something about your environment is different, but we can’t really help with that.

AlexD202, 2023-05-24:
I will try that. Do i need to do anything when compiling to have the web included  in the image or is it included by default when i run the following?
. Setup evb-ast2500
bitbake obmc-phosphor-image

AlexD202, 2023-05-26:
I got the REST login to work.
The certificate that is automatically generated under /etc/ssl/certs/https/server.pem did not work. I had to make my own certificate.

Now I can log in using the following call:
curl -c cjar -b cjar -k -H "Content-Type: application/json" -X POST https://xx.xx.xx.xx/login -d "{\"data\": [ \"root\", \"0penBmc\" ] }"

returns:
{
  "data": "User 'root' logged in",
  "message": "200 OK",
  "status": "ok"
}

Any other API call returns not found:
curl -c cjar -b cjar -k -H "Content-Type: application/json" -X GET   https://xx.xx.xx.xx/xyz/openbmc_project/list

returns: 
Not Found

Any help would be appreciated...

edtanous, 2023-05-26:
I'm guessing you didn't enable the 'rest' option in meson.  I'm going to close this out, as the reported issue seems like it was not a problem, and seems to be an issue with your environment.  The best forum for these kinds of questions is likely discord or the mailing list.

closed 2023-05-26 by edtanous