openbmc/openbmc #3931: How To Display LM75 Value in Webui

Manikandanrama, 2023-05-18:
hi @derick-montague  

i have configure the lm75 temperature sensor Value in ipmtool.now i would like to show that Value in Webui. i don't know how to start from begginig.it also show in the busctl commad,but i dont know how to display the value in webui.could share the steps to show the sensor values in webui.

Thanks,
Manikandan

williamspatrick, 2023-05-18:
Please don't tag individuals in bug reports.  Derick ceased working on the project in Nov 2021 and was removed from the OWNERS file with ae09a4b3c03dcf86616820ab9d8c0c7ef9ebc69c .

This is not a webui problem.  Webui shows what is available in Redfish.  If it is in dbus but not in webui, you need to investigate the Redfish API.

williamspatrick, 2023-05-18:
The template for this repository has a template for "Questions / Support".  It clearly says:

> # STOP! READ BELOW.

> We do not field questions or feature requests by Github Issues!

Please don't ignore such things as it wastes people's time.

closed 2023-05-18 by williamspatrick