openbmc/openbmc #3932: python3 hatch-vcs-native fails to build

shenki, 2023-05-22:
Unsure if this is a oe/yocto issue, an openbmc issue, or a distro issue. I've seen it a few times so recording it here as an issue.

tmp/work/aarch64-linux/python3-hatch-vcs-native/0.3.0-r0/temp/log.do_compile.516284
```
DEBUG: Executing shell function do_compile
* Getting build dependencies for wheel...

Traceback (most recent call last):
  File "tmp/work/aarch64-linux/python3-hatch-vcs-native/0.3.0-r0/recipe-sysroot-native/usr/lib/python3.11/site-packages/build/__main__.py", line 375, in main
    built = build_call(
            ^^^^^^^^^^^
  File "tmp/work/aarch64-linux/python3-hatch-vcs-native/0.3.0-r0/recipe-sysroot-native/usr/lib/python3.11/site-packages/build/__main__.py", line 208, in build_package
    out = _build(isolation, builder, outdir, distribution, config_settings, skip_dependency_check)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "tmp/work/aarch64-linux/python3-hatch-vcs-native/0.3.0-r0/recipe-sysroot-native/usr/lib/python3.11/site-packages/build/__main__.py", line 147, in _build
    return _build_in_current_env(builder, outdir, distribution, config_settings, skip_dependency_check)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "tmp/work/aarch64-linux/python3-hatch-vcs-native/0.3.0-r0/recipe-sysroot-native/usr/lib/python3.11/site-packages/build/__main__.py", line 127, in _build_in_current_env
    missing = builder.check_dependencies(distribution)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "tmp/work/aarch64-linux/python3-hatch-vcs-native/0.3.0-r0/recipe-sysroot-native/usr/lib/python3.11/site-packages/build/__init__.py", line 369, in check_dependencies
    return {u for d in dependencies for u in check_dependency(d)}
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "tmp/work/aarch64-linux/python3-hatch-vcs-native/0.3.0-r0/recipe-sysroot-native/usr/lib/python3.11/site-packages/build/__init__.py", line 369, in <setcomp>
    return {u for d in dependencies for u in check_dependency(d)}
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "tmp/work/aarch64-linux/python3-hatch-vcs-native/0.3.0-r0/recipe-sysroot-native/usr/lib/python3.11/site-packages/build/__init__.py", line 179, in check_dependency
    if req.specifier and not req.specifier.contains(dist.version, prereleases=True):
                             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "tmp/work/aarch64-linux/python3-hatch-vcs-native/0.3.0-r0/recipe-sysroot-native/usr/lib/python3.11/site-packages/packaging/specifiers.py", line 905, in contains
    item = Version(item)
           ^^^^^^^^^^^^^
  File "tmp/work/aarch64-linux/python3-hatch-vcs-native/0.3.0-r0/recipe-sysroot-native/usr/lib/python3.11/site-packages/packaging/version.py", line 196, in __init__
    match = self._regex.search(version)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: expected string or bytes-like object, got 'NoneType'

ERROR expected string or bytes-like object, got 'NoneType'
```

williamspatrick, 2023-05-22:
It’s definitely a Yocto issue. I’ve seen the same thing on non-OpenBMC Yocto projects.

chaosding, 2023-09-25:
I also have this issue after syncing OpenBMC to [commit 59c98f4](https://github.com/openbmc/openbmc/commit/59c98f4cab8df2fa9c7bd483a79065368754844c).
I found that the "python3-hatch-vcs-native" use an older "python3-hatchling" version.
 ```
cat "tmp/work/x86_64-linux/python3-hatch-vcs-native/0.3.0--0/image/mnt/Data/obmc/openbmc/build/${machine_name}/tmp/work/x86_64-linux/python3-hatch-vcs-native/0.3.0-r0/recipe-sysroot-native/usr/lib/python3.11/site-packages/hatch_vcs-0.3.0.dist-info/WHEEL"
 
Wheel-Version: 1.0
Generator: hatchling 1.12.2
Root-Is-Purelib: true
Tag: py3-none-any
 ```
The "Generator" version should be "hatchling 1.14.1"

My solution is below:
```
bitbake -c do_cleanall python3-hatchling python3-hatch-vcs python3-hatch-vcs-native
bitbake python3-hatch-vcs-native
```
Maybe "clean build" will avoid this issue.

williamspatrick, 2023-09-25:
This is almost certainly an issue with upstream Yocto python packaging.  We don't have any bbclass modifications that I know of that would cause this.  Next time someone sees it, it would be good if we can report the issue upstream.

chaosding, 2023-10-04:
This problem also happens with another python package.
I tried to build the python3-jsonschema package, but I found the generator in the "WHEEL" file is an error version.

I think this issue would affect all Python packages that depend on the hatchling package.

williamspatrick, 2024-12-20:
This needs to be tackled upstream if it is still an issue.  I don't think I've seen this in a while.

closed 2024-12-20 by williamspatrick