openbmc/openbmc #3933: Unterminated String Literal Building Python 3.9+

KeepSummerSaf3, 2023-05-26:
Seems there are a few problems that crop up when using Python 3.9+

The collections abstract base classes (collections.abc) module was formerly part of the collections module, but I think, was moved in Python 3.8.11. This means several changes are required for python files within the bitbake config directory (see poky/bitbake/lib/bb/)

Even after applying fixes such that collections.abc can import the abstract base class "Mapping" and "MutableMapping"  an expansion error will occur with some of the bb expansion variables (SRCREV, FILESPATH, etc.) resulting in an unterminated string literal when attempting to pull or build various recipes.

```
ERROR: ExpansionError during parsing /drone/src/meta-phosphor/recipes-phosphor/mrw/mrw-native_git.bb
Traceback (most recent call last):
  File "/drone/src/poky/bitbake/lib/bb/data_smart.py", line 587, in DataSmart.getVar(var='FILESPATH', expand=True, noweakdefault=False, parsing=False):
         def getVar(self, var, expand=True, noweakdefault=False, parsing=False):
    >        return self.getVarFlag(var, "_content", expand, noweakdefault, parsing)
     
  File "/drone/src/poky/bitbake/lib/bb/data_smart.py", line 778, in DataSmart.getVarFlag(var='FILESPATH', flag='_content', expand=True, noweakdefault=False, parsing=False, retparser=False):
             if expand or retparser:
    >            parser = self.expandWithRefs(value, cachename)
             if expand:
  File "/drone/src/poky/bitbake/lib/bb/data_smart.py", line 416, in DataSmart.expandWithRefs(s='${@base_set_filespath(["/drone/src/meta-phosphor/recipes-phosphor/mrw/mrw-1.0+git${SRCPV}", "/drone/src/meta-phosphor/recipes-phosphor/mrw/mrw", "/drone/src/meta-phosphor/recipes-phosphor/mrw/files"], d)}', varname='FILESPATH'):
                     tb = sys.exc_info()[2]
    >                raise ExpansionError(varname, s, exc).with_traceback(tb) from exc
     
  File "/drone/src/poky/bitbake/lib/bb/data_smart.py", line 403, in DataSmart.expandWithRefs(s='${@base_set_filespath(["/drone/src/meta-phosphor/recipes-phosphor/mrw/mrw-1.0+git${SRCPV}", "/drone/src/meta-phosphor/recipes-phosphor/mrw/mrw", "/drone/src/meta-phosphor/recipes-phosphor/mrw/files"], d)}', varname='FILESPATH'):
                     try:
    >                    s = __expand_python_regexp__.sub(varparse.python_sub, s)
                     except SyntaxError as e:
  File "/drone/src/poky/bitbake/lib/bb/data_smart.py", line 114, in VariableParse.python_sub(match=<re.Match object; span=(0, 89), match='${@base_set_filespath(["/drone/src/meta-phosphor/>):
                     varname = '<expansion>'
    >            codeobj = compile(code.strip(), varname, "eval")

bb.data_smart.ExpansionError: Failure expanding variable FILESPATH, expression was ${@base_set_filespath(["/drone/src/meta-phosphor/recipes-phosphor/mrw/mrw-1.0+git${SRCPV}", "/drone/src/meta-phosphor/recipes-phosphor/mrw/mrw", "/drone/src/meta-phosphor/recipes-phosphor/mrw/files"], d)} which triggered exception SyntaxError: unterminated string literal (detected at line 1) (Var <FILESPATH>, line 1)
```

williamspatrick, 2023-05-26:
For which version are you building?  Something off master or a release branch?

KeepSummerSaf3, 2023-06-01:
After further review, this issue is not present in the master branch of openbmc. Closing this issue.; my apologies.

closed 2023-06-01 by KeepSummerSaf3