openbmc/openbmc #3936: Need info on openbmc build target on hardware target SYS-220H-TN24R (supermicro).

PrajwalMallikarjunS, 2023-06-13:
Hi,
We are exploring openbmc on supermicro (SYS-220H-TN24R) server.
As I could not see build target for hardware SYS-220H-TN24R, Need an information to build openbmc and flash on supermicroserver(SYS-220H-TN24R).
Thanks,
Prajwal

amboar, 2023-06-13:
This is a support request. The corresponding issue template says:

```
# STOP! READ BELOW.

We do not field questions or feature requests by Github Issues!

Questions or feature requests should be discussed on the mailing list or
Discord. Please see https://github.com/openbmc/openbmc#contact for info
on how to contact us.

Github Issues in openbmc/openbmc are for bugs only!
```

Please follow the suggestions there.

closed 2023-06-13 by amboar

zevweiss, 2023-06-13:
Also note that this is a common enough question that it's addressed in the README: https://github.com/openbmc/openbmc/#will-openbmc-run-on-my-acme-server-corp-xyz5000-motherboard