openbmc/openbmc #3937: qemux86 target fails to build on 2.14

macpijan, 2023-06-14:
**Note on Bugs**

I'm trying to follow: https://github.com/openbmc/docs/blob/master/development/dev-environment.md#alternative-yocto-qemu on 2.14 tag.

Build commands used:

```
source setup romulus
MACHINE=qemux86 bitbake obmc-phosphor-image
```

**Unexpected behavior you saw**

```
nux/12.2.0/ld: queue.os:/usr/src/debug/liburing/2.3+2.4+gitAUTOINC+f5cac78e62-r0/src/queue.c:388: more undefined references to `__stack_chk_fail_local' follow
| collect2: error: ld returned 1 exit status
| make[1]: *** [Makefile:82: liburing.so.2.4] Error 1
| make[1]: *** Waiting for unfinished jobs....
| make[1]: Leaving directory '/home/macpijan/projects/github/3mdeb/openbmc/build/romulus/tmp/work/core2-32-openbmc-linux/liburing/2.3+2.4+gitAUTOINC+f5cac78e62-r0/git/src'
| ERROR: oe_runmake failed
| WARNING: exit code 1 from a shell command.
| make: *** [Makefile:10: all] Error 2
ERROR: Task (/home/macpijan/projects/github/3mdeb/openbmc/meta-openembedded/meta-oe/recipes-support/liburing/liburing_2.3.bb:do_compile) failed with exit code '1'
```

**Expected behavior**

No such errors and the image builds correctly.

**To Reproduce**
Steps to reproduce the behavior:

```
source setup romulus
MACHINE=qemux86 bitbake obmc-phosphor-image
```

**Screenshots**

N/A

**OpenBMC Information:**

`67c9d4e715c7 (tag: 2.14.0) mickledore: subtree updates`

**Additional context**

Can be worked around by disabling stack protector, for example with:

```
diff --git a/meta-openembedded/meta-oe/recipes-support/liburing/liburing_2.3.bb b/meta-openembedded/meta-oe/recipes-support/liburing/liburing_2.3.bb
index 7d8c2f596d46..a9bbf41faddf 100644
--- a/meta-openembedded/meta-oe/recipes-support/liburing/liburing_2.3.bb
+++ b/meta-openembedded/meta-oe/recipes-support/liburing/liburing_2.3.bb
@@ -19,7 +19,7 @@ PV .= "+2.4+git${SRCPV}"
 S = "${WORKDIR}/git"
 
 DEPENDS:append:libc-musl = " libucontext"
-XCFLAGS = "-pthread"
+XCFLAGS = "-pthread -fno-stack-protector"
 XCFLAGS:append:libc-musl = " -lucontext"
```

macpijan, 2023-06-14:
Actually, the `liburing` build still fails for `qemux86` (but succeeds for the `romulus` MACHINE). Still looking into that.

williamspatrick, 2023-06-14:
I’m slightly confused why you ran setup with Romulus but then explicitly set machine to qemux86. Do you have this issue when using qemux86 as the setup parameter directly?

williamspatrick, 2023-06-14:
> That is what the linked docs suggest - just trying to follow it. Could you please point me how to build for qemu if this is not valid?

Interesting.  I'd have to try to work backwards as to what the original author of this doc was trying to achieve but it isn't how I've done it before.

`romulus` is a particular machine.  You can replace that parameter in the setup with any other supported machine.  So you can just directly do `qemux86`.  Since you've already built something in that build directory you passed as the last parameter, I'd replace that with a fresh directory name so you don't end up with some pollution between the romulus set up on the qemux86 one you're trying to build now.

To be honest I'm not sure it will make a difference one way or another with the specific error you're reporting.  This seems like something with one of the upstream recipes but I'm surprised nobody else has noticed this.

macpijan, 2023-06-20:
No, it does not make difference. I tried also clean build as you have suggested.

The workaround allowing the build to pass was:

```
+SECURITY_CFLAGS:remove:class-target = "-fstack-protector-strong"
```

I'm also surprised it was not noticed in Yocto upstream. I've yet to see if it reproduces on pure OE layers (with no OpenBMC). I failed to see such bug report, either.

Nevertheless, when trying to run qemu with: `runqemu slirp nographic`, it panics in a different ways, such as;

```
        Starting Telemetry...
         Starting Phosphor User Manager...
[  OK  ] Started Phosphor sysfs LED controller.
[  OK  ] Started Phosphor sysfs LED controller.
[  OK  ] Started Phosphor sysfs LED controller.
[    6.477880] ------------[ cut here ]------------
[    6.478032] Bad FPU state detected at restore_fpregs_from_fpstate+0x3a/0x70, reinitializing FPU registers.
[    6.478493] WARNING: CPU: 1 PID: 150 at arch/x86/mm/extable.c:126 fixup_exception+0x38e/0x3e0
[    6.478952] Modules linked in: sch_fq_codel
[    6.479267] CPU: 1 PID: 150 Comm: systemd-journal Not tainted 6.1.25-yocto-s-dirty-581dc1a #1
[    6.479464] Hardware name: QEMU Standard PC (Q35 + ICH9, 2009), BIOS rel-1.16.1-0-g3208b098f51a-prebuilt.qemu.org 04/01/2014
[    6.479899] EIP: fixup_exception+0x38e/0x3e0
[    6.480063] Code: 89 7e 30 e9 ae fd ff ff 8d 74 26 00 0f 0b ba 4c 19 27 de e9 b6 fe ff ff c6 05 aa 99 13 de 01 50 68 b4 e0 e7 dd e8 82 f0 00 00 <0f> 0b 5f 58 e9 03 fe ff ff 80 3d a8 99f
[    6.480538] EAX: 0000005e EBX: ddfbde14 ECX: f5ba9c7c EDX: f5ba9c78
[    6.480783] ESI: c25c1ee8 EDI: 0000000d EBP: c25c1e60 ESP: c25c1de0
[    6.480967] DS: 007b ES: 007b FS: 00d8 GS: 0033 SS: 0068 EFLAGS: 00000086
[    6.481174] CR0: 80050033 CR2: 007b018c CR3: 02763000 CR4: 001506d0
[    6.481501] Call Trace:
[    6.481755]  ? restore_fpregs_from_fpstate+0x3a/0x70
[    6.481921]  ? __switch_to_asm+0x59/0xf0
[    6.482014]  ? __switch_to_asm+0x53/0xf0
[    6.482090]  ? __switch_to_asm+0x4d/0xf0
[    6.482177]  ? __switch_to_asm+0x47/0xf0
[    6.482264]  ? __switch_to_asm+0x41/0xf0
[    6.482351]  ? finish_task_switch.isra.0+0x8a/0x260
[    6.482458]  ? __schedule+0x294/0x710
[    6.482550]  exc_general_protection+0x91/0x390
[    6.482677]  ? exc_bounds+0x90/0x90
[    6.482771]  handle_exception+0x133/0x133
[    6.482917] EIP: restore_fpregs_from_fpstate+0x3a/0x70
[    6.483670] Code: 89 d0 89 ca eb 0a 8d 76 00 db e2 0f 77 db 45 f4 3e 8d 74 26 00 8b 4d f4 23 05 e8 61 f5 dd 23 15 ec 61 f5 dd 8d 79 40 0f ae 2f <83> c4 04 5b 5f 5d c3 8d b4 26 00 00 004
[    6.483703] EAX: 00000007 EBX: c2cac500 ECX: c2cacd00 EDX: 00000000
[    6.483714] ESI: c2caccc0 EDI: c2cacd40 EBP: c25c1f50 ESP: c25c1f44
[    6.483720] DS: 007b ES: 007b FS: 00d8 GS: 0033 SS: 0068 EFLAGS: 00000046
[    6.483754]  ? exc_bounds+0x90/0x90
[    6.483774]  ? exc_bounds+0x90/0x90
[    6.483783]  ? restore_fpregs_from_fpstate+0x37/0x70
[    6.483795]  switch_fpu_return+0x44/0xd0
[    6.483815]  exit_to_user_mode_prepare+0x135/0x170
[    6.483826]  syscall_exit_to_user_mode+0x1b/0x40
[    6.483834]  __do_fast_syscall_32+0x5a/0xc0
[    6.483843]  do_fast_syscall_32+0x29/0x60
[    6.483853]  do_SYSENTER_32+0x15/0x20
[    6.483861]  entry_SYSENTER_32+0x98/0xf1
[    6.483869] EIP: 0xb7f5b549
[    6.484053] Code: 03 74 c0 01 10 05 03 74 b8 01 10 06 03 74 b4 01 10 07 03 74 b0 01 10 08 03 74 d8 01 00 00 00 00 00 51 52 55 89 e5 0f 34 cd 80 <5d> 5a 59 c3 90 90 90 90 8d 76 00 58 b86
[    6.484060] EAX: 00000001 EBX: 00000007 ECX: 00502290 EDX: 0000003e
[    6.484065] ESI: ffffffff EDI: 004eca10 EBP: b7c01e2c ESP: bf9d6f30
[    6.484071] DS: 007b ES: 007b FS: 0000 GS: 0033 SS: 007b EFLAGS: 00000246
[    6.484154] ---[ end trace 0000000000000000 ]---
[    6.485431] general protection fault, maybe for address 0x1: 0000 [#1] PREEMPT SMP
[    6.485525] CPU: 1 PID: 215 Comm: nscd Tainted: G        W          6.1.25-yocto-s-dirty-581dc1a #1
[    6.485538] Hardware name: QEMU Standard PC (Q35 + ICH9, 2009), BIOS rel-1.16.1-0-g3208b098f51a-prebuilt.qemu.org 04/01/2014
[    6.485555] EIP: entry_SYSENTER_32+0xe0/0xf1
[    6.485574] Code: 8b 54 24 30 8b 4c 24 3c 8e 64 24 24 5b 83 c4 08 5e 5f 5d 89 c4 eb 0b 0f 20 d8 0d 00 10 00 00 0f 22 d8 0f ba 34 24 09 9d 58 fb <0f> 35 c7 44 24 24 00 00 00 00 eb d2 6a6
[    6.485583] EAX: 00000001 EBX: 00000004 ECX: bfb17620 EDX: b7edb549
[    6.485588] ESI: 00007518 EDI: 00000000 EBP: b7ec7e2c ESP: ff82f000
[... 360 lines cut ...]
[    7.309466] Aborting core
[    7.310768] general protection fault, maybe for address 0x0: 0000 [#14] PREEMPT SMP
[    7.310992] CPU: 2 PID: 1 Comm: systemd Tainted: G      D W          6.1.25-yocto-s-dirty-581dc1a #1
[    7.311190] Hardware name: QEMU Standard PC (Q35 + ICH9, 2009), BIOS rel-1.16.1-0-g3208b098f51a-prebuilt.qemu.org 04/01/2014
[    7.311428] EIP: entry_SYSENTER_32+0xe0/0xf1
[    7.311538] Code: 8b 54 24 30 8b 4c 24 3c 8e 64 24 24 5b 83 c4 08 5e 5f 5d 89 c4 eb 0b 0f 20 d8 0d 00 10 00 00 0f 22 d8 0f ba 34 24 09 9d 58 fb <0f> 35 c7 44 24 24 00 00 00 00 eb d2 6a6
[    7.311970] EAX: 00000000 EBX: 0000003b ECX: bf841660 EDX: b7f33549
[    7.312201] ESI: b7a07e2c EDI: b750d6b0 EBP: 00000020 ESP: ff85a000
[    7.312354] DS: 007b ES: 007b FS: 0000 GS: 0033 SS: 0068 EFLAGS: 00000246
[    7.312510] CR0: 80050033 CR2: b792c000 CR3: 01020000 CR4: 001506d0
[    7.312708] Call Trace:
[    7.312790] Modules linked in: sch_fq_codel
[    7.313318] ---[ end trace 0000000000000000 ]---
[    7.313464] EIP: entry_SYSENTER_32+0xe0/0xf1
[... cut ...]
[    7.314298] ESI: ffffffff EDI: 0049adb0 EBP: b7db6e2c ESP: ff85a000
[    7.314427] DS: 007b ES: 007b FS: 0000 GS: 0033 SS: 0068 EFLAGS: 00000246
[    7.314544] CR0: 80050033 CR2: b792c000 CR3: 01020000 CR4: 001506d0
[    7.314805] Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b
[    7.315658] Kernel Offset: disabled
[    7.315835] ---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b ]---
```

I'm wondering if `qemux86` is the one that I should use, or maybe another qemu target has better chance here.

williamspatrick, 2023-06-20:
> I'm wondering if qemux86 is the one that I should use, or maybe another qemu target has better chance here.

If you're not confident in what you're doing it is probably better to start with a fully functional machine image that is supported in QEMU.  Romulus and Witherspoon are two such machines.  That will give you something complete to play with.

Beyond that, what is it you're trying to accomplish?  The QEMU targets aren't really full configurations and so they're rarely used by people doing OpenBMC development.

macpijan, 2023-06-20:
> If you're not confident in what you're doing it is probably better to start with a fully functional machine image that is supported in QEMU. Romulus and Witherspoon are two such machines.

Thanks, that is what I probably need. The initial document wasa bit misleading then, as it suggested to build for qemux86 at some point.

> Beyond that, what is it you're trying to accomplish?

Build and run anything with OpenBMC in QEMU, with no extra hardware.

williamspatrick, 2023-06-20:
> Build and run anything with OpenBMC in QEMU, with no extra hardware.

I'd suggest picking one of the machine configurations that shows up in QEMU then:

```
$ qemu-system-arm --machine help | grep bmc   
bletchley-bmc        Facebook Bletchley BMC (Cortex-A7)
fby35-bmc            Facebook fby35 BMC (Cortex-A7)
fp5280g2-bmc         Inspur FP5280G2 BMC (ARM1176)
fuji-bmc             Facebook Fuji BMC (Cortex-A7)
g220a-bmc            Bytedance G220A BMC (ARM1176)
kudo-bmc             Kudo BMC (Cortex-A9)
mori-bmc             Mori BMC (Cortex-A9)
palmetto-bmc         OpenPOWER Palmetto BMC (ARM926EJ-S)
qcom-dc-scm-v1-bmc   Qualcomm DC-SCM V1 BMC (Cortex A7)
qcom-firework-bmc    Qualcomm DC-SCM V1/Firework BMC (Cortex A7)
quanta-gbs-bmc       Quanta GBS (Cortex-A9)
quanta-q71l-bmc      Quanta-Q71l BMC (ARM926EJ-S)
rainier-bmc          IBM Rainier BMC (Cortex-A7)
romulus-bmc          OpenPOWER Romulus BMC (ARM1176)
sonorapass-bmc       OCP SonoraPass BMC (ARM1176)
supermicrox11-bmc    Supermicro X11 BMC (ARM926EJ-S)
tacoma-bmc           OpenPOWER Tacoma BMC (Cortex-A7)
witherspoon-bmc      OpenPOWER Witherspoon BMC (ARM1176)
```

closed 2023-09-25 by williamspatrick