openbmc/openbmc #3938: openBMC AST2500-EVB auto reboot

AmirRMoezi, 2023-07-01:
I have built openBMC for AST2500-EVB and when I run the .mtd image by qemu, the u-boot runs successfully but during the linux kernel boot, it stops automatically and u-boot runs again. This loops continuous infinitely. does any body know what is the problem? the terminal output is shown here. This happens even when running the FITimage on the real board.

root@osboxes:/home/osboxes/Desktop/openbmc/build/evb-ast2500/tmp/deploy/images/evb-ast2500# qemu-system-arm -M ast2500-evb -nic user -drive file=obmc-phosphor-image-evb-ast2500-20230612144736.static.mtd,format=raw,if=mtd -nographic

U-Boot 2019.04 (May 30 2023 - 06:05:47 +0000)

SOC : AST2500-A1 
RST : Power On 
LPC Mode : SIO:Enable : SuperIO-2e
Eth : MAC0: RGMII, , MAC1: RGMII, 
Model: AST2500 EVB
DRAM:  448 MiB (capacity:512 MiB, VGA:64 MiB, ECC:off)
MMC:   sdhci_slot0@100: 0, sdhci_slot1@200: 1
Loading Environment from SPI Flash... SF: Detected w25q256 with page size 256 Bytes, erase size 4 KiB, total 32 MiB
OK
In:    serial@1e784000
Out:   serial@1e784000
Err:   serial@1e784000
Net:   eth0: ethernet@1e660000
Warning: ethernet@1e680000 (eth1) using random MAC address - 92:c2:88:7d:f4:79
, eth1: ethernet@1e680000
Hit any key to stop autoboot:  0 
## Loading kernel from FIT Image at 20080000 ...
   Using 'conf-aspeed-ast2500-evb.dtb' configuration
   Trying 'kernel-1' kernel subimage
     Description:  Linux kernel
     Type:         Kernel Image
     Compression:  uncompressed
     Data Start:   0x2008012c
     Data Size:    3083360 Bytes = 2.9 MiB
     Architecture: ARM
     OS:           Linux
     Load Address: 0x80001000
     Entry Point:  0x80001000
     Hash algo:    sha256
     Hash value:   c1ed2b6ee1b997739ffd46f80e73f2a65a01389a571c2afad28632a2d11cf29c
   Verifying Hash Integrity ... sha256+ OK
## Loading ramdisk from FIT Image at 20080000 ...
   Using 'conf-aspeed-ast2500-evb.dtb' configuration
   Trying 'ramdisk-1' ramdisk subimage
     Description:  obmc-phosphor-initramfs
     Type:         RAMDisk Image
     Compression:  uncompressed
     Data Start:   0x20377a1c
     Data Size:    1082032 Bytes = 1 MiB
     Architecture: ARM
     OS:           Linux
     Load Address: unavailable
     Entry Point:  unavailable
     Hash algo:    sha256
     Hash value:   01a609ad06481fc8ea58318f4c1115d3ac339a9ba3fa73786d6092280b47b915
   Verifying Hash Integrity ... sha256+ OK
## Loading fdt from FIT Image at 20080000 ...
   Using 'conf-aspeed-ast2500-evb.dtb' configuration
   Trying 'fdt-aspeed-ast2500-evb.dtb' fdt subimage
     Description:  Flattened Device Tree blob
     Type:         Flat Device Tree
     Compression:  uncompressed
     Data Start:   0x20370ea0
     Data Size:    27311 Bytes = 26.7 KiB
     Architecture: ARM
     Hash algo:    sha256
     Hash value:   dd954d39662a2217b3b1b835554d5e22d8e8729174f8c53532327f425eb55a7e
   Verifying Hash Integrity ... sha256+ OK
   Booting using the fdt blob at 0x20370ea0
   Loading Kernel Image ... OK
   Loading Ramdisk to 8fef7000, end 8ffff2b0 ... OK
   Loading Device Tree to 8feed000, end 8fef6aae ... OK

Starting kernel ...
[... 211 lines cut ...]
   Verifying Hash Integrity ... sha256+ OK
## Loading fdt from FIT Image at 20080000 ...
   Using 'conf-aspeed-ast2500-evb.dtb' configuration
   Trying 'fdt-aspeed-ast2500-evb.dtb' fdt subimage
     Description:  Flattened Device Tree blob
     Type:         Flat Device Tree
     Compression:  uncompressed
     Data Start:   0x20370ea0
     Data Size:    27311 Bytes = 26.7 KiB
     Architecture: ARM
     Hash algo:    sha256
     Hash value:   dd954d39662a2217b3b1b835554d5e22d8e8729174f8c53532327f425eb55a7e
   Verifying Hash Integrity ... sha256+ OK
   Booting using the fdt blob at 0x20370ea0
   Loading Kernel Image ... OK
   Loading Ramdisk to 8fef7000, end 8ffff2b0 ... OK
   Loading Device Tree to 8feed000, end 8fef6aae ... OK

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 6.1.29-1ac953d (oe-user@oe-host) (arm-openbmc-linux-gnueabi-gcc (GCC) 13.1.1 20230520, GNU ld (GNU Binutils) 2.40.0.20230419) #1 Mon May 22 00:13:30 UTC 2023
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: VIPT aliasing data cache, unknown instruction cache
[    0.000000] OF: fdt: Machine model: AST2500 EVB

williamspatrick, 2023-07-01:
What code version did you build from and how did you select the build?

AmirRMoezi, 2023-07-01:
> What code version did you build from and how did you select the build?

Sorry I do not know what you mean. How can I find that? I built from the current openbmc source code and followed the exact tutorial on https://github.com/openbmc/openbmc. I selected obmc-phosphor-image for evb-ast2500 target.

williamspatrick, 2023-07-02:
`git describe --dirty && git diff` would be useful for us to confirm which commit you are building from.

I assume by "selected obmc-phosphor-image for evb-ast2500 target" you mean you did:

`source setup evb-ast2500 && bitbake obmc-phosphor-image`?

> obmc-phosphor-image-evb-ast2500-20230612144736.static.mtd

The numbers at the end are just a timestamp when the image was built, so that isn't very useful for knowing what is present in it.

williamspatrick, 2023-07-02:
I suspect you also asked this question on Discord today?

This was my response there: https://discord.com/channels/775381525260664832/775694683589574659/1124813137531641917

AmirRMoezi, 2023-07-02:
This is the output of "git describe --dirty":
2.15.0-dev-345-g15c7c77bf-dirty

and this is a part of the output of "git diff" :

diff --git a/.eslintrc.json b/.eslintrc.json
old mode 100644
new mode 100755
diff --git a/.github/ISSUE_TEMPLATE/bug_report.md b/.github/ISSUE_TEMPLATE/bug_report.md
old mode 100644
new mode 100755
diff --git a/.github/ISSUE_TEMPLATE/question.md b/.github/ISSUE_TEMPLATE/question.md
old mode 100644
new mode 100755
diff --git a/.github/pull_request_template.md b/.github/pull_request_template.md
old mode 100644
new mode 100755
diff --git a/.github/stale.yml b/.github/stale.yml
old mode 100644
new mode 100755
diff --git a/.gitignore b/.gitignore
old mode 100644
new mode 100755
diff --git a/.gitreview b/.gitreview
old mode 100644
new mode 100755
diff --git a/OWNERS b/OWNERS
old mode 100644
new mode 100755
diff --git a/README.md b/README.md
old mode 100644
new mode 100755
diff --git a/meta-amd/COPYING.MIT b/meta-amd/COPYING.MIT
old mode 100644
new mode 100755
diff --git a/meta-amd/COPYING.apache-2.0 b/meta-amd/COPYING.apache-2.0
old mode 100644
new mode 100755
diff --git a/meta-amd/LICENSE b/meta-amd/LICENSE
old mode 100644
new mode 100755
diff --git a/meta-amd/OWNERS b/meta-amd/OWNERS
old mode 100644
new mode 100755
diff --git a/meta-amd/README.md b/meta-amd/README.md
old mode 100644
new mode 100755
...

---------------------------------------------------------------------------------
and about the next question: Yes. I did "source setup evb-ast2500 && bitbake obmc-phosphor-image"

AmirRMoezi, 2023-07-02:
> I suspect you also asked this question on Discord today?
> 
> This was my response there: https://discord.com/channels/775381525260664832/775694683589574659/1124813137531641917

Yes. I also asked the question on Discord. I am going to answer your question there

williamspatrick, 2023-07-02:
```
diff --git a/meta-amd/README.md b/meta-amd/README.md
old mode 100644
new mode 100755
```

Any idea what is the deal with all of these permission changes?  What kind of OS are you building on?  If this is WSL, it wouldn't surprise me if it doesn't work as I'm not sure anyone has actively worked on testing it there.

AmirRMoezi, 2023-07-02:
> ```
> diff --git a/meta-amd/README.md b/meta-amd/README.md
> old mode 100644
> new mode 100755
> ```
>
> Any idea what is the deal with all of these permission changes? What kind of OS are you building on? If [...]

I remember I had encountered an error during openbmc building process. The error description was nearly "The postinstall intercept hook 'update_gio_module_cache' failed" as far as I remember. So, after googling it, I found a solution which was running the command "git restore poky" in openbmc root directory. After running the command, the problem solved and there was no error again. I remember from my searches that this command relates to permissions. can it be the cause of "all of these permission changes"?

williamspatrick, 2023-07-02:
I don't know what `git restore poky` would have done in this context.  Probably nothing good, because we don't use a raw poky tree.  I'd suggest resetting your image back to non-dirty and showing what the issues are you're finding.

You might as well do a git fetch / git rebase to ensure your repository is up to date.  It appears to almost be a month old based on the commit number you listed.

```
$ git rev-list origin/master "^15c7c77bfe" | wc -l
116
$ git log 15c7c77bfe -n1 | grep Date  
Date:   Fri Jun 9 14:20:28 2023 -0500
```

AmirRMoezi, 2023-08-22:
Finally, I found the solution of my problem. It relates to permission of files located in this path: 
openbmc/poky/scripts/postinst-intercepts
Their permission should be changed to 644. Then, after executing bitbake command again, OpenBMC will be simulated successfully and without any error.

closed 2023-08-22 by AmirRMoezi