openbmc/openbmc #3940: openbmc kernel panic

bmcopen, 2023-07-10:
Hello. 
I have compiled openbmc for ast2500-evb target using procedure described in https://github.com/openbmc/openbmc. after successfully compiling it, a group of images created shown here:

******************************************************************************************************
aspeed-ast2500-evb--6.1.34+git0+aa3bee4aa1-r0-evb-ast2500-20230710101918.dtb
aspeed-ast2500-evb-evb-ast2500.dtb
aspeed-ast2500-evb.dtb
core-image-minimal-evb-ast2500-20230710101918.qemuboot.conf
core-image-minimal-evb-ast2500-20230710101918.rootfs.manifest
core-image-minimal-evb-ast2500-20230710101918.rootfs.squashfs-xz
core-image-minimal-evb-ast2500-20230710101918.spdx.tar.zst
core-image-minimal-evb-ast2500-20230710101918.static.mtd
core-image-minimal-evb-ast2500-20230710101918.static.mtd.all.tar
core-image-minimal-evb-ast2500-20230710101918.static.mtd.tar
core-image-minimal-evb-ast2500-20230710101918.testdata.json
core-image-minimal-evb-ast2500.jffs2
core-image-minimal-evb-ast2500.manifest
core-image-minimal-evb-ast2500.qemuboot.conf
core-image-minimal-evb-ast2500.spdx.tar.zst
core-image-minimal-evb-ast2500.squashfs-xz
core-image-minimal-evb-ast2500.static.mtd
core-image-minimal-evb-ast2500.static.mtd.all.tar
core-image-minimal-evb-ast2500.static.mtd.tar
core-image-minimal-evb-ast2500.testdata.json
evb-ast2500-20230710101918.all.tar
evb-ast2500-20230710101918.tar
fitImage
fitImage--6.1.34+git0+aa3bee4aa1-r0-evb-ast2500-20230710101918.bin
fitImage-evb-ast2500.bin
fitImage-its--6.1.34+git0+aa3bee4aa1-r0-evb-ast2500-20230710101918.its
fitImage-its-evb-ast2500
fitImage-its-obmc-phosphor-initramfs-evb-ast2500--6.1.34+git0+aa3bee4aa1-r0-evb-ast2500-20230710101918.its
fitImage-its-obmc-phosphor-initramfs-evb-ast2500-evb-ast2500
fitImage-linux.bin--6.1.34+git0+aa3bee4aa1-r0-evb-ast2500-20230710101918.bin
fitImage-linux.bin-evb-ast2500
fitImage-obmc-phosphor-initramfs-evb-ast2500--6.1.34+git0+aa3bee4aa1-r0-evb-ast2500-20230710101918.bin
fitImage-obmc-phosphor-initramfs-evb-ast2500-evb-ast2500
flash-evb-ast2500
image-bmc
image-kernel
image-rofs
image-rwfs
image-u-boot
 obmc-phosphor-initramfs-evb-ast2500-20230710101918.qemuboot.conf
obmc-phosphor-initramfs-evb-ast2500-20230710101918.rootfs.cpio.xz
obmc-phosphor-initramfs-evb-ast2500-20230710101918.rootfs.manifest
obmc-phosphor-initramfs-evb-ast2500-20230710101918.spdx.tar.zst
obmc-phosphor-initramfs-evb-ast2500-20230710101918.testdata.json
obmc-phosphor-initramfs-evb-ast2500.cpio.xz
obmc-phosphor-initramfs-evb-ast2500.manifest
obmc-phosphor-initramfs-evb-ast2500.qemuboot.conf
obmc-phosphor-initramfs-evb-ast2500.spdx.tar.zst
obmc-phosphor-initramfs-evb-ast2500.testdata.json
u-boot-evb-ast2500-v2019.04+gitAUTOINC+5b90e2801f-r0.bin
u-boot-evb-ast2500.bin
u-boot.bin
vmlinux
vmlinux--6.1.34+git0+aa3bee4aa1-r0-evb-ast2500-20230710101918.bin
vmlinux-evb-ast2500.bin
****************************************************************************************************

I want to run openbmc on a real ast2500 EVB. So, I programmed the SPI flash with u-boot image : "u-boot-evb-ast2500-v2019.04+gitAUTOINC+5b90e2801f-r0.bin"

then I created a tftp server and put the FIT image there : "fitImage-obmc-phosphor-initramfs-evb-ast2500--6.1.34+git0+aa3bee4aa1-r0-evb-ast2500-20230710101918.bin". I also tested all other FIT images listed above.

when I turn on the board, the u-boot images runs successfully. But, when I get the FIT Image using dhcp command and run it by bootm command, it issues a kernel panic which relates to rootfs ,I think. the output of the terminal from UART serial port is shown here:

@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
U-Boot 2019.04 (May 30 2023 - 06:05:47 +0000)

[... 257 lines cut ...]
Eth : MAC0: RGMII, , MAC1: RGMII, 
Model: AST2500 EVB
DRAM:  496 MiB (capacity:512 MiB, VGA:16 MiB, ECC:off)
MMC:   sdhci_slot0@100: 0, sdhci_slot1@200: 1
Loading Environment from SPI Flash... SF: Detected w25m512jv with page size 256 Bytes, erase size 4 KiB, total 64 MiB
*** Warning - bad CRC, using default environment

In:    serial@1e784000
Out:   serial@1e784000
Err:   serial@1e784000
Net:   
Warning: ethernet@1e660000 (eth0) using random MAC address - 4e:b4:59:0c:e9:59
eth0: ethernet@1e660000
Warning: ethernet@1e680000 (eth1) using random MAC address - d6:51:ab:ba:b4:53
, eth1: ethernet@1e680000
Hit any key to stop autoboot:  2  1  0 
Wrong Image Format for bootm command
ERROR: can't get kernel image!
ast# 
@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@

when I want to run the openbmc on qemu, I choose the MTD file : "core-image-minimal-evb-ast2500-20230710101918.static.mtd", and it runs successfully without any error, but in real board I get the kernel panic error.
Meanwhile, I compiled core-image-minimal in ubuntu 22.04 LTS running on vmware.

Does anybody know what is the problem and how can I solve it?

williamspatrick, 2023-07-10:
This is not a bug report.  The issue template for "I have a question..." clearly says:
> We do not field questions or feature requests by Github Issues!
And then proceeds to describe how to get support.

I have attempted to answer your question here, but I am closing the issue and not providing further support.

> I want to run openbmc on a real ast2500 EVB. So, I programmed the SPI flash with u-boot image : "u-boot-evb-ast2500-v2019.04+gitAUTOINC+5b90e2801f-r0.bin"

> then I created a tftp server and put the FIT image there : "fitImage-obmc-phosphor-initramfs-evb-ast2500--6.1.34+git0+aa3bee4aa1-r0-evb-ast2500-20230710101918.bin". I also tested all other FIT images listed above.

> when I turn on the board, the u-boot images runs successfully. But, when I get the FIT Image using dhcp command and run it by bootm command, it issues a kernel panic which relates to rootfs ,I think. 

This method didn't end up loading a rootfs, so it is expected that you ended up with the kernel panic saying it cannot find a rootfs.

Please try booting `flash-evb-ast2500`.

closed 2023-07-10 by williamspatrick

mdmillerii, 2023-07-11:
As mentioned in our flash layout documentation for [MTD](https://github.com/openbmc/docs/blob/master/architecture/code-update/flash-layout.md#jffs2-on-mtd-partition) the `init` script source package is [here](  https://github.com/openbmc/openbmc/blob/master/meta-phosphor/recipes-phosphor/initrdscripts/obmc-phosphor-initfs.bb).  The `init` script has [options](https://github.com/openbmc/openbmc/blob/e6c34c0667cfce0c535fe6451dc3e66d077413c0/meta-phosphor/recipes-phosphor/initrdscripts/files/obmc-init.sh#L208).  The default is to process options supplied in locations that are accessible at the U-boot prompt and a file supplied by a bitbake recipe extention.  (Configuring a full option list is also possible but will affect factory reset).

There is an option to download one image using kernel `ip=` and optionally flash it 
- download the `rofs` and then perform a code update for the fit
- invoke the debug shell to download multiple images
- download the complete `bmc` flash image.

It's also possible to build an initramfs containing the images to be flashed.