openbmc/openbmc #3941: Boot failure with evb-ast2600

Ma5300408, 2023-07-12:
I have built a image with  u-boot version **Aspeed-v2019.04** and Linux version **Linux-dev6.1.29** , demo board nor-flash using macronix **mx25l51245g** . 
First, I show the board information in the first paragraph. Then I show the problem I met , can anyone tell me some information to fix this situation ? Log said Squashfs have some issue , but the filesystem type in **rofs = mtd4 squashfs**   rwfs = mtd5 jffs2 Is auto generated by Yocto project bitbake , I didn't touch this part after image built . 

> U-Boot SPL 2019.04 (May 15 2023 - 08:09:51 +0000)
> Trying to boot from RAM
> 
> U-Boot 2019.04 (May 15 2023 - 08:09:51 +0000)
> 
> SOC: AST2620-A3 
> RST: Power On 
> eSPI Mode: SIO:Enable : SuperIO-2e
> Eth: MAC0: RMII/NCSI, MAC1: RMII/NCSI, MAC2: RMII/NCSI, MAC3: RMII/NCSI
> Model: AST2600 EVB
> DRAM:  already initialized, 1 GiB (capacity:1024 MiB, VGA:0 MiB, ECC:off)
> MMC:   
> sdhci_slot0@100: 1, sdhci_slot1@200: 2, emmc_slot0@100: 0
> Loading Environment from SPI Flash... SF: Detected mx66l51235l with page size 256 Bytes, erase size 64 KiB, total 64 MiB
>  ***Warning - bad CRC, using default environment
> In:    serial@1e784000
> Out:   serial@1e784000
> Err:   serial@1e784000
> Model: AST2600 EVB
> Net:   Could not get PHY for eth0: addr 0
> Failed to initialize PHY: -19
> eth-1: ftgmac@1e670000Could not get PHY for eth1: addr 0
> Failed to initialize PHY: -19
> , eth-1: ftgmac@1e690000
> Hit any key to stop autoboot:  2  1  0 
>  Loading kernel from FIT Image at 83000000 ...
>    Using 'conf-aspeed-ast2600-evb.dtb' configuration
>    Verifying Hash Integrity ... OK
>    Trying 'kernel-1' kernel subimage
>      Description:  Linux kernel
>      Type:         Kernel Image
>      Compression:  uncompressed
>      Data Start:   0x8300012c
>      Data Size:    3826752 Bytes = 3.6 MiB
>      Architecture: ARM
>      OS:           Linux
>      Load Address: 0x80001000
>      Entry Point:  0x80001000
>      Hash algo:    sha256
>      Hash value:   185b2a38840f05216e81b7650e1f9ee4dfb6eb76e114bfc78cc391e7efc230f7
>    Verifying Hash Integrity ... sha256+ OK
>  Loading ramdisk from FIT Image at 83000000 ...
>    Using 'conf-aspeed-ast2600-evb.dtb' configuration
>    Verifying Hash Integrity ... OK
>    Trying 'ramdisk-1' ramdisk subimage
>      Description:  obmc-phosphor-initramfs
>      Type:         RAMDisk Image
>      Compression:  uncompressed
>      Data Start:   0x833b0504
>      Data Size:    1039504 Bytes = 1015.1 KiB
>      Architecture: ARM
>      OS:           Linux
>      Load Address: unavailable
>      Entry Point:  unavailable
>      Hash algo:    sha256
>      Hash value:   5e1a3fd9d769b8f6a240ca792926612e39a3a5383016d45f7b2979252bae066f
>    Verifying Hash Integrity ... sha256+ OK
>  Loading fdt from FIT Image at 83000000 ...
>    Using 'conf-aspeed-ast2600-evb.dtb' configuration
>    Verifying Hash Integrity ... OK
>    Trying 'fdt-aspeed-ast2600-evb.dtb' fdt subimage
>      Description:  Flattened Device Tree blob
>      Type:         Flat Device Tree
>      Compression:  uncompressed
>      Data Start:   0x833a6680
>      Data Size:    40376 Bytes = 39.4 KiB
>      Architecture: ARM
[... 223 lines cut ...]
[  OK  ] Started BMC health monitoring.
[  OK  ] Started Phosphor Download Manager.
[  OK  ] Finished Clear one time boot overrides.
[  OK  ] Started Phosphor Log Manager.
[  OK  ] Started Avahi mDNS/DNS-SD Stack.
[  OK  ] Started Serial Getty on ttyS4.
[  OK  ] Reached target Login Prompts.
         Starting Phosphor Ldap config updater...
[  OK  ] Started Rsyslog config updater.
[  OK  ] Started System Logging Service.
[  OK  ] Started Phosphor DBus Service Discovery Manager.
[   75.122926] SQUASHFS error: xz decompression failed, data probably corrupt
[   75.130745] SQUASHFS error: Failed to read block 0x86a948: -5
[  OK  ] Started Phosphor Settings Daemon.
[  OK  ] Started Phosphor Dump Manager.
**[   75.315530] SQUASHFS error: xz decompression failed, data probably corrupt
[   75.323324] SQUASHFS error: Failed to read block 0x87f930: -5
[   75.391061] SQUASHFS error: xz decompression failed, data probably corrupt
[   75.398950] SQUASHFS error: Failed to read block 0x886fbc: -5
[   75.491517] SQUASHFS error: xz decompression failed, data probably corrupt
[   75.499346] SQUASHFS error: Failed to read block 0x886fbc: -5
[   75.505825] SQUASHFS error: Unable to read data cache entry [886fbc]
[   75.512973] SQUASHFS error: Unable to read page, block 886fbc, size 6cbc
[   75.520552] SQUASHFS error: Unable to read data cache entry [886fbc]
[   75.527707] SQUASHFS error: Unable to read page, block 886fbc, size 6cbc**

williamspatrick, 2023-07-12:
What image did you flash onto the module?  The log is that the image is corrupted. Either:

- Something went wrong in the build process. 
- You flashed on the wrong image and left partial data from a previous image. 
- Something went wrong in your flashing process. 
- Your NOR chip is bad. 
- Your board is having bad signal integrity between the BMC and NOR chip.

closed 2023-07-12 by williamspatrick

IgorLinux1990, 2023-11-14:
How you solved this problem, thnaks?