openbmc/openbmc #3942: Inconsistency in MAC address persistency

cm-jishnu, 2023-07-18:
**Unexpected behaviour you saw**
Though the "persist-mac" is enabled by default, Updated MAC address
is not persisted across reboot on all interfaces.

**Expected behaviour**
Consistent behaviour on all ethernet interfaces.
If "persist-mac" doesn't promise persistency across reboot then the
behaviour should be the same for all interfaces.

**To Reproduce**
Steps to reproduce the behaviour:
1. Update the MAC address through redfish or bustle command
2. Verify the MAC address is updated in d-bus and network config file
3. Reboot the system
4. Check the MAC address value of all interfaces

**Screenshots**
If applicable, add screenshots to help explain your problem.

**OpenBMC Information:**
 - /etc/os-release:
 - uname -a:
 - Any additional patches or changes:

**Additional context**
Add any other context about the problem here.

cm-jishnu, 2023-07-18:
@sunharis

williamspatrick, 2023-07-18:
You didn’t fill out half the information in this template, like what version you are running. 

Which interfaces are working and which aren’t?  Is it only eth0 that works?

Isn’t there a config file in phosphor-networkd that controls which interfaces this is enabled for (and where/how to obtain the data for them)?

williamspatrick, 2023-09-25:
No response from originator.

closed 2023-09-25 by williamspatrick