openbmc/openbmc #3943: Load kernel on evb-ast2600 board at error memory address

Ma5300408, 2023-07-19:
First time boot OpenBMC the default configure environment value **bootcmd** is **run bootspi** , and the bootspi describe some line code to execute and it can successfully boot into OpenBMC , but the second time to boot the board show following message on my console .
   **Loading kernel from FIT Image at 20100000** , but in default configure is load kernel at 0x83000000 not 20100000 . U-boot load incorrect address in memory then cause kernel load failed . I have a workaround to command **env default -f -a  ,  saveenv** to recover bootcmd to run bootspi . 
Above way doesn't what I want . I hope I can rewrite the **board-spl.h** in U-boot **include/configs** to hard write the configure enviroment . I see the OpenBMC U-boot build folder have the configure file **#define BOOTCOMMAND "run bootspi"** , then I expect this way will work on my board . But the result this way doesn't work .

What I want to do is without type any command in boot process and make board can normally boot-up. 
---------------------------------------------------------------------------------------------------------------------------------------------------
U-Boot SPL 2019.04 (May 15 2023 - 08:09:51 +0000)
Trying to boot from RAM

U-Boot 2019.04 (May 15 2023 - 08:09:51 +0000)

SOC: AST2620-A3
RST: Power On
eSPI Mode: SIO:Enable : SuperIO-2e
Eth: MAC0: RMII/NCSI, MAC1: RMII/NCSI, MAC2: RMII/NCSI, MAC3: RMII/NCSI
Model: AST2600 EVB
DRAM:  already initialized, 1 GiB (capacity:1024 MiB, VGA:0 MiB, ECC:off)
MMC:

sdhci_slot0@100: 0, sdhci_slot1@200: 1, emmc_slot0@100: 2
Loading Environment from SPI Flash... SF: Detected mx66l51235l with page size 256 Bytes, erase size 64 KiB, total 64 MiB
OK
In:    serial@1e784000
Out:   serial@1e784000
Err:   serial@1e784000
Model: AST2600 EVB
Net:   Could not get PHY for eth0: addr 0
Failed to initialize PHY: -19
eth-1: ftgmac@1e670000Could not get PHY for eth1: addr 0
Failed to initialize PHY: -19
, eth-1: ftgmac@1e690000
Hit any key to stop autoboot:  0
 Loading kernel from FIT Image at 20100000 ...
   Using 'conf-aspeed-ast2600-evb.dtb' configuration
   Verifying Hash Integrity ... OK
   Trying 'kernel-1' kernel subimage
     Description:  Linux kernel
     Type:         Kernel Image
     Compression:  uncompressed
     Data Start:   0x20100128
     Data Size:    3453920 Bytes = 3.3 MiB
     Architecture: ARM
     OS:           Linux
     Load Address: 0x80001000
     Entry Point:  0x80001000
     Hash algo:    sha256
     Hash value:   f28f6132b546caa2950d2c07a4ade9dabb078468bf92a82962d15b573cf0b1f8
   Verifying Hash Integrity ... sha256 error!
Bad hash value for 'hash-1' hash node in 'kernel-1' image node
Bad Data Hash
ERROR: can't get kernel image!

------------------------------------------------------------------------------------------------------------------------------------------------
ast# printenv
baudrate=115200
bootargs=console=ttyS4,115200n8 root=/dev/ram rw
**bootcmd=bootm 20100000**
bootdelay=2
bootfile=all.bin
**bootspi=fdt addr 20100000 && fdt header get fitsize totalsize && cp.b 20100000 ${loadaddr} ${fitsize} && bootm; echo Error loading kernel FIT image**
fdtcontroladdr=bdf8c328
gatewayip=192.168.0.1
ipaddr=192.168.0.45
loadaddr=0x83000000
netmask=255.255.255.0
serverip=192.168.0.81
stderr=serial@1e784000
stdin=serial@1e784000
stdout=serial@1e784000
verify=yes

Environment size: 475/131068 bytes
ast#
-------------------------------------------------------------------------------------------------------------------------------------------------

mdmillerii, 2023-09-25:
Note that the fit has to be copied into RAM before the crypto engine can calculate the hash for signature comparison.  This is likely why you see a RAM address, although I haven't followed what you are trying to accomplish.

amboar, 2023-09-25:
> Note that the fit has to be copied into RAM before the crypto engine can calculate the hash for signature comparison. This is likely why you see a RAM address, although I haven't followed what you are trying to accomplish.

There's also an issue where there's an initial mismatch between the address contained in `bootcmd` and the kernel address on the flash. This later gets fixed when the u-boot environment is updated by userspace (once userspace has been reached). I haven't root-caused the problem.

```
Warning: ethernet@1e660000 (eth0) using random MAC address - c6:4c:58:c9:e6:c4
eth0: ethernet@1e660000
Hit any key to stop autoboot:  0 
Wrong Image Format for bootm command
ERROR: can't get kernel image!
ast# printenv bootcmd
bootcmd=bootm 20100000
```
```
ast# setenv bootcmd bootm 20080000
ast# boot
## Loading kernel from FIT Image at 20080000 ...
   Using 'conf-aspeed-bmc-opp-romulus.dtb' configuration
   Trying 'kernel-1' kernel subimage
```
https://discord.com/channels/775381525260664832/775694683589574659/1098858893804843068

LookOuta, 2024-05-22:
Hello, Could you tell me how to solve this problem ? I met the same problem in using the AST2500. 

ast# 
DRAM Init-V12-DDR4
0abc1-8Gb-Done
Read margin-DL:0.3901/DH:0.3882 CK (min:0.30)

U-Boot 2019.04 (Feb 29 2024 - 08:14:28 +0000)

SOC : AST2500-A2 
RST : WDT3 - Boot
RST : Power On 
LPC Mode : SIO:Enable : SuperIO-4e
Eth : MAC0: RMII/NCSI, , MAC1: RGMII, 
Model: AST2500 EVB
DRAM:  960 MiB (capacity:1024 MiB, VGA:64 MiB, ECC:off)
MMC:   sdhci_slot0@100: 0, sdhci_slot1@200: 1
Loading Environment from SPI Flash... SF: Detected gd25b512m with page size 256 Bytes, erase size 4 KiB, total 64 MiB
*** Warning - bad CRC, using default environment

In:    serial@1e784000
Out:   serial@1e784000
Err:   serial@1e784000
Net:   
Warning: ethernet@1e660000 (eth0) using random MAC address - 9a:e0:ce:93:73:c9
eth0: ethernet@1e660000Could not get PHY for eth1: addr 0
Failed to initialize PHY: -19
, eth-1: ethernet@1e680000
Hit any key to stop autoboot:  0 
Wrong Image Format for bootm command
ERROR: can't get kernel image
------------------------------------------------------

The following is my modification of bootcmd:

ast# printenv bootcmd
bootcmd=bootm 20100000
ast# setenv bootcmd bootm 20080000
ast# boot
Wrong Image Format for bootm command
ERROR: can't get kernel image!